cartographer-project/cartographer_ros #397: Documentation: Tuning guide

ojura, 2017-06-28:
It would be nice if there was a comprehensive tuning guide with a little more in-depth explanation of the parameters available via Lua configuration files, implications what happens when they are changed, and recommendations for when making adjustments (e.g. what to do if Cartographer fails to recognize loop closures because they end up being too far away). Although this would require some effort, I think it would be beneficial in long term, because the volume of tuning-related issues on Github should be decreased, saving you time.

ojura, 2017-07-06:
Some notes: A larger Huber scale adjusts the Huber loss function so that outliers have a larger effect. The parameter should be tuned so that it encompasses only inlier constraints.

Increasing the loop closure search window:  see `SPARSE_POSE_GRAPH.constraint_builder.fast_correlative_scan_matcher.linear_search_window` and `SPARSE_POSE_GRAPH.constraint_builder.max_constraint_distance` in `cartographer/configuration_files/sparse_pose_graph.lua`.

msy22, 2017-08-21:
What is the unit for the `SPARSE_POSE_GRAPH.constraint_builder.max_constraint_distance` parameter? Is it meters? pixels? voxels?

wohe, 2017-08-22:
Meters. Similar with other distances.

msy22, 2017-08-23:
Great, thanks for the clarification!

MichaelGrupp, 2020-11-06:
Closing, we have a tuning guide in the docs. Extension proposals via PRs are always welcome :+1:

closed 2020-11-06 by MichaelGrupp