cartographer-project/cartographer_ros #399: How to add a trimmer at runtime?

Mofef, 2017-06-28:
If I remember correctly yesterday @SirVer showed me that he just added a trimmer for a certain `trajectory_id` in the code. 
I wonder how this should be implemented on the long run, since the id of the trajectory, which is to be trimmed, is probably only known at run time. So will there be a ros service to add a trimmer to a trajectory one trimmer at a time?  Or do you want to integrate this functionality into the `/start_trajectory` service? Can I contribute to this?

SirVer, 2017-07-03:
My thought is this: extend start_trajectory_main.cc to take configuration for which trimmer to use for a new trajectory. For now this can be a bool: `pure_localization`. This will need some extending in the trajectory options and therefore changes to cartographer and cartographer_ros.

Mofef, 2017-07-03:
Ok, then `pure_localization` would hust be handed over as an argument to start_trajectory service, right? 

I came up with the alternative of having an add_trimmer service, because I thought there may be use cases in the future for which it would make sense 
* to add a trimmer after a trajectory was started (e.g. when a map was downloaded from the cloud) 
* to change the trimmer after the trajectory was started (e.g. change the number of submaps to keep)
* to add several different trimmers to one trajectory that trim according to different measures. 

Those would require to add trimmers independent of the start of a trajectory.

SirVer, 2017-07-03:
My concern is that we need another binary then, similar to the start_trajectory_main, since the trimmers will grow more options over time probably. For now my simple solution suffices, I think.

Mofef, 2017-07-03:
Ok I will start to work on this `pure_ localization` parameter then.

SirVer, 2017-07-05:
Fixed in #389.

closed 2017-07-05 by SirVer