cartographer-project/cartographer_ros #413: node crashes if any node subscribes to OccupancyGrid

jihoonl, 2017-07-05:
It crashes in [map_builder_bridge.cc#170](https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/map_builder_bridge.cc#L170). 

* rviz was subscribing to occupancy grid topic.
* trajectory_id 0 was for loaded map from serialized data.

```
F0705 19:26:54.898905 26894 map_builder_bridge.cc:170] Check failed: trajectory_options_.count(0) == 1 (0 vs. 1) 
[/cartographer_node FATAL 1499275614.899195430, 1498225574.110075237]: F0705 19:26:54.000000 26894 map_builder_bridge.cc:170] Check failed: trajectory_options_.count(0) == 1 (0 vs. 1) 
*** Check failure stack trace: ***575.107289   Duration: 1.908086 / 591.608141               
    @     0x7f9bec9f95cd  google::LogMessage::Fail()
    @     0x7f9bec9fb433  google::LogMessage::SendToLog()
    @     0x7f9bec9f915b  google::LogMessage::Flush()
    @     0x7f9bec9fbe1e  google::LogMessageFatal::~LogMessageFatal()
    @           0x51fe83  cartographer_ros::MapBuilderBridge::BuildOccupancyGrid()
    @           0x4fa7ee  cartographer_ros::Node::SpinOccupancyGridThreadForever()
    @     0x7f9be9064c80  (unknown)
    @     0x7f9bec5b76ba  start_thread
    @     0x7f9be8ad33dd  clone
    @              (nil)  (unknown)
```

SirVer, 2017-07-06:
This is sorta expected behavior. I am unclear what should be done here....

The problem is that we do not have scans from the loaded data, so we cannot generate an occupancy grid (since we do not have `trajectory_nodes`). And even if we had the scans, we would still not know which trajectory builder options (in this case: resolution and laser_scan_insert options) we should use for generating the ROS map.

I am not sure how this should be properly handled... probably by configuring the parameters for the published map through ROS somehow.

jihoonl, 2017-07-06:
One option would be having a separate ROS node that generates OccupancyGrid from submaps as you  suggested in [#343](https://github.com/googlecartographer/cartographer_ros/issues/343#issue-229398154). In this case, we can probably extend this node to provide other representations like [gridmap](https://github.com/ethz-asl/grid_map) for the future use case too.

Would it generate different map from the current occupancy grid map generation approach?

SirVer, 2017-07-06:
> Would it generate different map from the current occupancy grid map generation approach?

Probably. Currently we generate the occupancy grid by replaying all scans that we have in memory and inserting them using the current opinion of where they belong. A separate node would either

1) listen to the lasers (or other sources of points) themselve and use Cartographer's trajectory to periodically generate an occupancy grid. This would give the most similar results.
2) listen to the submaps similar to how the rViz plugin does it and create the occupancy grid by blending the submaps together (like the rViz plugin does). This would result in a map that is more similar to the visualization, but maybe not as good of quality.

Entropy512, 2017-07-17:
I encountered similar behavior with attempting to use WriteAssets when a map had been loaded - it dies at https://github.com/googlecartographer/cartographer_ros/blob/master/cartographer_ros/cartographer_ros/map_builder_bridge.cc#L86

For that particular use case, I was able to work around it by amending the WriteAssets service to accept a trajectory ID - This allows an occupancy grid to be written, however it will only write an occupancy grid for the current run's trajectory.  Anything from the loaded map that does not exist in the current trajectory will not be written out - and after reading through the discussion here I understand why.  I need to talk to management internally as to what company policy is with regards to me pushing my changes out to a public location, I'll publish those to a branch somewhere if I can obtain permission.

Is my understanding of the commit history for cartographer correct in that you now have scans from loaded data, leaving only the issue of unknown options remaining?  What about assuming that the existing scans were using the same options as for the currently running trajectory (typically in this case trajectory ID 1)?

SirVer, 2017-07-21:
This is now fixed by publishing a map using the Texture information from Cartographer directly. This does allow to generate a ROS map by launching Cartographer with a serialized state and running the `occupancy_grid_node`. If an offline ROS map (PGM and YAML) is desired, the [`map_saver`](http://wiki.ros.org/map_server#map_saver) can be used.

PR #436 (2017-07-21): Enable OccupancyGridNode and remove generating of assets in the node.
This will allow us to keep way less information in memory for SLAMing and fixes crashes related to multi trajectory (e.g. this fixes #413).

It also means that to get to an X-Ray or a map, users need to run the asset writer now after SLAMing, which is inconvenient.
commit 99da2f20 (2017-07-21): Enable OccupancyGridNode and remove generating of assets in the node. (#436)

This will allow us to keep way less information in memory for SLAMing and fixes crashes related to multi trajectory (e.g. this fixes #413).

It also means that to get to an X-Ray or a map, users need to run the asset writer now after SLAMing, which is inconvenient.
Remove dependency on YAML and delete more dead code.
closed 2017-07-21 by SirVer