cartographer-project/cartographer_ros #418: Cannot use my own bag to replace cartographer_paper_revo_lds.bag

liuhuaiyu, 2017-07-10:
[myissue.zip](https://github.com/googlecartographer/cartographer_ros/files/1134826/myissue.zip)

Thank you for filing an issue!

It would help us tremendously if you provid the following so that we can easily
reproduce your issue:

1. A link to a Git repository containing a branch of `cartographer_ros`
   containing all the configuration, launch, and URDF files required to
   reproduce your issue.
2. A link to a bag file we can use to reproduce your issue.

We will likely be unable to help you without both.

Please remove this boilerplate text before submitting your issue.

liuhuaiyu, 2017-07-10:
1. I edited /launch/demo_revo_lds.launch:25 
   <remap from="scan" to="horizontal" />
   Nothing else changed.
2. After step 1, I remaked it. 
   $catkin_make_isolated --install --use-ninja
   $source install_isolated/setup.bash
3. Using my own bag, test.bag, to roslaunch demo_revo_lds.launch.
4. I used UTM-30LX. Below shows the steps that I recorded my bag.
   $rosrun hokuyo_node hokuyo_node
   $rosbag record -O test.bag /scan

It turns out that:

1. [ WARN] [1499671086.289104514, 1499666123.004629894]: W0710 15:18:06.000000 28889 tf_bridge.cc:51] "horizontal_laser_link" passed to lookupTransform argument target_frame does not exist. 
2. Rvie shows nothing.

Would you please help me figure out what were the problems?

Thank you so much!

You can see my bag in the myissue.zip.

SirVer, 2017-07-10:
You did not provide what you changed in the configs, but looking at your bag your `/scan` topic has frame_id: laser and you likely didn't fix your urdf.

closed 2017-07-10 by SirVer

liuhuaiyu, 2017-07-10:
I fixed it and it worded well. Many thanks.