cartographer-project/cartographer_ros #421: Check failed: msg.range_min >= 0.f (-2.35183 vs. 0)

OZzhuma, 2017-07-11:
i Build & Installation follow this :
[image]
 
but i I have encountered the following problem:
Check failed: msg.range_min >= 0.f (-2.35183 vs. 0)
[image]

wohe, 2017-07-11:
Could it be that the bag you are using is pretty old? I remember that we uploaded fixed bags when we added the check, i.e. you would simply have to download the new demo bag. The old bags by mistake contained negative min ranges as the check says.

OZzhuma, 2017-07-12:
YES！The bag is pretty  old .I download the new bag and it works.

closed 2017-07-19 by SirVer