cartographer-project/cartographer_ros #424: More configuration problems, loop not closing

MasterEric, 2017-07-14:
I am having issues generating a loop-closed map from a bag file with significant overlap.
[image]
As seen in the map above, there is significant error by the end of the mapping; by 180 seconds there are symptoms of desyncronization (when it finishes the first loop it drives back down the road it was going on, but it starts to be crooked) and by 300 seconds the map is heavily duplicated. I believe this is an issue with my configuration, but I'm not sure which options need to be tweaked to resolve this (a tuning guide like #397 would be really helpful, rather than having to rely on the developers to spend time they could be spending solving bugs).

I am using the configuration you provided me earlier in issue #391. I have put this in MasterEric/cartographer_ros. 

I also noticed watching the map generate in RVIZ, the map appears to reset every 50-100 seconds, starting first at around 70 seconds into the bag. This does not seem normal.

Here is the bag file I am using, recorded with 1 Velodyne VLP-16 on top of a vehicle moving at around 7-10 miles per hour: https://mega.nz/#!AJ8TEb7R!D9JHWGzUhCQnYYfrUUNLWgWqL0H1RUPv6T2hnNUGUj8

I read that laser scans are being used for velocity estimation, however I have a geometry_msgs/TwistStamped, is there/should there be an option to provide this to cartographer to reduce error in velocity calculation?

SirVer, 2017-07-18:
I think there is more than a tuning issue here. Looking at your bag, your IMU data's `angular velocities.y` component is always 0. That can't be right and probably leads to a disagreement between scan matcher and IMU.  Also `linear_acceleration.z` looks a bit funny - plateauing at some spots, i.e. reporting exactly the same data for a few measurements.

> a tuning guide like #397 would be really helpful, rather than having to rely on the developers to spend time they could be spending solving bugs).

I agree that a tuning guide would be helpful now. Though the tuning issues are very helpful to nail down what the actual problems are people are expiring and also building up a really nice test set of example data for Cartographer spreading many different platforms and applications. The tuning guide we will eventually write will be more useful than the one we could have written at the beginning - which would not have been hepful at all.

> I also noticed watching the map generate in RVIZ, the map appears to reset every 50-100 seconds, starting first at around 70 seconds into the bag. This does not seem normal.

This comes from your trajectory drifting upwards, i.e. the lower submaps are faded out. This is likely due to non-calibrated IMU biases or a slightly wrong URDF.

> I read that laser scans are being used for velocity estimation, however I have a geometry_msgs/TwistStamped, is there/should there be an option to provide this to cartographer to reduce error in velocity calculation?

Cartographer uses odometry to estimate velocities as well, but using TwistStamped is not implemented. It would be nice to have if it improves the tracking accuracy.

closed 2017-07-18 by SirVer

MasterEric, 2017-07-24:
Hey @SirVer, sorry I haven't been able to get back to you until now.

Since I am getting my IMU data from the vehicle via CAN messages, there are two issues with the IMU data. The IMU data is filtered a bunch by the vehicle itself, meaning the plateaus are likely just very similar values that have been smoothed out. Additionally, the vehicle provides no pitch data (aka `angular_velocity.y`) whatsoever, but I don't understand how that would be an issue here; the recorded map contains no hills or slopes, and thus no significant changes in pitch.

I discovered an issue in which the linear acceleration for Y and Z were both negated, and a fix is in the bag linked below. Using it does not result in any fixes to the trajectory however.

https://mega.nz/#!NIcm1RoJ!RPIA9I4M9JW-nQxP5Ei6jmwMB0v-LY_hUZtSWai20eY

SirVer, 2017-07-25:
> the recorded map contains no hills or slopes

The shaking of the vehicle will already provide plenty of pitch from my experience with the google cars. I suggest you get yourself a cheap IMU and replace the poor one from your car. This also gives you the opportunity to mount it closer to the laser.

> I discovered an issue in which the linear acceleration for Y and Z were both negated, and a fix is in the bag linked below. Using it does not result in any fixes to the trajectory however.

It sounds to me like you have plenty of explanations in the issues with your sensor data, so I doubt this is a Cartographer issue altogether. I really suggest you try a different IMU.