cartographer-project/cartographer_ros #425: Configuring Cartographer for 3D Mapping with Velodyne 64 and IMU

saimanoj18, 2017-07-17:
Hi,

Thank you so much for this amazing piece of work. 

I am trying to get cartographer to work with my setup of Velo 64 and IMU and the bag files can be downloaded from https://uploadfiles.io/za278 (1.6 GB) and https://ufile.io/j6gtq (4.3 GB)

I would like to get the optimized trajectory and found that it can be obtained as a service call request.

I followed a post on how to get Kitti datasets to work with cartographer and I have modified only three files, namely demo_velo64_imu.launch, velo64_imu.launch and velo64_imu.lua, which are available at https://github.com/saimanoj18/cartographer_Velo64_imu. 

[image]

When I run with this setup, there is no error and the mapping or scan matching process does not start as well. I guess the problem is in TF. Could you help setup the system ?

One more question is, should IMU and Velodyne need to be calibrated or the TF should be right ? Or they cartographer automatically finds the transformation/calibration ? 

Roughly, the TF between Velo 64 and IMU is 
x = 1.2 
y = 0 
z = 2
roll = -0.7 (there is negative sign)
pitch = 2
yaw = -164 (there is negative sign)

Hoping to hear from you!

Thank you,
Sai

SirVer, 2017-07-19:
> One more question is, should IMU and Velodyne need to be calibrated or the TF should be right ? Or they cartographer automatically finds the transformation/calibration ?

You need to provide the transforms in a URDF, cartographer does not do extrinsic calibration.

Your configuration and data has some issues:

* Your [XML has invalid syntax](https://github.com/saimanoj18/cartographer_Velo64_imu/blob/master/demo_velo64_imu.launch#L28), so your remappings are not working. You use `<node /><remap ...>` but must use `<node><remap...></node>`.
* Your bags contain invalid frame ids, frame ids must not start with a leading `/` - and in fact they usually do not contain `/`, but I think that is just a convention.

closed 2017-07-19 by SirVer

GuangZhaoZhou, 2020-08-27:
Hi,
Thank you so much for this amazing work.
 which tools did you used to  calibrate between IMU and Velodyne ？
really appreciate your help