cartographer-project/cartographer_ros #426: Trajectory visualization is (arguably) in the wrong order

macmason, 2017-07-17:
When `cartographer_rviz` is being used to visualize multiple trajectories at the same time, they stack in newest-on-the-bottom order. I think that's backwards: newest should be drawn on top.

SirVer, 2017-07-18:
Mac, could you provide more arguments except you feel it is reversed? I feel the bikeshed should be blue.

@ojura You have an opinion on this?

ojura, 2017-07-18:
I think that we draw every trajectory on the same level, so the drawing order is actually undefined (or defined by random variables, such as insertion order into the vector of line strip markers, or the outcome of z fighting).

macmason, 2017-07-18:
Here's the specific experience I had: I'm slamming together multiple trajectories, one after another. I observed that each new trajectory was drawn under the previous ones (from what @ojura said, it sounds like I was seeing something that looked like that, rather than exactly that). 

This made it hard to watch the current (most recent) trajectory evolve, because it was usually behind the other ones (my robot happens to be roughly repeating a trajectory). I think the most-recent trajectory is the most interesting, so I figure it should be on top.

I like my bikesheds green.

macmason, 2017-07-18:
...and now I'm seeing them in a different order, so @ojura 's hypothesis is correct. Perhaps this should have been named "trajectory visualization should be in a deterministic order".

ojura, 2017-07-18:
I have already raised a little bit the z axis position of trajectory line strip markers in order to make them visible when juxtaposed with constraint markers. This z axis adjustment amount could be increased depending on the trajectory id, which will put the newest trajectory on top, and make the bikeshed green. :)

SirVer, 2017-07-19:
Dah, I misunderstood. I thought you were complaining about the ordering of the submaps in the selection list on the left, but you were talking about the z-position.

Ideally, we'd like to draw the submaps in the order they were created, i.e. oldest first and ignoring depth buffer newer ones on top. @ojura We could do this with z-axis fudging, but even better would be to turn off updating the z-buffer and using GL_LESS for drawing the submaps. Do you know if this is possible with ogre?

ojura, 2017-07-19:
@macmason there is also a bug in rviz with these markers related to bad depth testing, could it be related? https://github.com/ros-visualization/rviz/issues/1067#issuecomment-312281373
I mitigated the bug as described in the issue.
@SirVer I have no idea, OpenGL is not my stronger side. I'll ask around :)

macmason, 2017-07-19:
Could be related? (Note @SirVer I'm thinking about the markers that show the trajectory, not the submaps themselves.)

I'm not sure how to debug this; my OpenGL is pretty weak.