cartographer-project/cartographer_ros #429: Is there a function of save the map in online?

nnn112358, 2017-07-19:
I am not a bug, but please let me ask a question here.

Localization in Existing  for offline use are provided.(#315)
This sample launch files are offline_backpack_2d.launch　and　demo_backpack_2d_localization.launch.
(https://google-cartographer-ros.readthedocs.io/en/latest/)

By the way, This sample launch files  has a constraint of offline using rosbag files.
Cartographer_ros can save the map in offline, but cartographer_ros can save the map in online?

If cartographer_ros have the ability to save map in online, I'd like you to add documents how to save the map in online.
If there is no function, please consider addition of function.

damonkohler, 2017-07-20:
This is already supported. See our [ROS API documentation](https://google-cartographer-ros.readthedocs.io/en/latest/ros_api.html#services).

closed 2017-07-20 by damonkohler

nnn112358, 2017-07-20:
Thank you for your answer.
This is a great feature. I was impressed !