cartographer-project/cartographer_ros #437: Provide hints via RViz

kctess5, 2017-07-24:
I'm trying to use Cartographer to make a 2D map of an extremely large tunnel system. It usually works well, but occasionally it gets messed up, converges to a local minima, does not recover. I'd like to be able to provide hints via RViz that e.g. the place it's in now is very close to a place it has already been.

I think one easy way to do this would be to allow users to select a two points along the trajectory, drag/dropping the second point to the correct relative pose. This would result in an extremely high weight constraint in the solver which would hopefully guide it to the global convergence basin.

a. Does something like this already exist? I've seen allusions to similar functionality, but can't find it, so I suspect it is non-public.
b. Does this sound like a worthwhile idea? Is there a better way?
c. Implementation pointers would be appreciated

Cheers!

SirVer, 2017-07-24:
a) No. We never had manual loop closing hints in the system. Properly tuned it should never be needed.
b) I think not. It is a bandaid working around bugs or improper tuning. It might be a necessary bandaid, but so far I never saw a completely untunable case. Could you provide your data as a test case so I could give it a shot with tuning?
c) I think this could be implemented as a landmark contstraint. This itself is not implemented, but googlecartographer/cartographer#246 is the start.

kctess5, 2017-07-25:
Thanks for the response. Here's a gigantic rosbag where it fails consistently. I drove a [RACECAR](https://github.com/mit-racecar) around the MIT basement system for around an hour. I made sure to do the main loops as soon as possible, and drove each hallway at least twice (usually more). The car has a Hokuyo UST-10LX, odometry, and a Razor 9DOF IMU, all of which are included in the bag. The odometry is generally not terrible, but can be unreliable when on sharp turns.

https://www.dropbox.com/s/8mvsa2kx7q5f7qg/huge_basement_walk2.bag?dl=0

There's a few places where it consistently fails:
- First is a long and fairly featureless hallway below building 4 starting around time 1280 sec. The car is moving forwards the whole time but gets "stuck" and doesn't advance in the SLAM map, which makes it think the hallway is much shorter than it is. This causes an important loop closure to be missed. It's not surprising that scan matching doesn't work particularly well in that hallway, but I'd expect odometry to help it along more.
- Things really start to go down hill around 2450 seconds when it I guess bumps a wall and rotates the scan. It doesn't recover and from then on the whole map is rotated with respect to the previously mapped sections.

I've tried playing with some parameters but I'm not familiar enough to know how to best optimize them.

Here's the result.
[image]

It should look more like this (everything to the left of building 66/32).
[image]

damonkohler, 2017-07-25:
I was able to get better results by using some experimental code to improve odom integration, enabling correlative scan matching, and increasing the min score for global constraints to 0.75. This is an interesting test case for the odom integration I'm working on, so I'll keep experimenting with it.

Your bag file, however, is missing a static transform for the IMU and I do not see a URDF for your platform. Either a corrected bag file or a URDF would be helpful for continued tuning.

kctess5, 2017-07-25:
Sorry I forgot to include that - I have been downloading this repo: [https://github.com/mit-racecar/racecar-simulator](https://github.com/mit-racecar/racecar-simulator) and including this line in my racecar_2d.launch:

```
<param name="robot_description" command="$(find xacro)/xacro '$(find racecar_description)/urdf/racecar.xacro'" />
```

Maybe this is incorrectly configured, but I've been using this:
```
options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_imu_link",
  published_frame = "base_link",
  odom_frame = "odom",
  provide_odom_frame = false,
  use_odometry = true,
  use_laser_scan = true,
  use_multi_echo_laser_scan = false,
  num_subdivisions_per_laser_scan = 10,
  num_point_clouds = 0,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-2,
  trajectory_publish_period_sec = 50e-3,
}
MAP_BUILDER.use_trajectory_builder_2d = true
MAP_BUILDER.num_background_threads = 8.0
TRAJECTORY_BUILDER_2D.use_imu_data = true
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = false
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.ceres_solver_options.max_num_iterations = 50.0
MAP_BUILDER.sparse_pose_graph.constraint_builder.max_constraint_distance = 100.0
```

I found that enabling use_online_correlative_scan_matching caused worse results on the long and featureless hallways (but better on the feature-full places), so it's interesting to hear it worked for you.

I'm happy to provide anything else you need.

damonkohler, 2017-08-18:
With the new pose extrapolator that includes odometry, I was able to improve results. A section of the bag at the beginning has very bad IMU data, so I disabled the IMU. Also, near the end of the bag, the odometry becomes very bad and breaks the map.

You can see my configuration here: https://github.com/damonkohler/cartographer_ros/tree/racecar

Ideally, we can come up with a way to handle this kind of bad data better. But we're not there yet.

Would you be alright with us using your data as an official Cartographer demo? The long hallways make it quite interesting from a data diversity perspective.

SirVer, 2017-10-12:
@kctess5 Would you be alright with us using your data as an official Cartographer demo? The long hallways make it quite interesting from a data diversity perspective.

SirVer, 2017-11-14:
Closing, as we currently have no goal of providing manual hints via rViz and the bug became stale.

closed 2017-11-14 by SirVer

kctess5, 2018-03-18:
Sorry for the late response, but feel free to use this as an official demo!

kctess5, 2018-03-18:
@SirVer @damonkohler @gaschler bump in case you don't see this closed issue