cartographer-project/cartographer_ros #438: build issue

nobertgmuedu, 2017-07-24:
Hi i'm trying to build cartographer and having some problems.

I'm following these instructions:
https://google-cartographer-ros.readthedocs.io/en/latest/

Below is the output from running the steps, and i'm struggling to figure out what I need to do to fix this to get this up and running.

Any help our advice ?

Thank You much!

**ubuntu@alfresco:~$ sudo apt-get update**
Hit:1 http://us-east-1.ec2.archive.ubuntu.com/ubuntu xenial InRelease
Get:2 http://us-east-1.ec2.archive.ubuntu.com/ubuntu xenial-updates InRelease [102 kB] 
Get:3 http://us-east-1.ec2.archive.ubuntu.com/ubuntu xenial-backports InRelease [102 kB]                     
Hit:4 http://ppa.launchpad.net/webupd8team/java/ubuntu xenial InRelease                                                                           
Hit:5 http://security.ubuntu.com/ubuntu xenial-security InRelease                                        
Hit:6 http://packages.ros.org/ros/ubuntu xenial InRelease                          
Fetched 204 kB in 0s (434 kB/s)                      
Reading package lists... Done
**ubuntu@alfresco:~$ sudo apt-get install -y python-wstool python-rosdep ninja-build**
Reading package lists... Done
Building dependency tree       
Reading state information... Done
ninja-build is already the newest version (1.5.1-0.1ubuntu1).
python-rosdep is already the newest version (0.11.5-1).
python-wstool is already the newest version (0.1.13-1).
0 upgraded, 0 newly installed, 0 to remove and 3 not upgraded.
**ubuntu@alfresco:~$ mkdir catkin_ws**
**ubuntu@alfresco:~$ cd catkin_ws**
**ubuntu@alfresco:~/catkin_ws$ wstool init src**
Writing /home/ubuntu/catkin_ws/src/.rosinstall

update complete.
**ubuntu@alfresco:~/catkin_ws$ wstool merge -t src https://raw.githubusercontent.com/googlecartographer/cartographer_ros/master/cartographer_ros.rosinstall**
     Performing actions: 

     Add new elements:
  cartographer,  cartographer_ros,  ceres-solver

Config changed, maybe you need run wstool update to update SCM entries.
Overwriting /home/ubuntu/catkin_ws/src/.rosinstall

update complete.
**ubuntu@alfresco:~/catkin_ws$ wstool update -t src**
[cartographer] Fetching https://github.com/googlecartographer/cartographer.git (version None) to /home/ubuntu/catkin_ws/src/cartographer
Cloning into '/home/ubuntu/catkin_ws/src/cartographer'...
remote: Counting objects: 3759, done.
remote: Compressing objects: 100% (3/3), done.
remote: Total 3759 (delta 0), reused 1 (delta 0), pack-reused 3756
Receiving objects: 100% (3759/3759), 1.55 MiB | 0 bytes/s, done.
Resolving deltas: 100% (2975/2975), done.
Checking connectivity... done.
[cartographer] Done.
[cartographer_ros] Fetching https://github.com/googlecartographer/cartographer_ros.git (version None) to /home/ubuntu/catkin_ws/src/cartographer_ros
Cloning into '/home/ubuntu/catkin_ws/src/cartographer_ros'...
remote: Counting objects: 2094, done.
remote: Compressing objects: 100% (36/36), done.
remote: Total 2094 (delta 14), reused 9 (delta 3), pack-reused 2055
Receiving objects: 100% (2094/2094), 585.80 KiB | 0 bytes/s, done.
Resolving deltas: 100% (1513/1513), done.
Checking connectivity... done.
[cartographer_ros] Done.
[ceres-solver] Fetching https://ceres-solver.googlesource.com/ceres-solver.git (version 1.12.0rc4) to /home/ubuntu/catkin_ws/src/ceres-solver
Cloning into '/home/ubuntu/catkin_ws/src/ceres-solver'...
remote: Sending approximately 11.96 MiB ...
remote: Counting objects: 25, done
remote: Total 10406 (delta 6007), reused 10406 (delta 6007)
Receiving objects: 100% (10406/10406), 11.96 MiB | 0 bytes/s, done.
Resolving deltas: 100% (6007/6007), done.
[... 607 lines cut ...]
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/ubuntu/catkin_ws/build_isolated/cartographer_ros_msgs/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/ninja" "cmTC_41448"
[1/2] Building C object CMakeFiles/cmTC_41448.dir/CheckFunctionExists.c.o
[2/2] Linking C executable cmTC_41448
FAILED: : && /usr/bin/cc  -DCHECK_FUNCTION_EXISTS=pthread_create   CMakeFiles/cmTC_41448.dir/CheckFunctionExists.c.o  -o cmTC_41448  -rdynamic -lpthreads && :
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

Suggestions?

Thanks!

-Nathan

SirVer, 2017-07-24:
These are the relevant lines here:

~~~
CMake Error at /usr/share/catkin/cmake/catkinConfig.cmake:75 (find_package):
  Could not find a package configuration file provided by
  "message_generation" with any of the following names:

    message_generationConfig.cmake
    message_generation-config.cmake
~~~

You are missing `ros-kinetic-message-generation`. I do not know why this is not installed on your system, it should have been pulled in by the ROS dependency tools. Our CI should also show this error since we run through exactly the same step on a fresh install every time.

closed 2017-07-24 by SirVer

nobertgmuedu, 2017-07-24:
Hi, I did the following:

cd src
git clone https://github.com/ros/message_generation.git
git clone https://github.com/ros/message_runtime.git
cd ..
catkin_make_isolated --install --use-ninja

appears to have gotten past the error.   

Thank You for assistance.