cartographer-project/cartographer_ros #457: To SHA or not to SHA in docs?

SirVer, 2017-07-28:
In #449 and #445 I used SHA to refer to git blobs like files and directories. This guarantees that links in documentation are always correctly linking files and lines in files. This also means that documentation cannot bit rot, it will always be correct, but might not be updated to HEAD.

The alternative is to link `master` everywhere. This means that documentation will bit rot, links to lines will go stale. But it also means that the documentation is always referring to master, which it is likely also documenting.

Which of these approaches is the one we want to adopt?

macmason, 2017-07-28:
A pattern I've seen, but have never used myself: a presubmit check for "some documentation got updated on each commit, or a magic string in the commit message indicates no changes."

Then, at least, the odds of the documentation bitrotting go down.

(I'm avoiding voting on the actual issue, because I don't have a strong feeling either way.)

MichaelGrupp, 2020-10-17:
SHA seems to be the simplest approach for all parties, it should be clear from using the permalink view on GitHub that it is a snapshot and not master.

closed 2020-10-17 by MichaelGrupp