cartographer-project/cartographer_ros #460: What is .pbstream and can I read it

myu20, 2017-07-31:
Hi, I initially installed Cartographer in June 2017. At that time the trajectory output was a `.pb` file, from which I could get a human readable version with the command `protoc --decode cartographer.mapping.proto.Trajectory -I. $(find . -name '*.proto') < ~/.ros/map.pb`. The output was a timestamped trajectory of the form
```
node {
  timestamp: 636324486985516000
  pose {
    translation {
      x: 0.0011769832389950886
      y: 0.015897276467976743
      z: 0.00029333615062883135
    }
    rotation {
      x: 0.000467514026206124
      y: -0.010101004818702106
      z: -0.0072682970219754535
      w: 0.99992245848900119
    }
  }
}
```
Today I used `git pull` to update Cartographer and now Cartographer outputs a `.pbstream`. Does the `.pbstream` also contain a timestamped trajectory and is there a command to get a human readable version of it?

Thank you!

wohe, 2017-08-01:
Hi, yes and no. The `.pbstream` files contain much more information now, it is written by `MapBuilder::SerializeState()`. You can read it using the `ProtoStreamReader` in `cartographer/io/proto_stream.h`. The first proto message it contains is a `cartographer.mapping.proto.SparsePoseGraph` which can contain one or more trajectories. I hope this helps. Let me know if there are more questions.

myu20, 2017-08-01:
After I get a `mapping::proto::SparsePoseGraph graph` using the `ProtoStreamReader`, do I access the trajectories using `graph.trajectory_size()` and `graph.trajectory(0)`?

wohe, 2017-08-01:
Yes, for example. It's a normal protocol buffer message: https://developers.google.com/protocol-buffers/docs/reference/cpp-generated#repeatedmessage

myu20, 2017-08-01:
Great, thanks!

closed 2017-08-01 by myu20

litgm, 2017-08-17:
So reading this into python will be a little more intricate?

wohe, 2017-08-17:
But only a tiny bit. Using Python's `gzip` and `struct` modules, an implementation of `ProtoStreamReader` in Python should be rather straightforward. Let me know if you have specific questions.

woshidaye, 2018-09-10:
@myu20 Hi, I want to extract data from pbstream. How do you extract trajectory from pbstream?  Can you share your code for extracting data? I look forward to your reply.

myu20, 2018-09-10:
@woshidaye Hi, I used the following code:
```
cartographer::io::ProtoStreamReader reader(pbstream_filename);
cartographer::mapping::proto::PoseGraph pose_graph;
reader.ReadProto(&pose_graph);
```
You can access the trajectories with `pose_graph.trajectory(i)`, and from the trajectory you can access the trajectory nodes with `pose_graph.trajectory(i).node(j)`.

Relevant protos: [pose_graph](https://github.com/googlecartographer/cartographer/blob/2bd987ffb40e2d6667682c5c96432173f371094f/cartographer/mapping/proto/pose_graph.proto) [trajectory](https://github.com/googlecartographer/cartographer/blob/2bd987ffb40e2d6667682c5c96432173f371094f/cartographer/mapping/proto/trajectory.proto)

woshidaye, 2018-09-11:
@myu20 Thank you for your answer. I have another question.I want to save the submaps and pose-graph. Does the pbstream file contain this data? If not, what are your suggestions for saving this information? Looking forward to your reply again. Thank you.

myu20, 2018-09-11:
@woshidaye As I mentioned in my previous comment, you can read the pose graph with
```
cartographer::io::ProtoStreamReader reader(pbstream_filename);
cartographer::mapping::proto::PoseGraph pose_graph;
reader.ReadProto(&pose_graph);
```
I imagine you can read the submaps in a similar way. I would suggest looking at the [serialization proto](https://github.com/googlecartographer/cartographer/blob/213882a9b3d8aee10b7ea059e732f1632d3684de/cartographer/mapping/proto/serialization.proto) and the [WritePbStream method](https://github.com/googlecartographer/cartographer/blob/213882a9b3d8aee10b7ea059e732f1632d3684de/cartographer/io/internal/mapping_state_serialization.cc#L213) to see what you can read from the pbstream file. All of the relevant protos are probably in [cartographer/cartographer/mapping/proto](https://github.com/googlecartographer/cartographer/tree/213882a9b3d8aee10b7ea059e732f1632d3684de/cartographer/mapping/proto).

woshidaye, 2018-09-11:
@myu20 Thank you very much.

woshidaye, 2018-09-29:
@myu20 
I wrote a script according to your method. What file should I put this script under? Do we need to write the script as a ROS node?

Flandoe, 2018-11-29:
@woshidaye  Have you solved the problem? I met the same problem to compile the code. Thanks!