cartographer-project/cartographer_ros #463: I tried to run your demo: demo_backpack_2d_localization.launch, but I got a [FATAL] bug.

liuhuaiyu, 2017-08-01:
Thank you for filing an issue!

It would help us tremendously if you provid the following so that we can easily
reproduce your issue:

1. A link to a Git repository containing a branch of `cartographer_ros`
   containing all the configuration, launch, and URDF files required to
   reproduce your issue.
2. A link to a bag file we can use to reproduce your issue.

We will likely be unable to help you without both.

Please remove this boilerplate text before submitting your issue.

wohe, 2017-08-01:
Sorry, I could not reproduce this issue. Did you make sure to use the latest code, especially including #453? If you did and the problem persists: Could you please add some more context, i.e. when did this happen, what did the output say about the values of `current` and `time`, etc. Thanks.

liuhuaiyu, 2017-08-02:
I saw the output, but didn't record it. Sometimes `current` = `time`. So I changed pose_extrapolator.cc:125 `CHECK(imu_it->time < current)` to `CHECK(imu_it->time <= current)` and it worked.
But it seemed you have changed the code. I ran the #435 version, now used the latest code (#436) and it worked without any problem.
Thx

closed 2017-08-04 by SirVer