cartographer-project/cartographer_ros #472: tf transformation between map and base_frame is floating

kilmarnock, 2017-08-04:
I use ros_cartographer with an imu_um7 and a hokuyo 30lx. The map gets produced nicely.
The published transform from map to base_link is floating and has a constant rotation around the map frame:

`rosrun  tf tf_echo map base_link`
> At time 1501861249.096
> - Translation: [-7892.858, -12323.256, 0.000]
> - Rotation: in Quaternion [0.845, 0.209, 0.226, -0.438]
>             in RPY (radian) [-2.244, -0.601, 0.189]
>             in RPY (degree) [-128.543, -34.420, 10.857]
> At time 1501861250.096
> - Translation: [3310.326, -986.246, 0.000]
> - Rotation: in Quaternion [0.068, -0.152, -0.249, 0.954]
>             in RPY (radian) [0.214, -0.259, -0.539]
>             in RPY (degree) [12.253, -14.855, -30.866]
> At time 1501861251.096
> - Translation: [-23581.840, 4440.330, 0.000]
> - Rotation: in Quaternion [-0.122, -0.038, 0.235, 0.964]
>             in RPY (radian) [-0.256, -0.015, 0.480]
>             in RPY (degree) [-14.686, -0.863, 27.515]

`rostopic echo /imu/data `
> header: 
>   seq: 7921
>   stamp: 
>     secs: 1501835252
>     nsecs: 200998606
>   frame_id: imu_link
> orientation: 
>   x: 0.0292388603
>   y: 0.0773436672
>   z: 0.1662687429
>   w: 0.982573411
> orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> angular_velocity: 
>   x: -0.376964859662
>   y: -0.0414879759792
>   z: -0.2777762571
> angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
> linear_acceleration: 
>   x: -0.561970487193
>   y: 0.855681448182
>   z: 9.02666983562
> linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]

The output is quite stable. 

my config is
`cat backpack_2d.lua`
> include "map_builder.lua"

> include "trajectory_builder.lua"
> 
> options = {
>   map_builder = MAP_BUILDER,
>   trajectory_builder = TRAJECTORY_BUILDER,
>   map_frame = "map",
>   tracking_frame = "base_link",
>   published_frame = "base_link",
>   odom_frame = "odom",
>   provide_odom_frame = true,
>   use_odometry = false,
>   num_laser_scans = 1,
>   num_multi_echo_laser_scans = 0,
>   num_subdivisions_per_laser_scan = 10,
>   num_point_clouds = 0,
>   lookup_transform_timeout_sec = 0.2,
>   submap_publish_period_sec = 0.3,
>   pose_publish_period_sec = 5e-3,
>   trajectory_publish_period_sec = 30e-3,
> }
> 
> MAP_BUILDER.use_trajectory_builder_2d = true
> TRAJECTORY_BUILDER_2D.scans_per_accumulation = 10
> 
> return options

The tf tree looks good. the tf from map to base_link is entirely published by cartographer.
I assume the um7 driver is messing sth?
 Could you please give me a hint?

damonkohler, 2017-08-10:
I don't think I understand the issue. What do you mean by "map to base_link is floating?" I'm confused because you first state that "the map gets produced nicely. trajectory and /scan_matched_points2 are as expected." Sounds like everything is working?

kilmarnock, 2017-08-10:
Hi! Thank you for your time.
`rosrun tf tf_echo map base_link` shows that the position between two transforms is

> [-7892.858, -12323.256, 0.000
> [3310.326, -986.246, 0.000]
> [-23581.840, 4440.330, 0.000]

The rviz visualization shows a circular movement around the map frame. The numbers increase over time.
Everything is working, but the transforms are incorrect.

damonkohler, 2017-08-10:
Ah, I see the issue. If "provide_odom_frame" is true, you generally want published_frame == odom_frame.

So, change published_frame to "odom" and try again.

ojura, 2017-08-10:
Maybe it could be an issue in the recently merged extrapolator? IIRC so far it's not used for anything but publishing transforms.

damonkohler, 2017-08-10:
Please double check that you don't have some other node or bag file providing the odom transform. This is a rather common mistake. You can use the [tf filter](https://github.com/googlecartographer/cartographer_ros/blob/102a3b0db3f88e942c8a46ffbf57c92fa5072f48/cartographer_ros/launch/demo_pr2.launch#L32) to remove odom from your bag if it's there.

kilmarnock, 2017-08-12:
I am not able to reproduce the bahaviour after a complete restart of the system. What I observe is that the transform tree is broken after a rosbag loop / rosbag restart.
Steps to reproduce:
1) Start cartographer
2) Start a bag with /scan topic in it: `rosbag play scan.bag --clock --loop`

In the first loop, tf gets published, f.e.`map -> base_link`
In the second loop, tf is not published any more, the tf simply freezes.

ojura, 2017-08-12:
Regarding the previous version of your post - Cartographer does not handle backward jumps in time. You have to restart the node if you restarted playing the bag.

wohe, 2017-10-04:
Closing this since we cannot reproduce. Please re-open should you have more information.

closed 2017-10-04 by wohe

Magnusgaertner, 2018-03-07:
I am having the same issue, 
building a 2d map and trajectory from only laserscan data (using trajectory_builder_2d use_imu_data = false, use_online_correlative_scan_matching = true) displays a valid map and trajectory in rviz but produces weird transformations between map/odom and base_link. 

 ```
    header: 
      seq: 0
      stamp: 
        secs: 1520438896
        nsecs: 167912000
      frame_id: odom
    child_frame_id: base_link
    transform: 
      translation: 
        x: 19287.4468349
        y: -86139.647836
        z: -0.05
      rotation: 
        x: 0.0
        y: 0.0
        z: 0.00201267379083
        w: 0.99999797457

    header: 
      seq: 0
      stamp: 
        secs: 1520438896
        nsecs: 186653300
      frame_id: odom
    child_frame_id: base_link
    transform: 
      translation: 
        x: 63577.4245965
        y: -64820.7914003
        z: -0.05
      rotation: 
        x: 0.0
        y: 0.0
        z: -0.175923979231
        w: 0.984403755342

    header: 
      seq: 0
      stamp: 
        secs: 1520438896
        nsecs: 186765600
      frame_id: odom
    child_frame_id: base_link
    transform: 
      translation: 
        x: 63577.4246018
        y: -64820.7914066
        z: -0.05
      rotation: 
        x: 0.0
        y: 0.0
        z: -0.175923502492
        w: 0.984403840541
```