cartographer-project/cartographer_ros #476: Node::FinalizeTrajectory blocks callback servicing

ojura, 2017-08-07:
As I mentioned in #468, one final issue is that `MapBuilderBridge::FinalizeTrajectory()` (more precisely, when `SparsePoseGraph::RunFinalOptimization()` calls `SparsePoseGraph::WaitForAllComputations()`) can take a long time and make `Node::FinishTrajectory()` hold the mutex for too long. @wohe do you have a suggestion how to do more fine-grained locking?

wohe, 2017-08-08:
The question is whether fine-grained locking is the solution here: Originally, `FinishTrajectory()` was not just finishing the trajectory but also wrote a map. For this map to be correctly optimized, we had to call `RunFinalOptimization()`. Now that writing the result is separate, I wonder whether we should rather move `RunFinalOptimization()` out of `FinishTrajectory()`. We could then not hold the mutex during the call to `FinishTrajectory()`. Note though, that no new data should arrive then as this is an assumption `WaitForAllComputations()` makes. Thoughts?

ojura, 2017-08-08:
Great, that's more or less what I had in mind (when I mentioned "fine-grained locking").` I will prepare a PR and we can continue there.

ojura, 2017-08-08:
By the way, does `sensor_bridges_.erase(trajectory_id)` have to be called after `RunFinalOptimization`?

wohe, 2017-08-08:
I don't think so. You might want to check, but I think sensor bridges are only accessed until subscribers are removed which happens before the call to `FinishTrajectory()`.

PR #480 (2017-08-10): Release mutex in Node during final optimization
Enables servicing callbacks during the final optimization. This (along with #443 and #468) enables watching in RViz the progress of the final optimization using the offline node.

One question: should the online node also call `RunFinalOptimization()` in `Node::HandleFinishTrajectory`, or only at the end (which is handled by `FinishAllTrajectories`) ? If so, could this be done asynchronously for similar reasons (to avoid stopping callback servicing)?

> No new data should arrive then as this is an assumption WaitForAllComputations() makes.

This  [... cut]
commit 102a3b0d (2017-08-10): Release mutex in Node during final optimization (#480)

Enables servicing callbacks during final optimization, thus
enabling visualization of its progress in RViz. Fixes #476.
closed 2017-08-10 by damonkohler