cartographer-project/cartographer_ros #485: BUILD ERROR: the following packages/stacks could not have their rosdep keys resolved

duchengyao, 2017-08-16:
ubuntu 16.04, kinetic

```
$ rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ceres-solver: No definition of [eigen] for OS version []
cartographer: No definition of [eigen] for OS version []
```
How to resolve it ?
Thanks

duchengyao, 2017-08-16:
I resolved it by using `'rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y --os=ubuntu:xenial'`

Please write it to your wiki. Thanks

ojura, 2017-08-16:
Have you run `sudo rosdep init` and `rosdep update` before installing? Specifying the parameter you mentioned is not necessary on a fresh Ubuntu installation when installing Cartographer.

wohe, 2017-08-16:
This does not seem like a Cartographer ROS issue. Please re-open if you think this is incorrect. Adding `--os=ubuntu:xenial` should not be necessary. It is also only correct if you use xenial.

Your issue seems related to `rosdep`, e.g. see https://answers.ros.org/question/40081/rosdep-doesnt-recognize-os/ and https://github.com/ros-infrastructure/rosdep/issues/10 and references.

closed 2017-08-16 by wohe

samhoff20, 2020-10-18:
Had this issue for ubuntu: focal with ROS2 Foxy, came to say this solved my issue as well.

samhoff20, 2020-11-03:
Putting this here for reference, had this happen again on a different computer and had already forgot. `rosdep install --from-paths ros2-linux/share --ignore-src --rosdistro foxy -y --skip-keys "console_bridge fastcdr fastrtps osrf_testing_tools_cpp poco_vendor rmw_connext_cpp rosidl_typesupport_connext_c rosidl_typesupport_connext_cpp rti-connext-dds-5.3.1 tinyxml_vendor tinyxml2_vendor urdfdom urdfdom_headers"
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rosidl_adapter: No definition of [python3-empy] for OS version []
rcl_logging_noop: No definition of [python3-empy] for OS version []
tf2_eigen: No definition of [eigen] for OS version []
rcutils: No definition of [python3-empy] for OS version []
ament_clang_tidy: No definition of [clang-tidy] for OS version []
tf2_sensor_msgs: No definition of [eigen] for OS version []
launch: No definition of [python3-lark-parser] for OS version []
ament_pep257: No definition of [pydocstyle] for OS version []
laser_geometry: No definition of [eigen] for OS version []
rmw_cyclonedds_cpp: No definition of [cyclonedds] for OS [debian]
ros2pkg: No definition of [python3-empy] for OS version []
ament_clang_format: No definition of [clang-format] for OS version []
rosidl_parser: No definition of [python3-lark-parser] for OS version []
ros2doctor: No definition of [python3-ifcfg] for OS version []
ament_flake8: No definition of [python3-flake8] for OS version []
ament_mypy: No definition of [python3-mypy] for OS version []
ament_package: No definition of [python3-flake8] for OS version []
rcl_logging_log4cxx: No definition of [log4cxx] for OS version []
rviz_rendering: No definition of [eigen] for OS version []
orocos_kdl: No definition of [eigen] for OS version []
`

alicegong, 2020-12-31:
I'm still seeing this with ROS 2 foxy. How did you solve this? @samhoff20

JayantTeotia16, 2021-05-17:
Getting the same error. Did you find a solution? @alicegong

ztaye3, 2021-08-17:
@JayantTeotia16  @alicegong Have you found any solution ?

BartvanS, 2022-02-22:
For not wanting to include --os={os version} include something like this in your ~/.bashrc:
export ROS_OS_OVERRIDE=ubuntu:20.3
your linux mint version(which in my case is 20.3) can be found in the desktop "System info" app.

more information: http://wiki.ros.org/ROS/EnvironmentVariables#ROS_OS_OVERRIDE

adededede, 2023-06-14:
When I run this command I get the following error:
```ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rviz: No definition of [libogre-dev] for OS version [bullseye]```
Does any of you know how to solve that?

AktugHakan, 2023-08-17:
I am also getting the same error while following the [installation guide](https://docs.ros.org/en/foxy/Installation/Alternatives/Ubuntu-Development-Setup.html﻿﻿﻿﻿﻿﻿﻿).
When I run this command:
`rosdep install --from-paths src --ignore-src -y --skip-keys "fastcdr rti-connext-dds-5.3.1 urdfdom_headers" --rosdistro foxy`
it gives:
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros2doctor: No definition of [python3-ifcfg] for OS version [jammy]
```

I am trying to install the system on a virtual ARM Ubuntu 22.04 machine. Previous steps are completed without any error or warning.