cartographer-project/cartographer_ros #488: simulated odometry leads to poor map (bag attached)

BrannonKing, 2017-08-22:
I've been struggling to use simulated odometry with the current code (current as in a number of builds from the past 30 days). My Cartographer builds from last year seem to work better in that regard. I still get very good maps with CeresScanMatcher+Laser+IMU. I also (not in real time) get good results with OnlineRealTimeScanMatcher+Laser. I've attached my bag and config files to reproduce the "shaky" map from my odometry. I would appreciate it if someone could help me clarify whether my odometry data is complete junk or the transforms are wrong or whether the bug lies somewhere else. I have attempted to reduce the odometry influence with the weights, but they don't seem to affect the final output.

[bag.zip](https://github.com/googlecartographer/cartographer_ros/files/1243286/bag.zip)

jihoonl, 2017-08-24:
Could it be relevant to https://github.com/googlecartographer/cartographer/issues/467?

BrannonKing, 2017-08-24:
It's true that the changes to incorporate odometry into the pose extrapolator could be related, but I don't think the odometry has worked right for several months -- pre "extrapolator" times. In thinking about this further, it may be that my odometry is just too frequent for Cartographer. I've been bringing it in at 90 to 100 Hz. It could be related to the Cartographer issues dealing with high-frequency data causing velocity estimates to go bad. I've previously used 66Hz odometry without a problem, though. 

I'm not a ROS1 bag expert. Is there some easy way to play that bag whilst reducing the odometry down to 10Hz? Wouldn't I need to aggregate the odometry that is there somehow?

jpapon, 2017-08-31:
You can do something like this to produced an odometry topic throttled to 50hz: "rosrun topic_tools throttle messages /odometry 50.0 /odometry_throttle" if you're using the online node. Offline, you'd have to either just run the bag with this running and record it, or use something along the lines of the code in the rosbag cookbook to create a new bagfile which has every n odometry messages dropped.

BrannonKing, 2017-09-23:
I've tracked down most of my odometry issues and written them up here: https://github.com/googlecartographer/cartographer/issues/534. I'm certain it's not a problem with the ROS node portion.

closed 2017-09-23 by BrannonKing