cartographer-project/cartographer_ros #490: How to run offline node on multiple consecutive bagfiles?

Koisen, 2017-08-24:
Hello there :-)

(short disclaimer first: I am pretty new to google cartographer and actually ROS and SLAM in general so please bear with me)

I am wondering if it is possible to run the cartographer_offline_node on multiple consecutive bagfiles to recieve a map / trajectory based on the combined data of a number of bagfiles?

Background for this being the fact that my data of a 40ish minute field test is divided into a number of bags each containing about 60 seconds of testing.

So far i tried to list multiple - for starters just two - different bagfiles as arguments for the offline node. See below:

<launch>
  <node name="cartographer_offline_node" pkg="cartographer_ros" type="cartographer_offline_node"
    args="-configuration_directory /home/sua/google_cartographer/config
          -configuration_basename config.lua
          -urdf_filename $(find cartographer_ros)/urdf/backpack_2d.urdf
          -bag_filenames /home/sua/sensors_2017-05-17-11-38-36_38.bag
          -bag_filenames /home/sua/sensors_2017-05-17-11-39-36_39.bag"
      output="screen">
    <remap from="scan" to="/scan/pfodm30/front"/>
  </node>
</launch>

But if i use this the offline node only takes the second argument "-bag_filenames /home/sua/sensors_2017-05-17-11-39-36_39.bag" into account and creates the corresponding files.

I reckon i am  just using the wrong syntax to feed multiple bagfiles to the offline node.

Any education on how to do this correctly would be greatly appreciated :-)

wohe, 2017-08-24:
Hi! Yes, this is definitely possible. Whether you get good results with very short bags is another question, but given enough overlap it could work. Regarding your flag issue, it is a comma-separated list. There is also a `--help` for the command line flags, e.g.
```
    -bag_filenames (Comma-separated list of bags to process.) type: string
```
Just let me know if you have more questions.

Koisen, 2017-08-24:
Thank you very much for the response!

I edited my "bag_filenames" argument to a comma-seperated list and it solved my initial problem. Now the node processes one bagfile after another.

But:

After the node finishes the first bagfile it jumps back to the starting position of the first bag before starting with the second one. This results in a final .pgm file that shows the different maps overlapping each other. My goal obviously is to create a map / trajectory consisting of the multiple map parts combined and put together like pieces of a puzzle :-)

Any thoughts on what might be a smart way to achieve this?

wohe, 2017-08-24:
This is what I was referring to. You need enough overlap, so that a good transformation between the different trajectories is established. If you add a separate bag, we have to initially assume that this could come from a different source at an unknown pose.

Koisen, 2017-08-24:
Mhhh...ok. I guess there is zero time overlap between my multiple bagfiles because they were created one after the other consecutively during a single test drive. There should be overlap in the laser scan data though because it was collecting data continuously during the whole process - might i be able to build on that to solve the issue?

Is there any way to manually tell cartographer to always use the last position of one bagfile as the starting point for the following bagfile?

The whole problem doesn't seem to exist for the cartographer online SLAM. It is running smoothly throughout all my bagfiles if i just "rosbag play * --clock" all of them.

wohe, 2017-08-25:
1. There is currently no way to tell Cartographer explicitly that a new trajectory is connected to another trajectory. googlecartographer/cartographer#469 would solve this.
2. If the online SLAM works: you have (almost) no time gap between the bags? Then you should not present your data in separate bags to the offline node. This functionality is intended for separately collected bags, possibly even a different robot. Merge the bags into one and use this bag?

Koisen, 2017-08-27:
> Merge the bags into one and use this bag?

Thanks for the suggestion. I will give that a try.

ludennis, 2018-09-06:
Hi @Koisen, 

Just wanting to chip in a little by giving my two cents here. I think if your bags are just segments of a single vehicle run, maybe you can try to combine all the bags into one and then feeding that one bag to Cartographer.

To combine all the bags you can open three terminals, one with
> roscore

Another one with
> rosparam set use_sim_time true
> rosbag play --clock vehicle_route_*

And, lastly, the third terminal with command:
> rosbag record --buffsize=0 -a

Hope this helps, it's not a solution to the issue topic, but it's another way to go for a shot.

ojura, 2018-09-06:
@ludennis a better way would be to use the rosbag api directly:
https://answers.ros.org/question/10683/is-there-a-way-to-merge-bag-files/

ludennis, 2018-09-18:
Hi all,

For running multiple bag files, would it be possible to run Cartographer with multiple bag files AND with multiple configuration files?

Dennis

gaschler, 2018-09-18:
@ludennis 
Yes, this feature is implemented by offline_node.

```
DEFINE_string(
    configuration_basenames, "",
    "Comma-separated list of basenames, i.e. not containing any "
    "directory prefix, of the configuration files for each trajectory. "
    "The first configuration file will be used for node options. "
    "If less configuration files are specified than trajectories, the "
    "first file will be used the remaining trajectories.");
DEFINE_string(
    bag_filenames, "",
    "Comma-separated list of bags to process. One bag per trajectory. "
    "Any combination of simultaneous and sequential bags is supported.");
DEFINE_string(urdf_filenames, "",
              "Comma-separated list of one or more URDF files that contain "
              "static links for the sensor configuration(s).");
```

ludennis, 2018-09-19:
@gaschler thanks!

Will give it a try and report asap my result! Thanks, again!

Dennis Lu

maisa32, 2019-08-08:
I am facing the same issue as the OP: I have multiple bag files which were recorded using
`rosbag record --split` to limit the size of each rosbag. The rosbag were recorded consecutively in a single run by our robot (2 lidars and 1 imu), so there is *almost zero* time between each bag.

When I cartographer offline_node_3D with the rosbags, cartographer initialize the trajectory in each ROSBAG at the same point (this is the same problem the OP had). I wonder if there is now a solution to this and if so how can I implement it in the configuration files? I know that I could merge the bag files together but I would rather not as they are quite heavy (hours of recording).

Thanks!