cartographer-project/cartographer_ros #492: InitGoogleLogging() twice! error on ROS built from source [SOLVED]

at-wat, 2017-08-30:
The backend of the rosconsole library (that is linked from every ros nodes) can be selected from log4cxx, glog, and print, and also cartographer_ros uses glog.
If rosconsole is built with glog, InitGoogleLogging() is called twice and cartographer_ros dead just after launch, showing following error message.
`F0829 22:25:58.712759 28570 utilities.cc:326] Check failed: !IsGoogleLoggingInitialized() You called InitGoogleLogging() twice!`

Building rosconsole with log4cxx or print backend can avoid the error.
(It can be selected by the cmake arg `ROSCONSOLE_BACKEND`.)

It's just a tip for the built-from-source-ROS users.
Please close the issue if no more comment.

damonkohler, 2017-09-29:
Thanks! I've captured this in the FAQ.

closed 2017-09-29 by damonkohler

agilmor, 2018-06-08:
Instead of forcing `rosconsole` to use log4cxx, I think that we should use an option also into cartographer, right?

It's easy to do it by changing all these lines:
```
  google::InitGoogleLogging(argv[0]);
```
With:
```
#ifndef ROSCONSOLE_BACKEND_GLOG
  google::InitGoogleLogging(argv[0]);
#endif
```

And adding this into cartographer's ```CMakeLists.txt``` when desired:
```
add_definitions(-DROSCONSOLE_BACKEND_GLOG)
```

Ideally we should read `ROSCONSOLE_BACKEND`  with something like:
```
if(ROSCONSOLE_BACKEND STREQUAL "glog" OR "@ROSCONSOLE_BACKEND@" STREQUAL "glog")
    add_definitions(-DROSCONSOLE_BACKEND_GLOG)
endif()
```
But it doesn't work, not sure why...

Beck-Sisyphus, 2019-10-23:
Required to use rosconsole with glog and build-from-source, anyway we can initialize glog separately for rosconsole and our own project?

PR #515 (2017-09-29): Adds two new FAQ entries.
Fixes #492.
Obsoletes #371.