cartographer-project/cartographer_ros #497: how to provide  odometry data?

danningwang, 2017-09-04:
Hello, I am using cartographer to map a round corridor, but the result is not good enough. So I want to include odometry data in SLAM when using bagfile to improve the loop closure. 
map_frame = "map",   tracking_frame = "base_link",   published_frame = "base_footprint",
  odom_frame = "odom_combined",   provide_odom_frame = false,   use_odometry = true,
This is my lua file.
<remap from="odom" to="base/odometry_controller/odometry" />
In my launch file, I remap the odom topic to my odometry topic recorded in my bag file, but it shows  an error as following:
F0904 16:40:54.000000 26324 sensor_bridge.cc:33] Check failed: frame_id[0] != '/' ('/' vs. '/') The frame_id /base_footprint should not start with a /
So how to provide odometry data to "odom" topic? Could you explain it ?
Thank you for help!

danningwang, 2017-09-04:
remap from="odom" to="base/odometry_controller/odometry"
This is remap in my launch file.

wohe, 2017-09-04:
Hi. As the error indicates, the problem is with the frame name, not the topic name: In tf2, frame names do not contain a prefix, see http://wiki.ros.org/tf2#Frequently_asked_questions. You should make sure to (re)publish your odometry data in the frame `base_footprint` not `/base_footprint`.

danningwang, 2017-09-04:
Ok, Thanks a lot!

closed 2017-09-29 by damonkohler