cartographer-project/cartographer_ros #505: RViz slowdown, possible memory leak

ojura, 2017-09-13:
I mentioned in googlecartographer/cartographer#525 that RViz sometimes slows down on my machine after a few bag runs (even after completely restarting the Cartographer node) and I have to restart it so it feels snappy again (e.g. zooming and panning). I am not sure at all if it is related at all to the Submaps plugin and I haven't done any measurements, this is all pure speculation, but it might be worth looking into (not a high priority for me right now), so for now, I am just creating an issue.

@wohe suggested [heap profiling](http://goog-perftools.sourceforge.net/doc/heap_profiler.html).

jihoonl, 2017-09-13:
Note that a cpu usage of `cartographer_node` often hits 100% when I paused rosbag for a while(about 10mins). And it made rviz freeze. I had to resume for a moment and pause the bag again to unfreeze rviz. Maybe, some threads in `cartographer` run without sleep call if there is no data incoming?

Could it be relevant to your case?

wohe, 2017-09-13:
If you think there might be an issue with CPU usage: could you try to get some more data on where this CPU time is spent? For example there is https://perf.wiki.kernel.org/.

ojura, 2017-09-13:
I completely restart the cartographer node, so I don't think the other issue is related. I will look into this when I get the time, thanks on the link.

gaschler, 2018-02-16:
Closing as inactive, please feel free to re-open.

closed 2018-02-16 by gaschler