cartographer-project/cartographer_ros #507: rviz always dies when launching catgrapher_ros node

yinzixuan126, 2017-09-16:
[ WARN] [1505566301.586646655, 1504590505.371581463]: W0916 20:51:41.000000  9184 tf_bridge.cc:51] Lookup would require extrapolation into the past.  Requested time 1504590505.342596000 but the earliest data is at time 1504590505.363983870, when looking up transform from frame [laser] to frame [base_footprint]
[ INFO] [1505566301.660424321, 1504590505.392781053]: I0916 20:51:41.000000  9184 ordered_multi_queue.cc:172] All sensor data for trajectory 0 is available starting at '636401873053674402'.
[ INFO] [1505566301.660482399, 1504590505.392781053]: I0916 20:51:41.000000  9184 collated_trajectory_builder.cc:74] scan rate: unknown
[ WARN] [1505566301.681670460, 1504590505.398836009]: W0916 20:51:41.000000  9184 pose_extrapolator.cc:152] Queue too short for velocity estimation. Queue duration: 0 ms
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'Ogre::Exception'
  what():  OGRE EXCEPTION(5:): Could not find font Arial in MovableText::setFontName
================================================================================REQUIRED process [rviz-2] has died!
process has died [pid 9185, exit code -6, cmd /opt/ros/indigo/lib/rviz/rviz -d /home/will/cattographer_ws/install_isolated/share/cartographer_ros/configuration_files/demo_2d.rviz __name:=rviz __log:=/home/will/.ros/log/b5c06e20-9adb-11e7-b43c-1866da08c3ca/rviz-2.log].
log file: /home/will/.ros/log/b5c06e20-9adb-11e7-b43c-1866da08c3ca/rviz-2*.log
Initiating shutdown!
================================================================================
[base2laser-3] killing on exit
[rviz-2] killing on exit
[cartographer_node-1] killing on exit
Optimizing: Done.     
Optimizing: Done.     
shutting down processing monitor...
... shutting down processing monitor complete
done

yinnxinn, 2017-09-20:
rviz have replace the default font recently . i cloned the source from github and compiled it to solve this problem.

yinzixuan126, 2017-09-24:
@yinnxinn , which branch of rviz did u cloned? I cloned and complied its indigo-devel, but still have the same problem.

yinnxinn, 2017-09-24:
https://github.com/ros-visualization/rviz/pull/1141

davetcoleman, 2017-09-27:
I'm still having the same problem after building from source rviz

Mofef, 2017-09-27:
I had the same issue.

`sudo apt-get install ttf-mscorefonts-installer`
`sudo fc-cache`
and rebuilding cartographer_rviz solved it for me

Might be that I also rebuild rviz itself, so if the issue is not resolved try rebuilding rviz as well.

damonkohler, 2017-09-29:
This looks like an [rviz issue](https://github.com/ros-visualization/rviz/issues/1152), not a Cartographer issue.

closed 2017-09-29 by damonkohler