cartographer-project/cartographer_ros #508: Question about slice_pose !!

GMahmoud, 2017-09-19:
Hey guys,

I am trying to build the cartographer submaps with unity3D. But I am a little bit confused about the slice_pose and its frame. I am not sure that the pose is expressed in the map frame. 
So could you please give me some help on that and tell me which frame (position and orientation) are you using to express the slice_pose ?  

Cheers

closed 2017-09-20 by GMahmoud

wohe, 2017-10-02:
`slice_pose` is the pose of the textured rectangle which is used for visualizing a submap in the submap frame. To get the pose of the textured rectangle in the map frame, the submap pose must be multiplied by it.

GMahmoud, 2017-10-03:
Thank you @wohe for your answer. I got this by adding some rviz axis to the ogre node.

Cheers