cartographer-project/cartographer_ros #509: tuning 3d cartographer for my data

sunmk2006, 2017-09-21:
hello, I am tuning 3d cartographer for my vel-16 data,but there is a big jump during trajectory build, I can't solve it,can you help me?
the configuration files , launch file, urdf file and result png files is [here](https://github.com/sunmk2006/cartographer_ros/blob/master/slam.zip) ,and the bag file is here

[image]

cschuet, 2017-10-09:
@sunmk2006 We looked at your data and were somewhat able to reproduce your results.

The IMU data frequency is quite low (20Hz) and the data looks extremely noisy. We were able to cope with this somehow by telling Ceres to have really low confidence in the IMU data, but this is definitely a problem. It is also quite surprising that your bag contains **exactly** as many point cloud messages as it does IMU messages. How did you generate this data? Do you only query the IMU state when you receive a point cloud? In general it makes sense to sample the IMU much more frequently than that to allow Cartographer.

cschuet, 2017-10-09:
Closing for now since we believe the IMU data explains the poor performance. Please re-open if you have further questions and good luck tuning :)

sunmk2006, 2017-10-10:
I uploaded [a new bag file](http://pan.baidu.com/s/1eRGA4QQ)，The IMU data frequency is 125HZ,but the jump still exists.

sunmk2006, 2017-10-12:
[udp_jump.bag.zip](https://pan.baidu.com/s/1boUnNyB)
[udp_jump_synchronization.bag.zip](http://pan.baidu.com/s/1i44that)

In these two bag files, the laser data exists in the form of UDP packets, the difference is that the second files used the synchronization method mentioned earlier, and then cartographer will not have any warning information.

SirVer, 2017-10-13:
We further investigated the bag `2017-09-01-06-59-59.bag`. Our `rosbag_validate` tool gave the following output:

~~~
$ rosrun cartographer_ros cartographer_rosbag_validate -bag_filename ~/Downloads/2017-09-01-06-59-59.bag
I1013 17:11:03.875735 87882 rosbag_validate_main.cc:110] Time delta histogram for consecutive messages on topic "/imu" (frame_id: "imu"):
Count: 43977  Min: 0.007915  Max: 0.008078  Mean: 0.007997
[0.007915, 0.007931)                            Count: 3 (0.006822%)    Total: 3 (0.006822%)
[0.007931, 0.007948)                            Count: 1 (0.002274%)    Total: 4 (0.009096%)
[0.007948, 0.007964)                            Count: 0 (0.000000%)    Total: 4 (0.009096%)
[0.007964, 0.007980)                            Count: 0 (0.000000%)    Total: 4 (0.009096%)
[0.007980, 0.007996)                            Count: 28 (0.063670%)   Total: 32 (0.072765%)
[0.007996, 0.008013)    ####################    Count: 43934 (99.902222%)       Total: 43966 (99.974983%)
[0.008013, 0.008029)                            Count: 4 (0.009096%)    Total: 43970 (99.984085%)
[0.008029, 0.008045)                            Count: 4 (0.009096%)    Total: 43974 (99.993179%)
[0.008045, 0.008062)                            Count: 1 (0.002274%)    Total: 43975 (99.995453%)
[0.008062, 0.008078]                            Count: 2 (0.004548%)    Total: 43977 (100.000000%)
I1013 17:11:03.876127 87882 rosbag_validate_main.cc:110] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"):
Count: 5985  Min: 0.048640  Max: 1.489060  Mean: 0.052874
[0.048640, 0.192682)    ####################    Count: 5969 (99.732666%)        Total: 5969 (99.732666%)
[0.192682, 0.336724)                            Count: 2 (0.033417%)    Total: 5971 (99.766083%)
[0.336724, 0.480766)                            Count: 2 (0.033417%)    Total: 5973 (99.799500%)
[0.480766, 0.624808)                            Count: 1 (0.016708%)    Total: 5974 (99.816208%)
[0.624808, 0.768850)                            Count: 1 (0.016708%)    Total: 5975 (99.832916%)
[0.768850, 0.912892)                            Count: 1 (0.016708%)    Total: 5976 (99.849625%)
[0.912892, 1.056934)                            Count: 2 (0.033417%)    Total: 5978 (99.883041%)
[1.056934, 1.200976)                            Count: 0 (0.000000%)    Total: 5978 (99.883041%)
[1.200976, 1.345018)                            Count: 3 (0.050125%)    Total: 5981 (99.933167%)
[1.345018, 1.489060]                            Count: 4 (0.066834%)    Total: 5985 (100.000000%)
~~~

The IMU timing looks decent, the velodyne timing though shows that you have ~10 gaps > 0.5 seconds, i.e. there are 10 instances were you do not receive Velodyne data for more than half a second. That fully explains the jumps.

However, we further looked into your bag data and created the following plot:

[image]

The x-axis is the index of the packet in the rosbag, the y axis is the difference between the timestamp of the first sensor packet and the current packet, i.e. `t - t0`. I provided this data in https://gist.github.com/SirVer/337ea5cea8550a219a9fdf47d59ad2bb. 

There are a few interesting things to note here:

- Initially, your IMU and laser have a timing offset of 25 seconds. Over time, the difference in timestamp becomes smaller.
- Your velodyne not only has gaps in timing, but it also sometimes delivers the same timestamp for a few packets in a row.

This indicates that your IMU and your Velodyne are timed from different clocks that run with different speeds. It also shows that the clock for your Velodyne is not consistent. All of this explains the poor performance of Cartographer on your data.

What exactly is your capture setup?

I will close all of your open bugs now since your data is broken. Feel free to reopen a new bug report as soon as your timing is great.

closed 2017-10-13 by SirVer

sunmk2006, 2017-10-16:
I will recollect data and analyze it,Thank you for your kind help.

Duke-Allen, 2019-02-28:
@sunmk2006 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?