cartographer-project/cartographer_ros #511: how to tune parameters for 2d slam?

danningwang, 2017-09-26:
Hi there,
I record a corridor data using only front laser scan by topic name /base_laser_front/scan, even though my robot has front and rear laser scans and a topic name for that is /scan_unified. 
The problem is cartographer cannot launch successfully if <remap from="scan" to="base_laser_front/scan" /> in cartographer_jenny.launch. It always gave the error: [FATAL] [1506356539.647252067]: F0925 18:22:19.000000  6851 sensor_bridge.cc:33] Check failed: frame_id[0] != '/' ('/' vs. '/') The frame_id /base_laser_front_link should not start with a /. Then I also remap the frame name by using 
<node pkg="tf" type="tf_remap" name="tf_remapper" output="screen">
  <rosparam param="mappings">
    [{old: /base_laser_front_link, new: laser_front_link}]
  </rosparam>
</node>
in cartographer_sim.launch. And I also remap <remap from="scan" to="front_scan" /> in cartographer_jenny.launch. 
When I publish bag file using this command : rosbag play corr1_frontscan.bag /base_laser_front/scan:=front_scan laser_front_link:=/base_laser_front_link
But cartographer_ros node still cannot launch .  Could you help me to solve this problem? Thank you in advance!
this is my github https://github.com/danningwang/cartographer
a link to my bag file https://files.fm/u/7psc775b

BrannonKing, 2017-09-26:
You shouldn't have any slash characters in your lua file. It's not a problem with launch file but is a problem with the lua file that is specified there.

danningwang, 2017-09-26:
yes, I don't have any slash in my lua file and I tried to avoid slash for topic name and frame name in my launch file using remap. 
Last time cartographer can work using another bag file with scan_unified topic, but it cannot work with the above bag file with base_laser_front/scan topic. So I did the remap process, But it still gave error .

cschuet, 2017-10-06:
@danningwang: We looked into your issue by rebasing your branch on top of master.

- We used #527 to rewrite the frame_ids inside the rosbag rather than remapping it while playback. You remapped /tf's frame ids, but they were also wrong in the scan topic.
- Your bag seems to contain several holes (two minutes w/o data) in various topics. See also image below.
- roswtf reports severals errors regarding competing transform publishers. E.g.

```
ERROR TF multiple authority contention:
 * node [/play_1507304548228064619] publishing transform [base_link] with parent [base_footprint] already published by node [/robot_state_publisher]
 * node [/play_1507304548228064619] publishing transform [base_laser_front_link] with parent [base_link] already published by node [/robot_state_publisher]
 * node [/play_1507304548228064619] publishing transform [base_laser_rear_link] with parent [base_link] already published by node [/robot_state_publisher]
 * node [/base_link_to_base_footprint] publishing transform [base_link] with parent [base_footprint] already published by node [/play_1507304548228064619]
 * node [/laser_rear_to_base] publishing transform [base_laser_rear_link] with parent [base_link] already published by node [/play_1507304548228064619]
 * node [/laser_front_to_base] publishing transform [base_laser_front_link] with parent [base_link] already published by node [/play_1507304548228064619]
 * node [/robot_state_publisher] publishing transform [base_link] with parent [base_footprint] already published by node [/base_link_to_base_footprint]
 * node [/robot_state_publisher] publishing transform [base_laser_front_link] with parent [base_link] already published by node [/laser_front_to_base]
 * node [/robot_state_publisher] publishing transform [base_laser_rear_link] with parent [base_link] already published by node [/laser_rear_to_base]                                          
 * node [/laser_rear_to_base] publishing transform [base_laser_rear_link] with parent [base_link] already published by node [/robot_state_publisher]                                          
 * node [/laser_front_to_base] publishing transform [base_laser_front_link] with parent [base_link] already published by node [/robot_state_publisher]                                        
 * node [/base_link_to_base_footprint] publishing transform [base_link] with parent [base_footprint] already published by node [/robot_state_publisher]                                       
 * node [/robot_state_publisher] publishing transform [base_link] with parent [base_footprint] already published by node [/play_1507304548228064619]                                          
 * node [/play_1507304548228064619] publishing transform [base_link] with parent [base_footprint] already published by node [/base_link_to_base_footprint]                                    
 * node [/robot_state_publisher] publishing transform [base_laser_front_link] with parent [base_link] already published by node [/play_1507304548228064619]                                   
 * node [/robot_state_publisher] publishing transform [base_laser_rear_link] with parent [base_link] already published by node [/play_1507304548228064619]                                    
 * node [/play_1507304548228064619] publishing transform [base_laser_rear_link] with parent [base_link] already published by node [/laser_rear_to_base]
 * node [/play_1507304548228064619] publishing transform [base_laser_front_link] with parent [base_link] already published by node [/laser_front_to_base]
```

[image]

All of these issues are not related to cartographer directly but rather the data acquisition and ros configuration.

closed 2017-10-06 by cschuet