cartographer-project/cartographer_ros #520: Question about the alpha vector

GMahmoud, 2017-10-04:
Hey guys,
I am trying to visualize the submaps of cartographer for  Android device. 
And I have a quick question concerning the signification of the alpha vector in the submap. Does it represent the opacity ?  In the Ogre texture you used PF_BYTE_RGB as parameter and I am using argb32 as codec for my texture. I don't know how to fuse all the submaps to get a global map. I put the alpha value but it get only borders (no clearing of the map).
[image]
[image]

wohe, 2017-10-05:
Our textures only use value (premultiplied alpha) and alpha (representing opacity as you say). If I remember correctly we use an RGB texture with Ogre because RG texture was not possible or caused issues with Ogre. If you look at `ogre_submap.cc`, you see that we store the value in R and the alpha in G, and read it from there in the shader `submap.frag`. Does that answer your question?

Mofef, 2017-10-05:
We had once the same issue/confusion.
If you want resemble the compositing of submaps like in rviz using normal alpha-blending you first need to transform the given "alpha" and "value" values like so:
(assuming everything is in float normalized to 0. ...1.)
*    `initialize the output to the same shape as the input (i.e. width * height * 2 channels)`
*    `result[alpha] = input[alpha] + input[value]`
*    `result[value] = 1.0 where input[value]>input[alpha] otherwise 0.0`

Afterwards you can just apply the over operator as in https://en.wikipedia.org/wiki/Alpha_compositing.

wohe, 2017-10-05:
I strongly suggest to use premultiplied alpha for best visual quality. Any reason not to simply change the blend function? If you want to use RGBA you just copy (value, value, value, alpha) into the texture.

GMahmoud, 2017-10-05:
Thank you guys (@wohe and @Mofef)  for your answers. 
I checked my texture with your information and I got better result.
So glad.

Cheers,
MG.
[image]

Here's a new screenshot. I think I can close this issue ;)

closed 2017-10-05 by GMahmoud