cartographer-project/cartographer_ros #525: Offline mode not publishing to /clock

fjam100, 2017-10-05:
I was successfully able to run online mapping, but when I try to do it offline, nothing gets published to /clock and I get this error:
> terminate called after throwing an instance of 'std::runtime_error'
>   what():  Time is out of dual 32-bit range
Here's the launchfile I'm running:

launch>
  <param name="/use_sim_time" value="true" />

  <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />

  <node name="cartographer_offline_node" pkg="cartographer_ros"
      type="cartographer_offline_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename ranger.lua
          -bag_filenames $(arg bag_filenames)"
      output="screen">
    <remap from="scan" to="$(arg slam_lidar)" />
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />
</launch>

Here's the github link: [https://github.com/fjam100/cartographer-test](https://github.com/fjam100/cartographer-test).  Unfortunately, I can't share the bag file since it's too big.

SirVer, 2017-10-06:
The launch file looks good - I think it is likely an error in any other file or the bag. The github link does not seem to contain your changed files, so has no value for resolving this.

Please provide all files you are using or changing, including urdf, config, lua and launch files. Please also share a bag file on drive, dropbox.

SirVer, 2017-10-12:
Closing as this is not actionable in the current state for us. If you require further assistance, please provide the requested information an we'll reopen.

closed 2017-10-12 by SirVer