cartographer-project/cartographer_ros #528: Check failed: reader.ReadProto(&pose_graph_proto)

ghost069, 2017-10-07:
when I was running assets_writer_backpack_3d.launch,trying to construct point clouds from the results of the /finish_trajectory service, I encountered the following error:

> 
>  F1007 18:14:40.694540 16314 assets_writer_main.cc:153] Check failed: reader.ReadProto(&pose_graph_proto) 
> *** Check failure stack trace: ***
>     @     0x7faa468e2daa  (unknown)
>     @     0x7faa468e2ce4  (unknown)
>     @     0x7faa468e26e6  (unknown)
>     @     0x7faa468e5687  (unknown)
>     @           0x446674  cartographer_ros::(anonymous namespace)::Run()
>     @           0x44584c  main
>     @     0x7faa436caf45  (unknown)
>     @           0x446207  (unknown)
>     @              (nil)  (unknown)
> ================================================================================REQUIRED process [cartographer_assets_writer-2] has died!
> process has died [pid 16314, exit code -6, cmd /home/anthony/catkin_ws/install_isolated/lib/cartographer_ros/cartographer_assets_writer -configuration_directory /home/anthony/catkin_ws/install_isolated/share/cartographer_ros/configuration_files -configuration_basename assets_writer_backpack_3d.lua -urdf_filename /home/anthony/catkin_ws/install_isolated/share/cartographer_ros/urdf/backpack_3d.urdf -bag_filenames /home/anthony/Downloads/b3-2016-02-02-13-32-01.bag -pose_graph_filename /home/anthony/Downloads/laser_test1.bag.pdstream __name:=cartographer_assets_writer __log:=/home/anthony/.ros/log/53b5af06-ab48-11e7-8493-7ce9d31dd245/cartographer_assets_writer-2.log].
> log file: /home/anthony/.ros/log/53b5af06-ab48-11e7-8493-7ce9d31dd245/cartographer_assets_writer-2*.log
> Initiating shutdown!
> ================================================================================
> [cartographer_assets_writer-2] killing on exit
> 

I have no idea what's going on.The bag file was downloaded from[ the cartographer ROS documentation](https://google-cartographer-ros.readthedocs.io/en/latest/data.html),b3-2016-02-02-13-32-01.bag.And the pose graph file  was gotten by calling rosservice,just like the doc said.Although,there's some generated pictures in  ~/.ros,like xray_xy_all_color.png，xray_xy_all.png.But there's no ply file or other generated things.Here's the xray_xy_all_color.png
[image]

So,why did assets_writer_main.cc break down and how can I fix it to get a ply file?

ghost069, 2017-10-08:
I commented out CHECK(reader.ReadProto(&pose_graph_proto)); 
Run launch file.
I got:

 Check failed: pose_graph_proto.trajectory_size() == bag_filenames.size() (0 vs. 1) Pose graphs contains 0 trajectories while 1 bags were provided. This tool requires one bag for each trajectory in the same order as the correponding trajectories in the pose graph proto.

So that's why check failed.The pdstream file contains 0 trajectories.But how can I fix it?I did exactly what the documentation says.(At least I think so...)

ojura, 2017-10-08:
Commenting out the `CHECK` is definitely wrong. The first proto in the stream is the sparse poss graph proto and you have to read it, which is done inside the `CHECK`. 

To maintainers: the code inside the `CHECK` has side effects. Perhaps it would be more readable to not have code with side effects within asserts, e.g.

`const bool proto_read_successful = reader.ReadProto(&pose_graph_proto)); CHECK(proto_read_successful);`

SirVer, 2017-10-09:
> To maintainers: the code inside the CHECK has side effects. Perhaps it would be more readable to not have code with side effects within asserts, e.g.

that is the google style of doing things - it is just a stylistic thing and doesn't hurt. Actually it increases readability of the error, since the content of the CHECK is echoed on the console. Having an extra variable floating around is unpretty too.

@ghost069 Hard to know what went wrong. If you ran the exact command lines as [provided in the demo documentation](https://google-cartographer-ros.readthedocs.io/en/latest/#running-the-demos) it should just work. Make sure that you pulled both cartographer and cartographer_ros to the latest versions - they must always be synced to be compatible.

SirVer, 2017-10-12:
@ghost069 I cannot repro this without knowing the exact command lines and versions of cartographer_ros and cartographer you ran.

Closing for now until repro case is provided, will reopen immediately.

closed 2017-10-12 by SirVer