cartographer-project/cartographer_ros #529: Build a rosbag_wtf

SirVer, 2017-10-09:
Many of the tuning issues we receive in Cartographer are easy to explain

Some examples for inspiration what to check for with example issues where this showed:

- [ ] timing (#511, #317, googlecartographer/cartographer#377, googlecartographer/cartographer#193), most of these can be tracked with a histogram + min/max of delta timestamps for each sensor.
- [ ] incomplete IMU data (#424). Especially wrong units (G instead of m/s^2 or gravity being removed)
- [ ] fastest sensor is still very slow (#372)
- [ ] frames fighting: TF published from Cartographer are also in the bag (#356).
- [ ] scans are never changing shape (#365)
- [ ] ROS data type validation: For example invalid laser messages (#354, #270 ), invalid IMU messages (#273, #268 ).

ojura, 2017-10-09:
IIRC, Cartographer has a zero velocity assumption at the beginning of the bag, so checking for that might also be useful.

MichaelGrupp, 2020-10-26:
Should be done now in `cartographer_rosbag_validate`.

closed 2020-10-26 by MichaelGrupp