cartographer-project/cartographer_ros #531: the poor effect of repeated closed loop

sunmk2006, 2017-10-10:
hello, I am tuning 3d cartographer for my vel-16 data,but the effect of repeated closed loop is poor，I can't solve it,can you help me?
the configuration files , launch file, urdf file and result png files is [here ](https://github.com/sunmk2006/cartographer_ros/blob/master/slam_3d_loop.zip),and the bag file is [here](https://github.com/sunmk2006/cartographer_ros/blob/master/slam_3d_loop.zip).

[image]

Mofef, 2017-10-10:
On first glance SPARSE_POSE_GRAPH.constraint_builder.min_score = 0.7 seems very high to me try something like ~0.55

SirVer, 2017-10-11:
@sunmk2006 We mentioned some problems with your data in #509 already. Could you first comment on the issues before we further investigate? Most surprising to us is that you have the exact same number of IMU messages than VLP16 packets which seem nonphysical.

sunmk2006, 2017-10-11:
Sorry, I didn't respond to your comments in time. 

> the exact same number of IMU messages than VLP16 packets 

This phenomenon stems from a synchronous approach，locate the IMU data based on the received laser timestamp，Because my IMU device and laser device do not work at the same time, the data of the two are not collected simultaneously.The recently uploaded bag data file does not pay attention to the synchronization of the two, and directly to the cartographer processing.Is there any problem with this method?

SirVer, 2017-10-11:
> Is there any problem with this method?

Yes, it is. The synchronization is actually sub-optimal - it means that you integrate your IMU data for longer than necessary, leading to larger errors. Why are you doing this?

We will investigate your other bag in #509, I reopened the issue. Maybe a solution for this one comes out of that as well.

sunmk2006, 2017-10-11:
@Mofef  I have changed the parameter , but the problem still exists

sunmk2006, 2017-10-11:
> Why are you doing this?
Still based on data synchronization considerations，I think in advance to do data synchronization, the effect will be better.

SirVer, 2017-10-11:
> Still based on data synchronization considerations，I think in advance to do data synchronization, the effect will be better.

I do not understand that reasoning. Why should synchronized data collection better? More data will always be better than subsampling.

> I also did UDP processing, but the effect is not very good.

Not sure what you mean with this.

sunmk2006, 2017-10-12:
> More data will always be better than subsampling.
If the timestamp of the first IMU data in the bag file is 1000.000001 (second), the timestamp of the first laser data is 1020.000001 (second). Can the cartographer automatically synchronize the two data? 

> I also did UDP processing, but the effect is not very good.
Later, I'll upload the bag file in UDP form to the Internet

SirVer, 2017-10-13:
> If the timestamp of the first IMU data in the bag file is 1000.000001 (second), the timestamp of the first laser data is 1020.000001 (second). Can the cartographer automatically synchronize the two data?

The offline node does this automatically. The online node could too, but ROS cannot deal with this (as it tries to transform the data and TF buffers are not that long). It seems unphysical that your sensor offset is that big and constant - how did that happen?

Closing this following the analysis in https://github.com/googlecartographer/cartographer_ros/issues/509#issuecomment-336484002.

ojura, 2017-10-14:
If there's a fixed offset between the two sensors, just account for it in their timestamps. No need to change the sample rate.

SirVer, 2017-10-16:
@ojura It isn't a fixed offset, see our analysis in the linked comment.

sunmk2006, 2017-10-19:
[image]

sunmk2006, 2017-10-25:
the new configuration files and png files is [here](https://github.com/sunmk2006/cartographer_ros/blob/master/configuration_files_frame.zip)
[image]

SirVer, 2017-10-25:
@sunmk2006 Could you host your bags somewhere else? Downloading your bag from baidu takes > 10 hours here in Germany and the connection terminates regularly. Google Drive is better replicated and would be preferable.

sunmk2006, 2017-10-25:
OK， but I can't use Google drive, I will reduce the bag file or try another cloud disk.

sunmk2006, 2017-10-25:
@SirVer hello,a new link is [here](http://www.babel.cc/share.do?s=1114663790085936),You can test the download speed over there, and if the speed is still slow, I'll look for other cloud disks

SirVer, 2017-10-25:
Does the new link requires an account? I have trouble with the chinese.

sunmk2006, 2017-10-25:
No account required，please open the link, Put the mouse on the line of test_03.bag.zip，[Click the second buttons that appear](https://github.com/sunmk2006/cartographer_ros/blob/master/20171025191607.jpg)

sunmk2006, 2017-10-25:
https://github.com/sunmk2006/cartographer_ros/blob/master/20171025191607.jpg

SirVer, 2017-10-25:
that link opens a different looking page for me: 
[image]

sunmk2006, 2017-10-25:
ok，i have an account 18037109891 and the passwd is 632851

sunmk2006, 2017-10-26:
I added two more links. See if you can use them
[link1](https://v2.fangcloud.com/share/a51cdc038358d5f61361731996) 
[link2](http://pan.baidu.com/s/1bp8Qnin)

gaschler, 2017-10-26:
Timing of sensor data now seems fine, we had a look into the graphs. Some gaps in laser data are still present but not in the first seconds, so they don't explain why tracking fails.

I had a look at your bag file with
`rqt_bag ~/Downloads/test_03.bag`
and observed that IMU acceleration in z-axis is positive in this bag.
For this, the urdf setting 
`<child link="imu" />  <origin xyz="0 0 0" rpy="3.1416 0 0" />`
cannot be correct. Gravity should be pointing down.

Perhaps you can share photos of how the sensors are attached to your robot?

sunmk2006, 2017-10-26:
sorry,  I should update the urdf file,the imu to base_link rpy should "0 0 3.1415926"

>   <joint name="imu_link_joint" type="fixed">
    <parent link="base_link" />
    <child link="imu" />
    <origin xyz="0 0 0" rpy="0 0 3.1415926" />
  </joint>

gaschler, 2017-10-26:
I also tried this, quality was low again. Are you entirely sure the timing of the imu matches the laser scans?

Investigating this further will be hard without 
- photos of the setup to ensure the urdf transforms are correct
- a fork that we can simply run and test without any copy-paste.

sunmk2006, 2017-10-27:
Okay, I'll upload a picture of the device later，below is a URDF model, the relative position is correspond with actual position of equipment , the problem is that in the ROS coordinates  x-->forward,  y-->left, z-->up, but x-->left,but my own coordinates x-->right,y-->forward, z-->up, because the IMU around the Z axis rotation to 180 degrees (yaw = 3.1415926 rad) ,the coordinates of IMU is x-->left, y-->backward, z-->up,How can I tell the cartographer system about these changes?

[image]

SirVer, 2017-10-30:
> How can I tell the cartographer system so many changes about the coordinates?

Through the URDF.

> In addition, is it possible that those configuration parameters (lua files)causes the problem?

Not the Lua files. But if your URDF is wrong your IMU and Laser are constantly disagreeing, so that would be definitively lead to the fragility we observe.

gaschler, 2017-10-30:
As I said, I observed that IMU acceleration in z-axis is positive in your bag.
In my eyes, this does not match the upwards IMU z-axis in the first picture.
Before any fine tuning, you need to correct the URDF first.

closed 2017-10-30 by gaschler

sunmk2006, 2017-11-01:
my imu device model is xsens MTI-G-700 ,I use the official equipment software to do an experiment, when the device is placed upright, Z acceleration is positive ,when placed upside down, Z acceleration is negative

gaschler, 2017-11-02:
If your IMU reads (negative) -10m/s^2, your URDF settings should be that the z-axis (blue) of IMU points upward.
If it reads positive, z-axis should be configured downward.

For the other axes, we would need more information.