cartographer-project/cartographer_ros #533: Nearest frontier based exploration of unknow environment with cartographer shows incorrect map

lxsang, 2017-10-10:
Hello, i perform a simulation of  nearest frontiers based exploration using cartographer with gazebo + turtlebot. I've used the latest version of cartographer_ros + cartographer_turlebot (as shown on the cartographer document website), i used the default configuration, and used the demo_lidar_2d.lauch as the base of my launch file (adding navigation stack, frontiers detection and allocation, etc.). Cartographer built a very strange map as shown in  [this video](https://youtu.be/ioBIqXGjpkU).  I don't know if i'm missing some thing, could you please help me? The bag file can be found [here](https://github.com/lxsang/online_stuffs/blob/master/2017-10-10-17-44-17.bag.zip?raw=true).

With the same launch file, by replacing cartographer with gmapping, i could correctly explore the environment and build the map as show in [this video](https://youtu.be/so0gh9UxZzw). But i do really want to use cartographer for an evaluation.

Thanks

wohe, 2017-10-11:
Hi, From the video it looks like Cartographer is finding too many outliers when doing loop closure. The configuration you use was tuned for real data and not simulated data. This can make a huge difference. You can have a look at the [tuning guide](https://google-cartographer-ros.readthedocs.io/en/latest/tuning.html) but it seems to not contain much regarding loop closure. Assigning to @cschuet to investigate further.

lxsang, 2017-10-12:
Hello, thanks for turning guide tip, i tried to adjust the parameters but the problem still persist. I'll let you know if i find something else.

SirVer, 2017-10-12:
@lxsang You can also turn on visualization of constraints in rViz - they are published as markers. That might give you  a clue where wrong loop closures are found. I would start trying to turn [`SPARSE_POSE_GRAPH.constraint_builder.min_score`](https://github.com/googlecartographer/cartographer_turtlebot/blob/master/cartographer_turtlebot/configuration_files/turtlebot_urg_lidar_2d.lua#L49) upwards towards 1., maybe 0.8

Let us know if you cannot make progress and we can have a look.

lxsang, 2017-10-12:
@SirVer thanks for the hint, the constraints graph is really helpful. I tried to tune the SPARSE_POSE_GRAPH.constraint_builder.min_score value beginning with 0.75 upto 0.95. Here goes some results (I stopped the simulation where the problem occurred )
### min_score=0.75
[image]
### min_score=0.8
[image]
### min_score=0.85
[image]
### min_score=0.925
[image]

It looks like that, for this simulation, value greater than **0.9** gives better result, but if i understand correctly, it also means that we are losing  the beautiful loop closure correction of Cartographer, as marked in the last figure (min_score=0.925)

SirVer, 2017-10-16:
I looked into your data a bit now and here are my observations:

- Your bag contains funky TF (from a previous run with previous timestamps maybe?) and the turtlebot does not come with an easy URDF. I had to do some magic to get only the transforms that I am interested in.
- I had to increase the min score and decrease the loop closing search space. I was not able to get rid of the wrong loop closure constraint otherwise - the data is just to perfect and the scan matches too good. This is unfortunate, but a 7 m search space for your small loops is quite large anyways.
- Your robot turns very fast, so I dabbled with increasing the num subdivisions, this seemed to have improved quality, but also costs performance.
- The bag you posted shows the problem, but ends immediately after that. It would be fun to see the full bag and the corresponding results.

You can find the changes I did in [this branch](https://github.com/googlecartographer/cartographer_turtlebot/compare/master...SirVer:issue/533?expand=1). The map I got from this is attached.

[image]

closed 2017-10-16 by SirVer

lxsang, 2017-10-16:
Thanks for the help, i really appreciate it.
> Your bag contains funky TF (from a previous run with previous timestamps maybe?) and the turtlebot does not come with an easy URDF

I saw these TF errors, they weren’t shown when i performed the simulation, but when i replayed the recorded bag file, i got these errors. Does cartographer publish something to the TF anyway?
And sorry for the missing URDF, i just add it [here](https://raw.githubusercontent.com/lxsang/online_stuffs/master/turtlebot.urdf) in case you need it.

>Your robot turns very fast

+1. I didn't notice it, it's something i can fix.

> The bag you posted shows the problem, but ends immediately after that. It would be fun to see the full bag and the corresponding results.

I've recorded a new bag file, so if you have time to play with it, you can [find it here](https://drive.google.com/file/d/0Bz6hIak80E5wNk9nZVVjdVl0T2c/view?usp=sharing) (using the same configuration). I stopped the simulation because it's an autonomous exploration, so the robot strongly depends on the map that it built. If the map is incorrect, the robot may have a conflict between it trajectory planification (global planner) and the groundtruth (local planner). Therefore, some time the robot oscillates or is blocked, and that will, IMHO, accumulate the errors.
Thanks again

SirVer, 2017-10-17:
> And sorry for the missing URDF, i just add it here in case you need it.

Thanks. Only using it makes everything much smoother. I updated my branch to use your urdf.

The new bag is funny because the robot drives through a wall at one point :). I further restricted loop closure to deal with the large featureless room a bit better. The system behaves okay, but not perfect, see screenshot.

[image]

lxsang, 2017-10-17:
Thanks, it looks much better.

> The new bag is funny because the robot drives through a wall at one point :)

This happens sometime in simulation (Gazebo), when there is a conflict and the robot tries to  reach its global goal through the wall (the local planner is supposed to avoid this behaviour but sometime it doesn't :) ). The physical system in Gazebo is not perfect though.

lxsang, 2017-10-18:
One last thing, it looks like cartographer is terrain-sensitive, and needs a tuning of parameters for different kind of terrains/datasets (for example the linear search window) . So it is not obvious to have an optimal set of parameters for all kind of terrains. Is it correct ? This may have a negative effect on autonomous exploration where the terrain is unknown, therefore, it is not possible to find an optimal set of parameters for that terrain at the exploration time.
thanks

lxsang, 2017-10-18:
Here is the final result with your tuning:

[image]

I'm now working on this terrain, and obviously, some configurations need new values
[image]

SirVer, 2017-10-19:
> One last thing, it looks like cartographer is terrain-sensitive, and needs a tuning of parameters for different kind of terrains/datasets (for example the linear search window) 

This is the only parameter in Cartographer that can be sensitive to the terrain. All others are tuned for the set of sensors. We have ideas how to make this parameter not depend on the environment - basically by tying it to the distance to the last loop closure (a bit more sophisticated than that, but you get the idea). Alas we had no time to investigate this yet.

I also believe that you have a harder time in simulation than you would have in reality because your environment is so perfectly self similar everywhere, making for easy wrong loop closures.

joyde3382, 2019-04-08:
@lxsang Any chance you could share the source code? Right now i'm able to generate a live map with cartographer and would like to use it for autonomous navigation/exploration :)