cartographer-project/cartographer_ros #534: test Cartographer using VLP-16

Vincent1923, 2017-10-12:
Hello!
I have a VLP-16 - Velodyne LiDAR (3d LiDAR) that publish a pointCloud2 /velodyne_points msg and a IMU that publish a /imu. I tested cartographer ROS using the VLP-16 and IMU. I have tried with backpack_2d.launch and backpack_3d.launch but I got no results, even modifying configuration files.

The configuration files , launch file and urdf file is [here](https://gist.github.com/Vincent1923/9b4dd8bbfd9301dd408e89f6e21cead7)

and the bag file is [here](https://pan.baidu.com/s/1hrDiI1Y)

SirVer, 2017-10-12:
See #259 for an example of a user that ran with exactly your setup. You very likely require a URDF with the correct frame names, a launch file and a Lua configuration based on the backpack_3d.(lua|launch) example.

[These are the configuration files](https://github.com/googlecartographer/cartographer_ros/compare/master...ozgudum:master) that worked for the user. The issue is a bit older, so you should start from our files and modify them instead of just trying to copy these. 

Closing as this is a configuration issue with ROS, not really about Cartographer. Let us know if you require further help with getting everything up and running. If you do, please post console output and which commands you are running exactly.

closed 2017-10-12 by SirVer

Vincent1923, 2017-10-13:
Thank you for your answer. @SirVer 

I modify the configuration file vlp16-imu.lua based on the backpack_3d.lua. When i used the configuration files above, I still got no results. This is the command i am using
  roslaunch cartographer_ros demo_vlp16-imu.launch bag_filename:=${HOME}/Downloads/2017-09-19-22-43-47.bag

I notice on the terminal , it always print out below info. 
[ WARN] [1507877651.988053206, 1505832303.729148703]: W1013 14:54:11.000000 18329 tf_bridge.cc:51] "map" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1507877652.091040837, 1505832303.936919835]: W1013 14:54:12.000000 18329 tf_bridge.cc:51] "map" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1507877652.192541342, 1505832304.139026499]: W1013 14:54:12.000000 18329 tf_bridge.cc:51] "map" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1507877652.296966175, 1505832304.348513401]: W1013 14:54:12.000000 18329 tf_bridge.cc:51] "map" passed to lookupTransform argument source_frame does not exist. 
[ WARN] [1507877652.398519789, 1505832304.550059293]: W1013 14:54:12.000000 18329 tf_bridge.cc:51] "map" passed to lookupTransform argument source_frame does not exist.

Are my configuration files still wrong? [These are the files i am using.](https://gist.github.com/Vincent1923/9b4dd8bbfd9301dd408e89f6e21cead7)

Thank you!

SirVer, 2017-10-13:
This line is still wrong:

https://gist.github.com/Vincent1923/9b4dd8bbfd9301dd408e89f6e21cead7#file-vlp16-imu-lua-L40

Try putting there 1 or 2 - the velodyne driver always outputs full revolutions instead of individual UDP packets worth of points. Did this fix it?

Vincent1923, 2017-10-13:
Putting there 1 or 2 still can not fix this problem.
/home/vincent/Pictures/Screenshot from 2017-10-13 17-29-03.png

Vincent1923, 2017-10-13:
Putting there 1 or 2 still can not fix this problem.

SirVer, 2017-10-13:
Running our new validator tool on your bag gives:

~~~
$ rosrun cartographer_ros cartographer_rosbag_validate -bag_filename ~/Downloads/2017-09-19-22-43-47.bag
I1013 14:45:53.088878 118386 rosbag_validate_main.cc:111] Time delta histogram for consecutive messages on topic "/imu" (frame_id: "map"):
Count: 3046  Min: 0.015220  Max: 0.117025  Mean: 0.044480
[0.015220, 0.025401)                            Count: 1 (0.032830%)    Total: 1 (0.032830%)
[0.025401, 0.035581)                            Count: 1 (0.032830%)    Total: 2 (0.065660%)
[0.035581, 0.045762)     ###################    Count: 2923 (95.961914%)        Total: 2925 (96.027580%)
[0.045762, 0.055942)                            Count: 20 (0.656599%)   Total: 2945 (96.684174%)
[0.055942, 0.066123)                            Count: 0 (0.000000%)    Total: 2945 (96.684174%)
[0.066123, 0.076303)                            Count: 0 (0.000000%)    Total: 2945 (96.684174%)
[0.076303, 0.086483)                            Count: 75 (2.462245%)   Total: 3020 (99.146423%)
[0.086483, 0.096664)                            Count: 25 (0.820749%)   Total: 3045 (99.967171%)
[0.096664, 0.106844)                            Count: 0 (0.000000%)    Total: 3045 (99.967171%)
[0.106844, 0.117025]                            Count: 1 (0.032830%)    Total: 3046 (100.000000%)
I1013 14:45:53.089287 118386 rosbag_validate_main.cc:111] Time delta histogram for consecutive messages on topic "/velodyne_points" (frame_id: "velodyne"):
Count: 1343  Min: 0.099314  Max: 0.102361  Mean: 0.100871
[0.099314, 0.099619)                            Count: 1 (0.074460%)    Total: 1 (0.074460%)
[0.099619, 0.099923)                            Count: 0 (0.000000%)    Total: 1 (0.074460%)
[0.099923, 0.100228)                            Count: 1 (0.074460%)    Total: 2 (0.148920%)
[0.100228, 0.100533)                            Count: 1 (0.074460%)    Total: 3 (0.223380%)
[0.100533, 0.100837)                 #######    Count: 441 (32.836933%) Total: 444 (33.060314%)
[0.100837, 0.101142)           #############    Count: 891 (66.344009%) Total: 1335 (99.404320%)
[0.101142, 0.101447)                            Count: 5 (0.372301%)    Total: 1340 (99.776619%)
[0.101447, 0.101752)                            Count: 2 (0.148920%)    Total: 1342 (99.925537%)
[0.101752, 0.102056)                            Count: 0 (0.000000%)    Total: 1342 (99.925537%)
[0.102056, 0.102361]                            Count: 1 (0.074460%)    Total: 1343 (100.000000%)
~~~

This shows that your IMU messages are published in the map frame: "/imu" (frame_id: "map"). That should probably be "imu_link".

Vincent1923, 2017-10-13:
Thank you! @SirVer 
But I still do not understand what it means. Is that means I have to change the 'imu_link' in the file vlp16-imu.urdf? Or is that the frame id name "map" is wrong in my bag?

SirVer, 2017-10-16:
> Or is that the frame id name "map" is wrong in my bag?

This is wrong, `map` is by convention an unchanging frame that all other frames are relative too. You should read [REP 105](http://www.ros.org/reps/rep-0105.html) and probably do a ROS tutorial or two.

Vincent1923, 2017-10-16:
OK. I will read this and do some ROS tutorials later. If I still have problems, can I ask you again.
Thank you very much! @SirVer

Vincent1923, 2017-10-31:
I still don't understand the problem with the frame id. Is the frame id wrong when I record the bag file? @SirVer

SirVer, 2017-11-02:
Yes, that is the case.

Vincent1923, 2017-11-02:
Thank you very much! I will try to change the  frame id and have a try again.

GeoBIMpro, 2018-01-06:
Hi @Vincent1923, could you solve your problem?
I am not able to tune proper parameters for my setup too.
Below are the links for bag file and urdf:
Bag file: [https://drive.google.com/file/d/19F1FkYd47ycefaxcSpYByEo8Q8p8jpx1/view?usp=sharing](bag file)
URDF: [https://gist.github.com/GeoBIMpro/ca1f0d5a29ad8a983ef13dd022c61746](urdf)

gaschler, 2018-01-08:
@GeoBIMpro Please open a separate issue, thanks.

Rolanding, 2018-08-19:
@Vincent1923 have you sloved the problem?

Rolanding, 2018-08-19:
@Vincent1923 I have the same problem, and I changed my frame_id and topic ，but still can't solve it. 
Look forward to your reply. Thanks  a lot.

Duke-Allen, 2019-03-02:
@Vincent1923 @gaschler @SirVer 
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?