cartographer-project/cartographer_ros #540: Test cartographer with 3D LIDAR while no data

timoonboru, 2017-10-16:
Hello!
I have a 3d LiDAR bag containing a pointCloud2 /fullscan msg and a IMU /imu/data msg. 
I tested cartographer ROS with this bag.

The configuration files , launch file and urdf file is here
[configuration.zip](https://github.com/googlecartographer/cartographer_ros/files/1385935/configuration.zip)

Note that in .launch,  
I have been remapped:
remap from="points2" to="/fullscan" /
remap from="imu" to="/imu/data" /

In .urdf , 
The frame id of "/fullscan" is "rslidar"
The frame id of  "/imu/data" is "imu_link"
Hence in XXX.urdf I changed in correct position.

In .lua,
I have changed that 
num_point_clouds = 1.

The ROS report an error with "Queue waiting for data:(0, points2)" when I try to test catographer with bag， and in RVIZ there is no data.
Could you help me or give some suggests? Thank you!

SirVer, 2017-10-16:
The error message indicates that your laser data never makes it to Cartographer, i.e. either is dropped because it cannot be transformed (which would indicate a wrong URDF) or is just not on the topics that you think.

I took a quick look over the changes you made to the configuration files and they seem fine and correct for the configuration of your sensor as described in your post. My guess is that what ends up in your bag
 looks different than what you think - or that I missed an error in your config files.
 
Without your bag it is impossible to help you here. Closing until this data is provided.

timoonboru, 2017-10-17:
Thank you for your answer. @SirVer 
I have been upload the test bag in the issue page.
It's worth noting that  the link shoud be copy to anther window to open it.
Waiting for your news.

timoonboru, 2017-10-24:
Thank you for your answer. @SirVer
I have solved this problem by editing the LIDAR driver.

In practical use I have another problem.
When I SLAM outdoors with cartographer, it seems can‘t generate the trajectory in real time as the map become bigger and bigger. In my opinion it is caused by the increase numbers of submaps , whch occupy more and more memory and cost more and more calculation amount with global optimization.
Regardless of the hardware causes，I attemp to retain the latest numbers submaps（eg：the latest 100 submaps ） to ensure the real-time with cartographer.
Is the method practicable? Could you give me some better suggests? @SirVer
Thank you again！

SirVer, 2017-10-24:
> Regardless of the hardware causes，I attemp to retain the latest numbers submaps（eg：the latest 100 submaps ） to ensure the real-time with cartographer.

You can do that with some coding using a trimmer. Look at how pure localization is implemented in cartographer_ros. In general I would advise to tune your system for real time though to get the best possible map.

Rolanding, 2018-08-20:
@timoonboru I really want to know how did you solve this problem. I meet the same thing, and I'm quite sure my  configuration files are right, especially the frame_id and topics.
Looking forward to your reply, thanks a lot.

daiencan, 2018-09-08:
I also meet the same problem and I want to know how did you solve this problem.
Looking forward to your reply, thanks a lot.

CherryChenw, 2019-07-23:
> Thank you for your answer. @SirVer
> I have solved this problem by editing the LIDAR driver.
> 
> In practical use I have another problem.
> When I SLAM outdoors with cartographer, it seems can‘t generate the trajectory in real time as the map become bigger and bigger. In my opinion it is caused by the increase numbers of submaps , whch occupy more and more memory and cost more and more calculation amount with global optimization.
> Regardless of the hardware causes，I attemp to retain the latest numbers submaps（eg：the latest 100 submaps ） to ensure the real-time with cartographer.
> Is this method practicable? Could you give me some better suggests? @SirVer
> Thank you again！

I also meet the same problem and  how do you fix the driver?

Vineet-Pandey, 2021-03-29:
I am running the cartographer using a 3D Lidar and an IMU, and I am facing this issue. Was this problem solved? because when I look at my Lidar data, every data has a unique time stamp and I am publishing a static tf on the ```/tf_static``` topic. Am i missing anything?

johuber, 2021-03-30:
> I am running the cartographer using a 3D Lidar and an IMU, and I am facing this issue. Was this problem solved? because when I look at my Lidar data, every data has a unique time stamp and I am publishing a static tf on the `/tf_static` topic. Am i missing something?
> I am receiving the following response:
> `[ WARN] [1617046190.073522848]: W0329 15:29:50.000000 14731 ordered_multi_queue.cc:155] Queue waiting for data: (0, points2)`

this means cartographer is not receiving any point cloud messages. Are you sure that your cartographer node subscribes to the correct topic? (you can check with `rosbag info`)

Vineet-Pandey, 2021-04-07:
> this means cartographer is not receiving any point cloud messages. Are you sure that your cartographer node subscribes to the correct topic? (you can check with `rosbag info`)

My apologies for a delayed response. I have remapped both IMU and the 3D lidar topics in the launch file so i think node is subscribing to the correct topics.  Upon checking the ```rosbag info``` i received the following output:
```path:        2021-04-07-12-09-25.bag
version:     2.0
duration:    2:35s (155s)
start:       Apr 07 2021 12:09:25.15 (1617811765.15)
end:         Apr 07 2021 12:12:00.82 (1617811920.82)
size:        278.9 MB
messages:    52218
compression: none [312/312 chunks]
types:       cartographer_ros_msgs/SubmapList  [73b1e412208f0787050395996f6143db]
             diagnostic_msgs/DiagnosticArray   [60810da900de1dd6ddd437c3503511da]
             diagnostic_msgs/DiagnosticStatus  [d0ce08bc6e5ba34c7754f563a9cabaf1]
             nav_msgs/Odometry                 [cd5e73d190d741a2f92e81eda573aca7]
             ros_mscl/filter_heading_msg       [b90b446055a9cca548055d2a70764344]
             ros_mscl/filter_heading_state_msg [ca75e8c156e8551037af1d0f6d896ef2]
             ros_mscl/filter_status_msg        [44b6cdd5e336c9a880e785b76ea6be97]
             ros_mscl/status_msg               [b19b8bec6f96c609664c27dbc51ddf60]
             rosgraph_msgs/Log                 [acffd30cd6b6de30f120938c17c593fb]
             sensor_msgs/Imu                   [6a62c6daae103f4ff57a132d6f95cec2]
             sensor_msgs/MagneticField         [2f3b0b43eed0c9501de0fa3ff89a45aa]
             sensor_msgs/PointCloud2           [1158d486dd51d683ce2f1be655c3c181]
             tf2_msgs/TFMessage                [94810edda583a504dfda3829e70d7eec]
             visualization_msgs/MarkerArray    [d155b9ce5188fbaf89745847fd5882d7]
topics:      /constraint_list                311 msgs    : visualization_msgs/MarkerArray   
             /diagnostics                     85 msgs    : diagnostic_msgs/DiagnosticArray  
             /diagnostics_agg                155 msgs    : diagnostic_msgs/DiagnosticArray  
             /diagnostics_toplevel_state     155 msgs    : diagnostic_msgs/DiagnosticStatus 
             /gx5/device/status              163 msgs    : ros_mscl/status_msg              
             /gx5/imu/data                 15508 msgs    : sensor_msgs/Imu                  
             /gx5/mag                      15511 msgs    : sensor_msgs/MagneticField        
             /gx5/nav/filtered_imu/data     1553 msgs    : sensor_msgs/Imu                  
             /gx5/nav/heading               1554 msgs    : ros_mscl/filter_heading_msg      
             /gx5/nav/heading_state         1554 msgs    : ros_mscl/filter_heading_state_msg
             /gx5/nav/odom                  1554 msgs    : nav_msgs/Odometry                
             /gx5/nav/status                1554 msgs    : ros_mscl/filter_status_msg       
             /landmark_poses_list           5183 msgs    : visualization_msgs/MarkerArray   
             /livox/lidar                   1555 msgs    : sensor_msgs/PointCloud2          
             /rosout                          69 msgs    : rosgraph_msgs/Log                 (4 connections)
             /rosout_agg                      52 msgs    : rosgraph_msgs/Log                
             /submap_list                    518 msgs    : cartographer_ros_msgs/SubmapList 
             /tf_static                        1 msg     : tf2_msgs/TFMessage               
             /trajectory_node_list          5183 msgs    : visualization_msgs/MarkerArray 
```
However, i received the following output upon validating the rosbag using the cartographer library:

``` E0407 12:23:16.955684  6302 rosbag_validate_main.cc:351] frame_id "livox_frame" on topic /livox/lidar has serialization time 1617811765.358983125 but sensor time 763.099137830 differing by -1.61781e+09 s.
W0407 12:23:16.955950  6302 rosbag_validate_main.cc:103] frame_id sensor_wgs84 time 1617811765448710339: IMU linear acceleration is 0.0230335 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 0000.020284 -0.00989991 0-0.0045929
E0407 12:23:16.955976  6302 rosbag_validate_main.cc:316] Sensor with frame_id "sensor_wgs84" jumps backwards in time, i.e. timestamps are not strictly increasing. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
E0407 12:23:16.955988  6302 rosbag_validate_main.cc:327] frame_id "sensor_wgs84" is send on multiple topics. It was seen at least on /gx5/nav/odom and /gx5/nav/filtered_imu/data
E0407 12:23:16.956342  6302 rosbag_validate_main.cc:351] frame_id "livox_frame" on topic /livox/lidar has serialization time 1617811765.459779798 but sensor time 763.198977830 differing by -1.61781e+09 s.
W0407 12:23:16.956384  6302 rosbag_validate_main.cc:103] frame_id sensor_wgs84 time 1617811765550800365: IMU linear acceleration is 0.0111621 m/s^2, expected is [3, 30] m/s^2. (It should include gravity and be given in m/s^2.) linear_acceleration 00.00400424 00.00549399 -0.00885296
E0407 12:23:16.956398  6302 rosbag_validate_main.cc:316] Sensor with frame_id "sensor_wgs84" jumps backwards in time, i.e. timestamps are not strictly increasing. Make sure that the bag contains the data for each frame_id sorted by header.stamp, i.e. the order in which they were acquired from the sensor.
[... cut ...]
[0.065020, 0.075856)	                    	Count: 0 (0%)	Total: 1553 (50%)
[0.075856, 0.086693)	                    	Count: 0 (0%)	Total: 1553 (50%)
[0.086693, 0.097529)	                 ###	Count: 537 (17.2891%)	Total: 2090 (67.2891%)
[0.097529, 0.108366]	             #######	Count: 1016 (32.7109%)	Total: 3106 (100%)
```
for which I am unable to understand the exact issue since I see the timestamps increasing in the Lidar and the IMU messages. Could you please advise on the possible root cause? TIA.

Vineet-Pandey, 2021-04-12:
Also, when I check the node subscriptions, i receive the following response:
```
Node [/cartographer_node]
Publications: 
 * /constraint_list [visualization_msgs/MarkerArray]
 * /landmark_poses_list [visualization_msgs/MarkerArray]
 * /rosout [rosgraph_msgs/Log]
 * /scan_matched_points2 [sensor_msgs/PointCloud2]
 * /submap_list [cartographer_ros_msgs/SubmapList]
 * /tf [tf2_msgs/TFMessage]
 * /trajectory_node_list [visualization_msgs/MarkerArray]

Subscriptions: 
 * /gx5/nav/filtered_imu/data [sensor_msgs/Imu]
 * /livox/lidar [sensor_msgs/PointCloud2]
 * /tf [tf2_msgs/TFMessage]
 * /tf_static [tf2_msgs/TFMessage]

Services: 
 * /cartographer_node/get_loggers
 * /cartographer_node/set_logger_level
 * /finish_trajectory
 * /get_trajectory_states
 * /read_metrics
 * /start_trajectory
 * /submap_query
 * /trajectory_query
 * /write_state

contacting node http://USHAP0-D90TSRTX-UX:46603/ ...
Pid: 8188
Connections:
 * topic: /rosout
    * to: /rosout
    * direction: outbound (60769 - 127.0.0.1:58726) [18]
    * transport: TCPROS
 * topic: /tf
    * to: /cartographer_node
    * direction: outbound
    * transport: INTRAPROCESS
 * topic: /submap_list
    * to: /cartographer_occupancy_grid_node
    * direction: outbound (60769 - 127.0.0.1:58728) [19]
    * transport: TCPROS
 * topic: /tf
    * to: /cartographer_node (http://USHAP0-D90TSRTX-UX:46603/)
    * direction: inbound
    * transport: INTRAPROCESS
 * topic: /tf
    * to: /robot_state_publisher (http://USHAP0-D90TSRTX-UX:45477/)
    * direction: inbound (47812 - USHAP0-D90TSRTX-UX:52397) [20]
    * transport: TCPROS
 * topic: /tf_static
    * to: /robot_state_publisher (http://USHAP0-D90TSRTX-UX:45477/)
    * direction: inbound (47820 - USHAP0-D90TSRTX-UX:52397) [21]
    * transport: TCPROS
 * topic: /livox/lidar
    * to: /livox_lidar_publisher (http://USHAP0-D90TSRTX-UX:34063/)
    * direction: inbound (40452 - USHAP0-D90TSRTX-UX:58597) [22]
    * transport: TCPROS
 * topic: /gx5/nav/filtered_imu/data
    * to: /gx5/ros_mscl_node (http://USHAP0-D90TSRTX-UX:43805/)
    * direction: inbound (38726 - USHAP0-D90TSRTX-UX:44291) [23]
    * transport: TCPROS
 ```
which indicates that the cartographer is subscribing to the correct topics. Could someone advise on the possible root cause? TIA