cartographer-project/cartographer_ros #553: Writing ROS compatible maps

cagataysari, 2017-10-23:
Hello, I have been following #475 and #548. My odom, sensors etc.. everything works fine. I can write ros maps with tips in #475 with map_server but I want to write it with asset_writer but I could not figure out how to do it. Tips will be appreciated thanks.

SirVer, 2017-10-23:
@cagataysari Have you read through https://google-cartographer-ros.readthedocs.io/en/latest/assets_writer.html? Could you outline what you have tried?

cagataysari, 2017-10-23:
I went through this documentation, I did not really try it out since I am not using a bag file, I am running cartographer_ros while simulating my robot in gazebo so I did not use offline writing I want to write a yaml or pgm file when I am done. Should i record everything to a bag file then run asset_writer with the steps explained on this documentation and if i follow will i get a ros compatible pgm or yaml file ?? I want to figure out the easiest solution to write a map file compatible with ros. In summary I am running cartographer_ros , I see I am done with map in rviz then I wanna write the map with asset writer without recording bag files etcc.. I wanna do something similar to map_server map_saver with assert_writer. Thanks

SirVer, 2017-10-23:
> Should i record everything to a bag file then run assets writer with the steps explained on this documentation and if i follow will i get a ros compatible pgm or yaml file ?? 
>  Should I edit lua file to write ros compatible pgm file ?

Yes to both. You have to configure the `write_ros_map` step in the pipeline.

> I wanna write the map with assets writer without recording bag files etcc.. I wanna do something similar to map_server map_saver with assets writer.

That we do not currently support. Eventually we want to support writing a map from a .pbstream, but this is still in the works.

closed 2017-10-23 by cagataysari

guilhermelawless, 2017-10-23:
Here's a sample for using the new `write_ros_map` assets_writer

```
options = {
  tracking_frame = "base_laser_front_link",
  pipeline = {
    {
      action = "min_max_range_filter",
      min_range = 0.0,
      max_range = 5.,
    },
    {
      action = "write_ros_map",
      range_data_inserter = {
        insert_free_space = true,
        hit_probability = 0.55,
        miss_probability = 0.49,
      },
      filestem = "map",
      resolution = 0.05,
    }
  }
}
```

cagataysari, 2017-10-23:
thank you very much @guilhermelawless