cartographer-project/cartographer_ros #563: asset_writer define map resolution

Changliu52, 2017-10-30:
Hi,

I wonder if there is any parameter I can set to reduce the resolution for generating the point cloud (PLY file)?
It is very difficult to open the high resolution ply using meshlab to do any processing.

Thank you in advance

SirVer, 2017-10-30:
The PLY file contains all points that are not filtered out by any other points processor. If you just want to look at the point cloud, use https://github.com/googlecartographer/point_cloud_viewer which can deal with billions of points. If you really want to subsample your points, use a [`fixed_ratio_sampler`](https://github.com/googlecartographer/cartographer/blob/master/cartographer/io/fixed_ratio_sampling_points_processor.cc) in your pipeline configuration before the PLY writing step.

The configuration should look something like this:

```lua
{
   action = "fixed_ratio_sample",
   sampling_ratio = 0.01, 
}
```
This will only keep 1% of points.

closed 2017-10-30 by SirVer

Changliu52, 2017-10-30:
Great! Thank you @Holger Rapp. This exactly what I was looking for.

Chang