cartographer-project/cartographer_ros #567: Building error while running catkin_make_isolated --install --use-ninja

shahabty, 2017-11-01:
Hello,
I have been trying to re-install Cartographer. I have removed everything. But, when I tried to install it again I faced this error:
[135/260] Building CXX object CMakeFiles/cartographer.dir/cartographer/common/lua_parameter_dictionary.cc.o
FAILED: /usr/bin/g++    -isystem /usr/include/eigen3 -isystem /home/nabaviss/Desktop/catkin_ws/install_isolated/include -isystem /home/nabaviss/torch/install/include -isystem /usr/include/cairo -isystem /usr/include/glib-2.0 -isystem /usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/pixman-1 -isystem /usr/include/freetype2 -isystem /usr/include/libpng12 -I. -I/home/nabaviss/Desktop/catkin_ws/src/cartographer -isystem /usr/src/gmock/gtest/include -O3 -DNDEBUG    -pthread -std=c++11 -fPIC  -Wall -Wpedantic -Werror=format-security -Werror=missing-braces -Werror=reorder -Werror=return-type -Werror=switch -Werror=uninitialized -O3 -DNDEBUG -MMD -MT CMakeFiles/cartographer.dir/cartographer/common/lua_parameter_dictionary.cc.o -MF CMakeFiles/cartographer.dir/cartographer/common/lua_parameter_dictionary.cc.o.d -o CMakeFiles/cartographer.dir/cartographer/common/lua_parameter_dictionary.cc.o -c /home/nabaviss/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc
/home/nabaviss/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc: In function ‘void cartographer::common::{anonymous}::QuoteStringOnStack(lua_State*)’:
/home/nabaviss/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc:49:24: error: ‘lua_pushglobaltable’ was not declared in this scope
   lua_pushglobaltable(L);         // S: ... string globals
                        ^
/home/nabaviss/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc: In lambda function:
/home/nabaviss/Desktop/catkin_ws/src/cartographer/cartographer/common/lua_parameter_dictionary.cc:332:3: error: control reaches end of non-void function [-Werror=return-type]
   };
   ^
cc1plus: some warnings being treated as errors
[135/260] Building CXX object CMakeFiles/cartographer.dir/cartographer/mapping/proto/sparse_pose_graph.pb.cc.o
ninja: build stopped: subcommand failed.
<== Failed to process package 'cartographer': 
  Command '['/home/nabaviss/Desktop/catkin_ws/install_isolated/env.sh', 'ninja', '-j8', '-l8']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /home/nabaviss/Desktop/catkin_ws/build_isolated/cartographer && /home/nabaviss/Desktop/catkin_ws/install_isolated/env.sh ninja -j8 -l8

Command failed, exiting.

Please tell me why I got this error and how to solve this.

gaschler, 2017-11-02:
This is unrelated to cartographer_ros, please follow your issue in the cartographer project, closing.

closed 2017-11-02 by gaschler

reyanshsolis, 2019-05-21:
same error while installing Cartographer_ros
Please Help