cartographer-project/cartographer_ros #571: Non-sorted data added to queue: '(0, points2)'

Calvin4016, 2017-11-03:
Dear Cartographer developers,

I have a couple issues that are I could not fix on my own and I'm seeking help. When I launch a lightly modified version of lua, urdf, and launch files found here: https://gist.github.com/Vincent1923/9b4dd8bbfd9301dd408e89f6e21cead7 , I received the following error message: (first it's showing "IMU not yet initialized", then shown with a FATAL error, I've highlighted them with Bold font)

SUMMARY
========

PARAMETERS
 * /pointcloud_to_laserscan/angle_increment: 0.0087
 * /pointcloud_to_laserscan/angle_max: 3.14
 * /pointcloud_to_laserscan/angle_min: -3.14
 * /pointcloud_to_laserscan/concurrency_level: 1
 * /pointcloud_to_laserscan/max_height: 1.0
 * /pointcloud_to_laserscan/min_height: 0.0
 * /pointcloud_to_laserscan/range_max: 130.0
 * /pointcloud_to_laserscan/range_min: 0.4
 * /pointcloud_to_laserscan/scan_time: 0.3333
 * /pointcloud_to_laserscan/target_frame: velodyne
 * /pointcloud_to_laserscan/transform_tolerance: 0.01
 * /pointcloud_to_laserscan/use_inf: True
 * /robot_description: <...>
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /use_sim_time: True
 * /velodyne_nodelet_manager_cloud/calibration: /home/ANT.AMAZON....
 * /velodyne_nodelet_manager_cloud/max_range: 130.0
 * /velodyne_nodelet_manager_cloud/min_range: 0.4
 * /velodyne_nodelet_manager_driver/device_ip: 
 * /velodyne_nodelet_manager_driver/frame_id: velodyne
 * /velodyne_nodelet_manager_driver/model: VLP16
 * /velodyne_nodelet_manager_driver/pcap: 
 * /velodyne_nodelet_manager_driver/port: 2368
 * /velodyne_nodelet_manager_driver/read_fast: False
 * /velodyne_nodelet_manager_driver/read_once: False
 * /velodyne_nodelet_manager_driver/repeat_delay: 0.0
 * /velodyne_nodelet_manager_driver/rpm: 600.0

NODES
  /
    cartographer_node (cartographer_ros/cartographer_node)
    cartographer_occupancy_grid_node (cartographer_ros/cartographer_occupancy_grid_node)
    playbag (rosbag/play)
    pointcloud_to_laserscan (pointcloud_to_laserscan/pointcloud_to_laserscan_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    static_transform_publisher3 (tf/static_transform_publisher)
    velodyne_nodelet_manager (nodelet/nodelet)
    velodyne_nodelet_manager_cloud (nodelet/nodelet)
    velodyne_nodelet_manager_driver (nodelet/nodelet)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[velodyne_nodelet_manager-1]: started with pid [20416]
process[velodyne_nodelet_manager_driver-2]: started with pid [20417]
process[velodyne_nodelet_manager_cloud-3]: started with pid [20418]
process[pointcloud_to_laserscan-4]: started with pid [20419]
process[robot_state_publisher-5]: started with pid [20425]
process[static_transform_publisher3-6]: started with pid [20437]
process[cartographer_node-7]: started with pid [20447]
process[cartographer_occupancy_grid_node-8]: started with pid [20452]
process[rviz-9]: started with pid [20464]
process[playbag-10]: started with pid [20474]
[ INFO] [1509674459.047857414]: I1102 19:00:59.000000 20447 configuration_file_resolver.cc:40] Found '/home/ANT.AMAZON.COM/hwguo/Cartographer_ws/install_isolated/share/cartographer_ros/configuration_files/vlp16-imu.lua' for 'vlp16-imu.lua'.
[ INFO] [1509674459.048119201]: I1102 19:00:59.000000 20447 configuration_file_resolver.cc:40] Found '/home/ANT.AMAZON.COM/hwguo/Cartographer_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1509674459.048157569]: I1102 19:00:59.000000 20447 configuration_file_resolver.cc:40] Found '/home/ANT.AMAZON.COM/hwguo/Cartographer_ws/install_isolated/share/cartographer/configuration_files/map_builder.lua' for 'map_builder.lua'.
[ INFO] [1509674459.048211947]: I1102 19:00:59.000000 20447 configuration_file_resolver.cc:40] Found '/home/ANT.AMAZON.COM/hwguo/Cartographer_ws/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1509674459.048243295]: I1102 19:00:59.000000 20447 configuration_file_resolver.cc:40] Found '/home/ANT.AMAZON.COM/hwguo/Cartographer_ws/install_isolated/share/cartographer/configuration_files/sparse_pose_graph.lua' for 'sparse_pose_graph.lua'.
[ INFO] [1509674459.048364437]: I1102 19:00:59.000000 20447 configuration_file_resolver.cc:40] Found '/home/ANT.AMAZON.COM/hwguo/Cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[ INFO] [1509674459.048395148]: I1102 19:00:59.000000 20447 configuration_file_resolver.cc:40] Found '/home/ANT.AMAZON.COM/hwguo/Cartographer_ws/install_isolated/share/cartographer/configuration_files/trajectory_builder.lua' for 'trajectory_builder.lua'.
[... 62 lines cut ...]
[0.094830, 0.096828)	                    	Count: 0 (0.000000%)	Total: 4 (0.312989%)
[0.096828, 0.098825)	                    	Count: 8 (0.625978%)	Total: 12 (0.938967%)
[0.098825, 0.100822)	                ####	Count: 243 (19.014084%)	Total: 255 (19.953053%)
[0.100822, 0.102819)	    ################	Count: 1009 (78.951485%)	Total: 1264 (98.904541%)
[0.102819, 0.104816)	                    	Count: 9 (0.704225%)	Total: 1273 (99.608765%)
[0.104816, 0.106814)	                    	Count: 2 (0.156495%)	Total: 1275 (99.765259%)
[0.106814, 0.108811)	                    	Count: 1 (0.078247%)	Total: 1276 (99.843506%)
[0.108811, 0.110808]	                    	Count: 2 (0.156495%)	Total: 1278 (100.000000%)
I1102 19:10:40.236456 24648 rosbag_validate_main.cc:154] Time delta histogram for consecutive messages on topic "/scan" (frame_id: "velodyne"):
Count: 2556  Min: -0.100869  Max: 0.110808  Mean: 0.050430
[-0.100869, -0.079701)	                    	Count: 1 (0.039124%)	Total: 1 (0.039124%)
[-0.079701, -0.058534)	                    	Count: 0 (0.000000%)	Total: 1 (0.039124%)
[-0.058534, -0.037366)	                    	Count: 0 (0.000000%)	Total: 1 (0.039124%)
[-0.037366, -0.016198)	                    	Count: 0 (0.000000%)	Total: 1 (0.039124%)
[-0.016198, 0.004970)	          ##########	Count: 1276 (49.921753%)	Total: 1277 (49.960876%)
[0.004970, 0.026137)	                    	Count: 0 (0.000000%)	Total: 1277 (49.960876%)
[0.026137, 0.047305)	                    	Count: 0 (0.000000%)	Total: 1277 (49.960876%)
[0.047305, 0.068473)	                    	Count: 0 (0.000000%)	Total: 1277 (49.960876%)
[0.068473, 0.089640)	                    	Count: 0 (0.000000%)	Total: 1277 (49.960876%)
[0.089640, 0.110808]	          ##########	Count: 1279 (50.039124%)	Total: 2556 (100.000000%)

I am trying to get permission from my professor to upload to bag but I am not optimistic. Is there a way you can help me out with what I've got here?

Thank you!
-Calvin

gaschler, 2017-11-03:
Hi Calvin,
thanks for sharing your fork. You can post the results of /devel_isolated/cartographer_ros/lib/cartographer_ros/cartographer_rosbag_validate 
In case this doesn't help, we would need to look at the bag file.

Calvin4016, 2017-11-03:
Thanks for your prompt feedback. I was able to identify the issue and it was indeed a human error (caused by another node overwriting the correct tf and imu data) and not a Cartographer issue.

closed 2017-11-03 by Calvin4016

letsbelopez, 2017-12-08:
@Calvin4016 I think we have a similar setup and I'm getting similar output when running the validate tool on my bag file. I don't have a IMU though, so I'm just using 2d options. 

Would you be able to explain what your solution was or if you could take a look at my issue #623  please?