cartographer-project/cartographer_ros #579: Initial Pose Wiring

cschuet, 2017-11-07:
The initial pose implementation in https://github.com/googlecartographer/cartographer/pull/606 is ready to merge but as of yet untested.

Initial poses need to be wired up in cartographer_ros (probably have an initial_pose topic) and tested that they work correctly.

cschuet, 2017-11-07:
@Mofef @damienrg @apr8 @ojura @jihoonl Any interested in taking this?

damienrg, 2017-11-07:
I can do it.
Considering implementation, I think that I should add an optional pose in start trajectory rather than subscribing to another one topic as we have to create a new trajectory in any case.
Is it ok for you?

cschuet, 2017-11-08:
That sounds like a good approach to me.

jihoonl, 2017-11-08:
Can you take a look #513? It was a follow PR of our initialpose implementation. Surely, it can be tuned to use together with googlecartographer/cartographer#606

cschuet, 2017-11-08:
Ah, nice! @jihoonl do you want to adapt that PR then since it is already written? Or would you rather have @damienrg take that over? I am currently busy with other work items and I don't want to get in the way of this feature.

jihoonl, 2017-11-08:
Sure no problem. I will simply rename `initialpose` as `relative_pose` and add `int32 to_trajectory` in `StartTrajectory.srv`. I am thinking to use frame_id field as indicator if this relative_pose is used. How do you think?

cschuet, 2017-11-08:
I would avoid sentinel values to implement an Optional<>. Could you define an InitialPose submessage and have that as an optional field?

jihoonl, 2017-11-08:
Sadly, ROS message doesn't support optional field...

cschuet, 2017-11-08:
I see. How about a repeated field as a poor-man's optional? Is that something people do? @SirVer what do you suggest here?

SirVer, 2017-11-08:
My 2c: Add `relative_pose` and `trajectory_id` to `TrajectoryOptions.msg`, since it is not ROS specific (StartTrajectory.srv takes `options` which is everything concerning the SLAM and `topics` which is ROS specific). And I think `trajectory_id = -1` as sentinel value is the ROS way of life here.

damienrg, 2017-11-08:
I set to_trajectory_id to -1. If you want I just pushed the [initial-pose-branch](https://github.com/damienrg/cartographer_ros/tree/initial-pose) on which I am working to see how I started.

cschuet, 2017-11-08:
We also need to add the time when the robot saw the other robot under the transformation 'relative_pose'.

damienrg, 2017-11-08:
Maybe I should open a PR with with actual work to discuss on it?
For now I have encountered the following "flaws":
- initial pose is not handle like pure localization, I think it should be added in proto::TrajectoryBuilderOptions or there should be distinction between "optional arguments" and "mandatory ones".
- the way I handle time is not perfect but for now I do not have a better idea

cschuet, 2017-11-08:
Happy to look at any PRs. It's not ideal to write two implementations when only one gets merged, so maybe discuss among yourselves @damienrg  and @jihoonl who wants to put in the work to write this thing.

jihoonl, 2017-11-08:
Um since im in vacation, I will be pretty slow until next week. So, @damienrg feel free to take over.

cschuet, 2017-11-08:
Enjoy your well-deserved vacation then :-D

wohe, 2017-11-08:
Reading this discussion, I wonder if it would not be better to adapt googlecartographer/cartographer#606 to include this information in the trajectory options proto message, similar to how pure localization is handled. This way, there would not even be a need for any change in ROS.

cschuet, 2017-11-08:
Yes, that'd make sense, but I think it should be a follow-up PR on top of googlecartographer/cartographer#606.

damienrg, 2017-11-08:
What does it mean? Fix #579 to test googlecartographer/cartographer#606, create a new PR in cartographer and drop the fix for #579.
I am more with @wohe to update googlecartographer/cartographer#606.

SirVer, 2017-11-08:
> I am more with @wohe to update googlecartographer/cartographer#606.

+1.

cschuet, 2017-11-08:
Updated googlecartographer/cartographer#606.

cschuet, 2017-11-08:
Alright googlecartographer/cartographer#606 is merged. @damienrg  I will let you take it from here. Thanks.

damienrg, 2017-11-08:
Since https://github.com/googlecartographer/cartographer/pull/606 and as @wohe mentioned there is nothing to do in cartographer_ros as it is handle in TrajectoryBuilderOptions and thus read by start trajectory node.

closed 2017-11-24 by wally-the-cartographer

kajep09, 2017-12-07:
Would it be possible to get a short description or an example of how to use the initial_pose option in the configuration files?

doisyg, 2017-12-13:
Indeed, an example would be much appreciated

damienrg, 2017-12-13:
You can not set it in configuration files, you need to call `start_trajectory_main` with initial_pose. 
For example you can use it this way:
- start slam with cartographer_node
- call finish trajectory when the robot does not move
- call `start_trajectory_main --initial_pose '{to_trajectory_id = 0, relative_pose = { translation = { 0., 0., 0. }, rotation = { 0., 0., 0., } } }' `

Just keep in mind that the position is relative to the trajectory `to_trajectory_id` at the specified `timestamp` (if not specified it means `now`).

See https://github.com/googlecartographer/cartographer/pull/642#issuecomment-345177084 to have another one example.

doisyg, 2017-12-13:
Thanks for your answer, but it is still very unclear to me.

- start slam with cartographer_node 
 => OK

- call finish trajectory when the robot does not move 
=> Done by calling the ros service `/finish_trajectory`

- call `start_trajectory_main --initial_pose '{to_trajectory_id = 0, relative_pose = { translation = { 0., 0., 0. }, rotation = { 0., 0., 0., } } }'`
 => How to do that? there is a ros service `/start_trajectory` but there is no field for ` --initial_pose` , is `start_trajectory_main` suppose to be a service or something else?

damienrg, 2017-12-13:
I meant `rosrun cartographer_ros cartographer_start_trajectory ... -initial_pose '{to_trajectory_id = 0, relative_pose = { translation = { 0., 0., 0. }, rotation = { 0., 0., 0., } } }'`.

doisyg, 2017-12-18:
Thanks! This is working.
For the ones willing to try with pure localization:

1.  Start cartographer node in pure localization mode. By default the map trajectory ID is 0 and the node automatically starts a new one (ID 1)
2. call /finish_trajectory for ID 1
3. `rosrun cartographer_ros cartographer_start_trajectory ... -initial_pose '{to_trajectory_id = 0, relative_pose = { translation = { 0., 0., 0. }, rotation = { 0., 0., 0., } } }'`.

NTUwilliam, 2019-03-02:
@doisyg @damienrg 
I'm wondering the correct **steps** for setting `initial_pose` for **pure localization mode**. Is it like this? :

1. Start cartographer node in pure localization mode by:
```
roslaunch cartographer_ros demo_backpack_2d_localization.launch \
   load_state_filename:=${HOME}/xxxx/xxxx.bag.pbstream \
   bag_filename:=${HOME}/xxxx/xxxx.bag
```

2. How to call `/finish_trajectory` for the map trajectory ID? Which means, what should I specify on the command line?

3. `rosrun cartographer_ros cartographer_start_trajectory ... -initial_pose '{to_trajectory_id = 0, relative_pose = { translation = { 0., 0., 0. }, rotation = { 0., 0., 0., } } }`

Are the steps correct?

Thanks!!

ojura, 2019-03-02:
Do not confuse Google flags when you are executing Cartographer executables directly (`-flag=name`, for some types of flags you can also use a space instead of equals) with ROS node or launch file arguments (these use`:=`).

Also, offline_backpack_2d.launch is not pure localization. It is regular mapping which produces a .pbstream file which you can then load as a map for pure localization with `-load_state_filename`. Please carefully examine the provided example launch files and the documentation on readthedocs.

NTUwilliam, 2019-03-04:
@ojura 
Thanks for all your guidance. I've made it work!!

@doisyg @damienrg 
I think the correct **steps** for setting `initial_pose` for **pure localization mode** should be like this:

1. Start cartographer node in pure localization mode by:
```
roslaunch cartographer_ros demo_backpack_2d_localization.launch \
   load_state_filename:=${HOME}/xxx.bag.pbstream \
   bag_filename:=${HOME}/xxx.bag
```

2. Set the initial pose in **another Terminal** right after step 1:
`rosrun cartographer_ros cartographer_start_trajectory ... -configuration_directory '${HOME}/xxx/xxx/cartographer_ros/cartographer_ros/configuration_files' -configuration_basename 'backpack_2d_localization.lua' -initial_pose '{to_trajectory_id = 0, relative_pose = { translation = { 0, 0, 0}, rotation = {0., 0., 0., } } }'`

Thanks for all your works!!

NTUwilliam, 2019-03-11:
@ojura @cschuet @wohe 
Sorry for bothering you guys again.

Although I've successfully set the initial **translation** for the `pure_localization` process, I find it's still very hard for the robot to localize itself in large symmetric environment without the **rotation** information. 
And I'm still not clear about what determines the **orientation** of the `/map` frame, as shown in [issue /1218](https://github.com/googlecartographer/cartographer_ros/issues/1218), so it's quite hard for me to find the correct rotation information. 

So I'm wondering how can I get the **rotation** information (mainly the **Yaw** angle) to more accurately initialize the robot?

@doisyg @damienrg If you got any idea or advice, feel free to comment.

Thanks!!

soorajanilkumar, 2020-01-13:
There is an important field (timestamp) missing in the above syntax.  If it is not set, by default I think it takes the pose at the last time stamp of trajectory 0 as the reference frame, which can mess things up. 
I found this in [issue 913](https://github.com/googlecartographer/cartographer_ros/issues/913#issuecomment-411740137)

`rosrun cartographer_ros cartographer_start_trajectory  -configuration_directory <path-to-config> -configuration_basename <config-filename.lua> -initial_pose '{to_trajectory_id = 0, relative_pose = { translation = {0.0, 0.0, 0. }, rotation = { 0.0, 0.0, 0.0 } } ,timestamp = 0}'
`
This should set the initial pose of the new trajectory with respect to the map frame hopefully.

Hope this helps anyone looking at the issue in future.

eithwa, 2023-08-02:
Hello,

I would like to ask whether the start_trajectory function using rosservice has the same effect as the cartographer_start_trajectory node, as the master branch currently does not contain the cartographer_start_trajectory node.

Additionally, I'd like to inquire about the relative_to_trajectory_id parameter. If it is set to 0, does it refer to the first set of points in the trajectory_node_list to set the coordinates?

Currently, I am facing an issue where I use this function to update the coordinates to position: {x: -14.729079, y: 0.266100, z: 0.0}, but after updating the coordinates, the position becomes -14.723, 0.433098, trajectory id 0 is as follows:
```
sdmarkers:
header:
  seq: 0
  stamp:
    secs: 1596
    nsecs: 751000000
  frame_id: "map"
ns: "Trajectory 0"
id: 0
type: 4
action: 0
pose:
  position:
    x: 0.0
    y: 0.0
    z: 0.05
  orientation:
    x: 0.0
    y: 0.0
    z: 0.0
    w: 1.0
scale:
  x: 0.07
  y: 0.0
  z: 0.0
color:
  r: 0.207129895687
  g: 0.115499980748
  b: 0.769999980927
  a: 1.0
lifetime:
  secs: 0
  nsecs: 0
frame_locked: False
points:
  -
    x: 0.00551946445911
    y: -0.00345368672909
    z: 0.0
  -
  ```
Is there any way to update the coordinates accurately?

PR #610 (2017-11-24): Add initial_pose in start_trajectory_main.cc Fixes #579
Fixes #579 
Related to googlecartographer/cartographer#606

@damienrg @cschuet I followed most of the comments in googlecartographer/cartographer#642 except timestamp. Receiving timestamp sounds weird to me because trajectory should not start in past timestamp or future timestamp.
commit e1f65f31 (2017-11-24): Add initial_pose in start_trajectory_main.cc Fixes #579 (#610)

Fixes #579 
Related to googlecartographer/cartographer#606

@damienrg @cschuet I followed most of the comments in googlecartographer/cartographer#642 except timestamp. Receiving timestamp sounds weird to me because trajectory should not start in past timestamp or future timestamp.