cartographer-project/cartographer_ros #585: Simultaneous trajectories in offline node

ojura, 2017-11-08:
If we have multiple bags containing simultaneous trajectories (or even if they are all in one bag?), the offline node should support handling this similarly to the online node. For example, in the case with multiple bags, they could be collated before being pushed into the sensor queue.

SirVer, 2017-11-08:
This is an interesting suggestion. I think a RFC should describe how this should be configured.

ojura, 2017-11-08:
My thoughts on this so far:

- The proposed offline node configuration for multiple (possibly simultaneous) trajectories should closely follow the way multiple simultaneous trajectories can already be achieved in the online node, listed here for reference:
    - Performing multiple calls of the `StartTrajectory` service, with possibly different `TrajectoryOptions` and necessarily different sensor data topics (all specified in the service request).
    - Alternatively, the `start_trajectory` helper node is invoked multiple times, with possibly different `.lua` files for `TrajectoryOptions` and necessarily different sensor data topics (the recently merged PR #584 enables specifying different topics via topic remapping).
    - Specially, if a trajectory is finalized, we may start another independent trajectory which uses the same sensor data topics and possibly different `TrajectoryOptions`. This is already achievable in the offline node by simply loading multiple bags, which creates a trajectory for each bag, although `TrajectoryOptions` are necessarily the same for all trajectories. The ability to do this has to be preserved or expanded (*1).

- If we suppose that there is a mandated 1:1 relationship between bags and trajectories in the offline node, I propose the following configuration. For each bag (trajectory), the configuration would be a tuple consisting of:
    - `TrajectoryOptions` (given in a `.lua` file)
    - A set of sensor data topics
    - Boolean `use_bag_transforms` which allows us to ignore transforms in the bag
    - An optional `.urdf` file.
- Both `TrajectoryOptions` and the sensor data topics may be identical for multiple different bags. This ensures (*1). However, in addition to that, the trajectories may be concurrent.
  - Because the trajectories may be concurrent, due care would need to be taken to ensure that data published on the same topic, but in different bags, gets routed to the appropriate trajectory. Also, each trajectory should have its own transform buffer, to avoid interference between different bags. 
    - For example, we may have two identical separate robots driving concurrently; all the frames and topics are identical for both robots. The online node cannot handle this (we would have to ensure that the sensor data topics would be different by e.g. using different namespaces for each robot, as well as having different frames; conflicting transforms with same frames would be an unsolvable nightmare). 
     - However, if we had a separate bag for each robot, the offline node would have the advantage of knowing which data originated from which bag, so it could handle this case. As mentioned in (*1), the offline node already supports this for non-concurrent (sequential) trajectories of identical robot types (having identical `TrajectoryOptions` and `.urdf` by design).

- Going further, allowing multiple trajectories in a single bag and relaxing the 1 trajectory:1 bag restriction would mean that the configuration of each bag becomes a set of tuples mentioned above.

All the concurrent trajectories (whether in the same bag, or in different bags) would be collated. This implies that sequential trajectories would remain sequential, even if using the same topics.

ojura, 2017-11-09:
Okay, I am not finished yet :-)

There's another case which I believe is common - a single continous trajectory that is split into multiple bags (e.g. when using the split option of `rosbag record`).

We could handle also this with the following configuration structure:

- Configuration is given per trajectory. Each configuration is a tuple of:
  - A list of continuous bags for this trajectory
  - A set of sensor data topics
  - boolean `use_data_frames` and optional .urdf
- Multiple trajectories may share the same bags, but in that case they must differ in sensor data topics.
- As before, any concurrent trajectories are collated.

SirVer, 2017-11-09:
Please write an rfc, design decisions should go there.

ojura, 2017-11-09:
Will do. I just had to dump this in a comment while I had a clear picture.

ojura, 2017-11-09:
@SirVer googlecartographer/rfcs#4

SirVer, 2017-11-14:
Closing till RFC is merged, since it is not actionable till then.

closed 2017-11-14 by SirVer