cartographer-project/cartographer_ros #586: Running bagfile with cartographer_ros

shahabty, 2017-11-09:
Hello,
I would like to run cartographer on my own videos
I converted them into bag file. 
However, when I'm running the bagfile with backpack_2d launch file. rviz opens  and nothing happens.
I appreciate it if you could tell me how to write my own .launch and config files in order to run cartographer on my own bag files. 
Additional info: the data is RGB frames which I converted them to bagfile.

SirVer, 2017-11-09:
This lacks a lot of information and in its current form is not actionable. 

Please run through our [issue template](https://github.com/googlecartographer/cartographer_ros/blob/master/.github/ISSUE_TEMPLATE) and reopen a new issue providing as much of the asked for information as possible. Especially your data and your configuration files are important.

closed 2017-11-09 by SirVer

gaschler, 2017-11-09:
Hi Shahabty,
thanks for your interest in Cartographer.
_Cartographer requires range data_, RGB videos cannot be processed.
Please have a look at https://google-cartographer.readthedocs.io/en/latest/index.html