cartographer-project/cartographer_ros #607: occupancy grid origin no longer corresponds to initial point

BrannonKing, 2017-11-17:
Back before the creation of occupancy_grid_node, the origin published in the occupancy grid message corresponded to starting location & angle of the vehicle. occupancy_grid_node does not appear to work that way now; it uses the submap bounding box corner for an origin. My company relies heavily upon the proper origin -- the vehicle always starts in the same place. We don't use Cartographer for localization -- only mapping.

BrannonKing, 2017-11-21:
I got to the bottom of this: it's the same issue I've struggled with in other areas. Cartographer does not support rotational transforms on the tracking frame. Any rotation on the tracking frame breaks things like this. We need to catch that and chuck an error.

wohe, 2017-11-22:
I do not completely understand what you are reporting as an issue. My understanding is that you are concerned with the fact that the published `nav_msgs/OccupancyGrid` messages always contain grids without rotation. This is intentional and to my knowledge was always the case with Cartographer. While there is a yaw value in the message, [the documentation of map_server says](http://wiki.ros.org/map_server) that only a zero yaw is supported everywhere. Thus it seems safest to publish the map with a zero yaw.

Or are you concerned about something different?

BrannonKing, 2017-11-22:
If your tracking frame contains any non-zero roll/pitch/yaw in its transform then the occupancy grid produced is not rotated correctly. @SirVer 's comment (https://github.com/googlecartographer/cartographer/issues/534) was: "In 2D without IMU the tracking frame has to be z-aligned, orthogonal to the plane of movement." However, even with the IMU, my occupancy grids end up twisted around Z incorrectly when there is a non-zero roll/pitch/yaw in the tracking frame.

wohe, 2017-11-22:
I still do not understand what issue you see:

@SirVer was commenting on the fact that without an IMU the tracking frame must be gravity aligned, i.e. we assume no roll and pitch. Of course there can be yaw between tracking frame and map frame, i.e. the robot is allowed to rotate around the z-axis

I understand you expect a different rotation from the occupancy grid: this has nothing to do with the tracking frame, the occupancy grid contains pixels which have the same orientation as the map frame so that its reported yaw is zero as I explained above. Since we generate those pixels we can theoretically choose whatever orientation we would like. It seems you are unhappy with our choice? But maybe I misunderstand you here.

To clarify: Are you happy with the SLAM result but think the published occupancy grid is bad? Or is the SLAM result already broken for you, e.g. when looking at the visualization using submaps?

BrannonKing, 2017-11-22:
> Are you happy with the SLAM result but think the published occupancy grid is bad?

I'm happy with the shape. The rotation of the published occupancy grid is bad (when tracking frame's transform has non-zero yaw).

I expect that the initial orientation of the vehicle is significant -- that the occupancy grid output will keep the same orientation of the initial vehicle pose. That does not work in all situations.

BrannonKing, 2017-11-22:
When I go back to work next week I will post some images to clarify my meaning.

BrannonKing, 2017-11-27:
To try and further explain my issue, look at the image below. If the vehicle's initial position matched the blue arrow then the image is correct. If the vehicle matched the red arrow then the exported image is wrong. When using rotation transforms in the tracking frame I get the wrongly oriented output.
[image]

Entropy512, 2018-04-24:
OK, most of my previous experience with Cartographer was with 2D without an IMU - in all of my use cases, the tracking frame and published frame were different.  Also, I pretty much ignored your question because I thought "well, Cartographer has no way of knowing what your actual rotation was".

Now that I'm working on trying to integrate an IMU, I came across your question again, and I'm wondering if there was a nuance that you didn't make clear (it isn't to me):

Are you saying that you get bad output when the tracking frame is **rotated in relation to the published frame**?  E.g. you expect the map to be in relation to the published frame, but if the tracking frame is rotated relative to the published frame (for example, due to an IMU being mounted with a 90 degree yaw compared to base_link), then the map is rotated?

Now after my first attempts to integrate Cartographer with an IMU, I see that the tracking frame has to be co-located with the IMU frame - so offset x/y/z have to be 0, but can the IMU frame be rotated with respect to the tracking frame?  (e.g. a pure yaw/pitch/roll transform)?  Is that the solution?

Also - if I'm reading the Microstrain documentation correctly, +z is through the bottom of the sensor, while in ROS, +z is upwards - so if the Microstrain is facing upwards and the connector is towards the rear of the vehicle, is 180 degree roll followed by 180 degree yaw the correct orientation in my URDF?

BrannonKing, 2018-04-24:
Yes, the IMU can be rotated as long as it is colocated with the tracking frame. Publish different transforms for each as the tracking frame cannot be rotated.

wohe, 2018-04-25:
There are no constraints on the orientation. The IMU frame must be at the same location as the tracking frame. The published frame is not constrained in this way, i.e. it can be at a different location than the IMU frame or tracking frame.

Fanny-one, 2020-03-16:
Hi, @wohe @Entropy512 @BrannonKing 
I have met the similar problem. When there is a 90 deg yaw between the imu and base_link, the map_frame and base_link will create a 90 deg yaw. But I don't want this result. The detail have present in another issue.
https://github.com/googlecartographer/cartographer_ros/issues/1455
Thans in advance.

MichaelGrupp, 2020-10-26:
Drive-by comment while cleaning up old issues: I still don't completely understand the original issue as I agree with @wohe that this is expected behavior, from the description that was provided. An occupancy grid's orientation is relative to the respective map's origin and by "standard" (aka map_server) the map-origin to pixel-origin transform has no rotational component. So since our "map" frame ID of the occupancy_grid_publisher in Cartographer is the SLAM origin, this will be bound to where you start mapping.

Since you say that the orientation of the image is wrong, it depends what your frame of reference is. IMO the best practice is to have fixed inertial frame of reference as parent of the SLAM coordinate system. This allows alignment of the SLAM map inside a reference coordinate system to compensate for the more-or-less arbitrary start pose of SLAM.

closed 2020-10-26 by MichaelGrupp