cartographer-project/cartographer_ros #623: Not SLAMing with 2d Laser Scan and no IMU

letsbelopez, 2017-12-08:
First off thank you for making this technology accessible. Really appreciate it. 

I've gone through the issues and tuned the configurations from other examples I've seen from people trying to 2d SLAM without IMU, but I still can't get cartographer to start making a map.

I'm new to ROS, but I've gone through the tutorials and now I've hit a wall and don't know where to go. Hoping I've made this easy for someone to point me in a good direction.

My Setup: I have a Velodyne VLP-16 on a tripod on top of a library cart weighed down by sandbags. I role the cart around a loop of an office. I have the velodyne package running, which publishes sensor_msgs/LaserScan data to scan topic. I'm pretty sure I have the configs setup to not take in additional sensor data and to use correlative scan matching, but I couldn't find many examples of this so I may be wrong.

1. Link to gist of validate bag output: 
https://gist.github.com/letsbelopez/cb2d0a81130d9cc9d56f4a022155f174

2. Link to repo branch: 
https://github.com/letsbelopez/cartographer_ros/tree/Fresh

I'm using this launch file: 
https://github.com/letsbelopez/cartographer_ros/blob/Fresh/cartographer_ros/launch/demo_fresh_2d.launch

3. Link to bag file:
https://drive.google.com/file/d/1dEadzMIPdMJXB6V4TOFRhb7lxGlvLArq/view?usp=sharing

gaschler, 2017-12-11:
Without doing any deeper checks, I believe that you are missing a remap to forward the velodyne data to the cartographer_node.

Something like `<remap from="scan" to="horizontal_laser_2d" />`, as in backpack_2d.launch.

Then, use the command `rqt_graph` to see which data flows to the node.

Let me know if this doesn't help.

letsbelopez, 2017-12-11:
@gaschler  I appreciate the help on this. I tried the remap but had no success.

The velodyne package is publishing LaserScan messages on the scan topic, and cartographer is subscribing to the scan topic. That's why I thought that I didn't need a remap.

My graph after remapping
[image]

My graph without remapping.
[image]

Rviz Screen Shot
[image]

Entropy512, 2017-12-12:
Hmm...  2d with a VLP-16?  Regardless of any of the other issues you might be having, I thought none of the beams were at 0 degrees elevation on that unit which would cause issues with 2D-only mapping?

letsbelopez, 2017-12-12:
@Entropy512 thank you for the help. Definitely will help me to improve this map making project.

Looking at the VLP-16 manual [here](http://velodynelidar.com/docs/manuals/VLP-16%20User%20Manual%20and%20Programming%20Guide%2063-9243%20Rev%20A.pdf) it shows a table on page 21 with the lasers and vertical angles. None are at 0 degrees. Two lasers are at 1 and -1 degree. 

If the angle would cause issues for 2D, I would still like to continue to get cartographer to start drawing a map. Even if it's not a good map. Because cartographer hadn't started drawing the map, I thought it may have been some other issue.

letsbelopez, 2017-12-13:
I just got an IMU today. I'm using a VectorNav VN-100 and the corresponding [imu_vn_100 package](https://github.com/KumarRobotics/imu_vn_100). 

I recorded another bag file that can be found [here](https://drive.google.com/open?id=1XHtcu_FLkIE5PnvfsoP1mS6861j40zDy)

And the new rqt_graph:
[image]

Launching [this](https://github.com/letsbelopez/cartographer_ros/blob/Fresh/cartographer_ros/launch/demo_fresh_3d.launch) new launch file for 3d.

I'm still not getting cartographer to draw a map though.

gaschler, 2017-12-13:
frame_id did not match. Here is the config fix for you.

I downloaded your bag and config. roslaunch gave the following warning.
[ WARN] [1513170091.250466277, 1513118322.929107518]: W1213 14:01:31.000000 104146 tf_bridge.cc:52] "velodyne" passed to lookupTransform argument source_frame does not exist. 

Opening your bag file with rqt_bag showed that "scan" topic messages have header.frame_id "velodyne".
This frame_id needs to be configured in urdf and in the lua config.

```
diff --git a/cartographer_ros/configuration_files/fresh_2d.lua b/cartographer_ros/configuration_files/fresh_2d.lua
index 7bfd91f..7c9d094 100644
--- a/cartographer_ros/configuration_files/fresh_2d.lua
+++ b/cartographer_ros/configuration_files/fresh_2d.lua
@@ -19,8 +19,8 @@ options = {
   map_builder = MAP_BUILDER,
   trajectory_builder = TRAJECTORY_BUILDER,
   map_frame = "map",
-  tracking_frame = "base_link",
-  published_frame = "base_link",
+  tracking_frame = "velodyne",
+  published_frame = "velodyne",
   odom_frame = "odom",
   provide_odom_frame = false,
   use_odometry = false,
diff --git a/cartographer_ros/urdf/vlp16.urdf b/cartographer_ros/urdf/vlp16.urdf
index ee91f05..c7af19b 100644
--- a/cartographer_ros/urdf/vlp16.urdf
+++ b/cartographer_ros/urdf/vlp16.urdf
@@ -19,7 +19,7 @@
     <color rgba="0.2 0.2 0.2 1" />
   </material>
 
-  <link name="velodyne_link">
+  <link name="velodyne">
     <visual>
       <origin xyz="0.0 0.0 0.0" />
       <geometry>
@@ -33,7 +33,7 @@
 
   <joint name="velodyne_link_joint" type="fixed">
     <parent link="base_link" />
-    <child link="velodyne_link" />
+    <child link="velodyne" />
     <origin xyz="0 0 0" rpy="0 0 0" />
   </joint>
 </robot>
```

Then the 2D mapping works. Feel free to add IMU again or try 3D, should be equivalent.

closed 2017-12-13 by gaschler

joyde3382, 2019-03-21:
@letsbelopez Did you ever get it to work? I've been stuck with a very similar issue.

letsbelopez, 2019-03-21:
> @letsbelopez Did you ever get it to work? I've been stuck with a very similar issue.

Yes I did. gaschler's [comment](https://github.com/googlecartographer/cartographer_ros/issues/623#issuecomment-351397716) solved my problem: 

The frame names did not match. Look at the diff output he posted in the comment. Look at the values of the tracking frame, published frame in the .lua config and the child link value in the .urdf file

joyde3382, 2019-03-22:
@letsbelopez Ah i see. This might be a newbie question, but how do i know what my tracking and publishes frame should be set to? Is there a way to see it? I'm currently using an RPLidar.

letsbelopez, 2019-03-29:
@joyde3382 I think you can use TF Tree to see frames [tf view_frames](http://wiki.ros.org/tf/Debugging%20tools#Tf_tree_introspection)

I'm not sure if this is the code you're using but I was looking at [rplidar](http://wiki.ros.org/rplidar#Published_Topics) and under Parameters there's frame_id and default is set to 'laser.' I would test that out.