cartographer-project/cartographer_ros #624: Cartographer SLAM bad quality when mapping from above

monabf, 2017-12-12:
Hi all,

I have been using Cartographer for a few different things and want to thank you for making this great tool open source. I am trying to simulate a drone mapping a certain cube-like structure, and to send the simulated lidar data to Cartographer for SLAM. I managed to simulate the drone circling this structure, and to run Cartographer with that bag, and it worked fine. 

However, now I want the drone to fly in a cross-like pattern over the structure and map it from above. And suddenly, it doesn't seem like the SLAM catches on anymore: with odometry data provided it gives a reasonable trajectory but still drifts a lot, and without odometry it does not run at all. I have tried several things: making the drone much slower (which helped a lot), giving it an unrealistically large lidar swath so that it never loses sight of the structure during mapping (which changed nothing), tuning... But the trajectory and pointcloud outputted by Cartographer still aren't good enough. Maybe the SLAM algorithm has more trouble when the plane in which the drone moves is perpendicular to the plane in which the sensor is oriented. Otherwise I'm probably doing something wrong, and I would be very grateful for any kind of help.

I ran [rosbag_validate](https://gist.github.com/monabf/e0f0481ce87c7f8e32bfa57bb22e9aeb), and here is a link to the required [Git repo](https://github.com/monabf/cartographer_ros.git).
The structure the drone is mapping is confidential, therefore [here](https://drive.google.com/open?id=1VNicsLoFCaw9qH6xjrgVNasRb26R9Ybg) are two bags in which I replaced this structure with something similar found on the internet. One of the bags is a simulation of a regular swath lidar, the other of a very wide swath for debugging purposes. 

Thank you for your help!

gaschler, 2017-12-13:
In short, Cartographer currently does not support flying robots the way it extrapolates IMU input.
Currently, PoseExtrapolator and ImuTracker use IMU acceleration input only to extract gravity direction using a low-pass filter. Velocity is estimated by the difference of the last two scan matches or by odometry input. Yes, we decided against a Kalman filter. (Perhaps this explains why it cannot detect motion perpendicular to the plane in your case.)
We found this extrapolation approach best for mobile robots, cars, and sensors that are carried/moved around. 
To work with 3D laser scans from a flying vehicle, you would need to fork and make large changes to PoseExtrapolator and ImuTracker. 2D scans cannot be used with flying vehicles.

Concerning synthetic data, we found results gathered with synthetic data sometimes not very realistic. (Some simulation may be more realistic than other.)
Choosing the right algorithm and tuning should ideally be based on measurements.

monabf, 2017-12-13:
Thank you for your help. If I may, it should probably be mentioned somewhere that Cartographer can only be used as is for "rolling" robots that more or less only move inside a 2D plane. 

If I understood well, Cartographer does not support robots with 6 degrees of freedom because only pose and motion are estimated, but the attitude of the vehicle never is (would necessitate using tilt from odometry or full IMU data) . Is that what you think the problem is? Do you think the fact it cannot detect motion perpendicular to the plane comes from this lack of attitude estimation? Or could it rather come from the way the submaps are constructed, for example if there is an assumption somewhere stating they can only be around the robot and not under/over it? I would like to find out exactly what part of the algorithm gets lost when mapping under/over a robot, and if it does not come from the lack of attitude estimation and the lower accuracy of PoseExtrapolator then it is probably linked to scan matching.

Otherwise I agree that 2D scans cannot be used with flying vehicles (we only really use 3D) and that eventually we should use real measurements, but we were doing ok with the simulations as long as the drone moved more or less in a plane and was mapping around itself, comparable to a rolling robot.

gaschler, 2017-12-14:
1. How about adding Kalman filtering?
Cartographer did use an unscented Kalman filter in the past, it was removed in 
https://github.com/googlecartographer/cartographer/pull/378 and replaced by the current PoseExtrapolator approach. Both solutions are computationally efficient, but the current gives better results in practice. So, we are very hesitant to add a Kalman filter now.

2. Can the 3D pipeline work with 3D scans only under the camera?
The 3D pipeline uses HybridGrid, which is not just layers of ProbabilityGrid, but a voxel grid where all three dimensions are treated equally. So the data structure has no limitation here, they are "true 3D occupancy grids".
Scan matching in 3D does compare accumulated range data with a HybridGrid and then adds it to this and the previous one. You are right there is no grid-to-grid matching, but still this is unrelated to the question.
Yes, cartographer can very well recognize up-and-down motion. I would actually argue that laser range measurements pointing down will give very precise up-and-down results.
As I said, the only limitation is that accelerometer data is not used to predict translation.
Also, the angle of the 3D scanner should be wide in both directions, not just 4 stripes close to each other.

3. Generally, we would be glad to receive an open-source bag file with real data collected by an aerial vehicle, try to tune it and discuss good config parameters. It is always good to receive example data that is high quality and different from our exisiting scenarios.
I'd expect that for small, shaky quadrotors, IMU measurements would need to be high frequency and in sync with lidar time stamps.

monabf, 2017-12-14:
Thank you very much for your extensive answer! It is hard to know how 3D Cartographer works exactly since, to my knowledge, there is no paper explaining how it went from 2D to 3D.

After taking a closer look at the code, I can see that 2D mapping uses a 2D ProbabilityGrid but 3D uses a HybridGrid and therefore a real 3D occupancy grid, I missed that before. Great, that probably means that the use of IMU data is indeed the only thing stopping us from using Cartographer with a drone! We will implement the Kalman filter to use accelerometer data and see if that helps, we'll come back to you if any new problems come up.

As for the data, if you are interested in it as soon as we start doing real experiments mapping non-confidential structures I can post it here and we can discuss tuning and configuration!

closed 2017-12-20 by cschuet

monabf, 2018-01-09:
Hi again,

We ran simulations in which the drone is artificially stabilized (set IMU data manually so that drone doesn't tilt/vibrate/oscillate), and with those simulations Cartographer ran perfectly. This proves that you were right, and that the only thing stopping the SLAM from working properly is the fact that the faster drone dynamics are not taken into account by the Pose Extrapolator since it doesn't use all IMU data. 

We are now starting to implement a UKF inside the Pose extrapolator, using the [earlier Cartographer commit](https://github.com/googlecartographer/cartographer/pull/378) you talked about. Our new Cartographer version compiles but it does not run, several checks are triggered and even when they are all commented out the SLAM does not run properly (no trajectory output).

Here is the new [Git repo](https://github.com/monabf/cartographer_kalman_filter), the bag files are still [here](https://drive.google.com/open?id=1VNicsLoFCaw9qH6xjrgVNasRb26R9Ybg) (one with normal swath, one with wide swath, and one with wide swath and an artificially stable drone with constant IMU data, which runs with the usual Cartofgrapher version). The output of [rosbag_validate](https://gist.github.com/monabf/e0f0481ce87c7f8e32bfa57bb22e9aeb) on the first bag (normal swath) is still the same. 

Could you tell me if you can see why our version is not running ? Is there an obvious reason why numerous check functions that were not triggered before should be after our modifications ? Is there something missing/a mistake in what we did ?

Thank you in advance for your help!

gaschler, 2018-01-10:
Just to make sure, if you fork an old version of cartographer, you also need to use the corresponding version of cartographer_ros, there were some changes related to timestamps, among others.
It would be better to use github's fork mechanism (for both repos) so one can see what the diff to the old base is.

monabf, 2018-02-08:
Hi,
We now use the old UKF version of Cartographer in parallel with the current version, have aerial (drone) and underwater (rover) simulations and run Cartographer with the simulated lidar data. In the simulations the vehicle maps a certain structure, either by spiraling around it, or by going over it. Most of the time the UKF version of Cartographer performs much better. 
So I was just wondering: why exactly did you give up on the UKF and went to a much more basic state estimation? You said that gave better experimental results, I suppose that's only true for slow dynamics, more or less planar movements like with the bagpack? Did you ever figure out why the simple estimation works better than the UKF for you in those cases? I would very much like to know in which cases the UKF works better than the current Pose Extrapolator and why, so if you have any ideas or explanations or can tell me precisely why you chose to give up on the UKF that would be great!

Thanks for your help!

myavari, 2018-04-04:
Hi @monabf 

May I ask how far did you go with making the drone version of cartographer?

We need a SLAM library for drones and are very interested to use cartographer. I appreciate any hint or updates you share.

thanks

monabf, 2018-04-06:
Hi @myavari 

My quick fix for using Cartographer on a drone was to go back to an earlier version of the code, where a UKF was used in Pose Tracker for estimating the state of the vehicle, instead of the current Pose Extrapolator.
I used the versions of Cartographer from approximately June 30th 2017, commit number 17a22ed of cartographer and more or less corresponding 125aee3 of cartographer_ros. It worked rather ok, not as well with 3D than with 2D motion but still ok. Certain complex scenarios had trouble and I ended up more or less giving up on it, but if you want to do rather simple scenarios where the drone looks around itself and doesn't go too fast you should be fine.

Hope that helps!