cartographer-project/cartographer_ros #625: This file was generated by an older version of protoc

l1va, 2017-12-13:
Hi!
Running Ubuntu 16.04 and Gazebo7. All works ok but then i installed cartographer-ros.
It installs protoc 3.4.1. Cartographer is built successfully, but my project failed with :  
 ```
   In file included from /usr/include/gazebo7/gazebo/msgs/MessageTypes.hh:91:0,
                 from /usr/include/gazebo7/gazebo/msgs/msgs.hh:24,
                 from /usr/include/gazebo7/gazebo/physics/Entity.hh:28,
                 from /usr/include/gazebo7/gazebo/physics/Model.hh:32,
                 from /home/thesis/gazebo_animate_pose/animate_pose.cc:4:
/usr/include/gazebo7/gazebo/msgs/wrench_stamped.pb.h:17:2: error: #error This file was generated by an older version of protoc which is
/usr/include/gazebo7/gazebo/msgs/wrench_stamped.pb.h:18:2: error: #error incompatible with your Protocol Buffer headers. Please
/usr/include/gazebo7/gazebo/msgs/wrench_stamped.pb.h:19:2: error: #error regenerate this file with a newer version of protoc.
```
I understand that problem not in cartographer, but i believe you can help. Does cartographer can be build with older protoc? How to remove new and install old protoc? Or maybe any other ways i should check? How to work simultaneously with gazebo and cartographer if they require different protocs? 
I tried to install gazebo from source to build it with new protoc, but it failed too.
Many thanks!

ojura, 2017-12-14:
Have you tried cleaning your build tree?

gaschler, 2017-12-14:
This error message should appear when the protobuf compiler version during build time and the protobuf library version during runtime differ.
Protoc is the protobuf compiler used during build. In theory, you could have multiple versions installed, but it would be hard to manage.
Cartographer requires protobuf version >= 3.

Since, in your case, Gazebo requires protobuf library version 3.4.1, it will be easiest to uninstall all other versions, and fully clean and re-build cartographer(_ros).
Also, it is not necessary to install cartographer(_ros) in the system, so I see no reason why it should break other software (Gazebo).

l1va, 2017-12-14:
Thanks for the answer.
I understand the problem. 
Maybe i was not so clear, but vice verse - cartographer during installation installing protobuf 3.4.1 but gazebo was built with 2.6.1. In my project i use some message-files from gazebo (depends on it) and when i build my project - it fails. 
Cartographer was built successfully and it is not break anything and as you said it should not be installed, but cartographer force to install protobuf>=3. And protobuf >=3 is breaking work with gazebo. As i installed gazebo by apt (not from source) - i can not rebuild gazebo with new protobuf. When i tried to install gazebo from source and build with new protobuf - it failed too (but here i could make something wrong). 
The problem is that if i have to work with protobuf>=3 - i can't work with gazebo, if i will not install protobuf=>3 - i can't work with cartographer. The question - do you know how to manage work with too protobufs or can suggest any workaround (rebuild gazebo or install cartographer with protobuf<=3). And can you suggest right way of uninstalling protobuf and installing an old one?
ps. For now i manually downloaded protobuf 2.6.1 sources and installed it over protobuf 3.4.1 and seems my project is working with gazebo as before. It is because cartographer is not so necessary part in my project for now, but how to work with cartographer?

gaschler, 2017-12-14:
The least intrusive solution does not require any changes to system packages.
- Install protobuf compiler and (static) library 3.x not into the system but somewhere else.
- Configure cmake for cartographer[_ros] to use protobuf compiler 3.x and link the according static lib.
- If you create an application that links cartographer it needs to link the same protobuf library version.

l1va, 2017-12-15:
Thanks! It can be the solution. 
I installed protobuf to custom dir, but how to use it in catkin_make_isolated ? 
Am i on the right way? 
https://gist.github.com/l1va/630043c4a23e6b3f5c73c088ba6b62d5
What is wrong?

l1va, 2017-12-19:
Guys?

kdaun, 2018-01-17:
Based on @l1va 's script https://gist.github.com/kdaun/51b7d19bbcd5e0798c8415cd093078fb installs Cartographer with a local protobuf install. 
Are there specific reasons for a protobuf system install instead of a local install?
A local install is way less intrusive and therefore seems preferable to a system install.
I expect that the described Gazebo issue arises for other users as well.

gaschler, 2018-01-17:
You are right, when cartographer_ros is installed with catkin_make_isolated, it makes perfect sense to install protobuf in ./install_isolated.
System install is not necessary, and may have side effects.
@kdaun, is it possible to make a catkin package that does exactly that?

kdaun, 2018-01-17:
https://github.com/SafeDroneWare/protobuf_catkin provides a catkin integration for protobuf 3.4. I will check on the integration in the Cartographer workspace.

jfeldman24, 2018-10-23:
Is a detailed step-by-step start-to-finish resolution to this issue available? I have tried using kdaun's script  https://gist.github.com/kdaun/51b7d19bbcd5e0798c8415cd093078fb

 but I am still getting this error and am unable to compile cartographer while Protobuf2.6.1 is installed, despite protobuf3.4.1 being in my workspace from the above script. Instructions on what to change would be greatly appreciated!

jelavice, 2018-10-25:
The script doesn't work for me either. I also tried to install the protobuf 3.5.1 with a different prefix and then prepend this location to my PATH and LD_LIBRARY_PATH variables. It didn't work either.

l1va, 2018-10-26:
Firstly - to return broken 2.6:
To do it download release from here(https://github.com/protocolbuffers/protobuf/releases/tag/v2.6.1 ) , unzip it and do inside the dir:
``` 
./configure && make && make check && sudo make install
```
And how to install localy. I do it like that : 
https://gist.github.com/l1va/630043c4a23e6b3f5c73c088ba6b62d5

And after installing cartographer you have to use next command to build after changes in your project:
```    
catkin_make_isolated --install --use-ninja -DCATKIN_ENABLE_TESTING=false
source install_isolated/setup.bash
```

josephcoombe, 2018-12-06:
Did anyone figure out a way to use google_cartographer_ros and Gazebo together. 

I tried kdaun's script but it didn't work for me - fails during `catkin_make_isolated --install --use-ninja \
  -DCMAKE_PREFIX_PATH="${PWD}/install_isolated;${PWD}/protobuf/install;${CMAKE_PREFIX_PATH}"`?

I then tried to build `cartographer_ros` (with local protobuf 3 install) and my other packages (with system protobuf 2) in separate `catkin_ws`s via overlaying. However, (depending on how I did the overlay) either Gazebo or cartographer_ros would break when I tried to run them - due to Protobuf errors.

I uninstalled Gazebo binaries, installed system protobuf 3, and installed Gazebo from source. However the Gazebo build breaks when I do this.

ojura, 2018-12-06:
@josephcoombe Try Ubuntu 18.04 :-)

wirew0rm, 2019-02-19:
I have worked with @kdaun to improve the installation of a current protobuf3 when installing with ROS.
Starting from his local protobuf install script we came up with an approach using CMake's ExternalProject to allow cartographer to have it's own protobuf in `${catkin_workspace}/devel/opt/protobuf3_local` without any manual installation step.

I have compiled some [instructions](https://git.sim.informatik.tu-darmstadt.de/snippets/1) on how to have cartographer and voxblox (which uses the system protobuf) in the same workspace.
This is tested and works on Ubuntu 16.04 as well as 18.04.

This is the relevant additon to the [CMakeLists.txt](https://github.com/tu-darmstadt-ros-pkg/cartographer/blob/voxblox_workspace_compat/CMakeLists.txt#L44-L62). Maybe this would be worth including into the upstream makefile (only for catkin builds, of course)?

JerryZsy, 2019-04-12:
i am  confused by the problem ，in my project  with protobuf of 2.6.1. After making the so file, when i use it ,there will be such errors:error:
 #error This file was generated by an older version of protoc which is
 #error This file was generated by an older version of protoc which is

if I  in the source file, by #if 0 ，to stop version detection，it will be ok. But，i know it is not a good way to solve it.  And if if i use the newer version of protobuf, it also the same erro.

TouchDeeper, 2020-01-03:
Hello, Is there a script or example for cartographer (without ros) installation. I have installed protobuf 3.4.1 in a manually specified path.  And I modify the CMakelist.txt of cartographer according to the [answear](https://stackoverflow.com/questions/53651181/cmake-find-protobuf-package-in-custom-directory).
After `cmake .. -G Ninja`
protobuf3.4.1 can be found
```
-- Protobuf version : 3.4.1
-- Protobuf include path : /home/wang/CLionProjects/lidar_slam/L6/cartographer/protobuf/install/include
-- Protobuf libraries : /home/wang/CLionProjects/lidar_slam/L6/cartographer/protobuf/install/lib/libprotobuf.a
-- Protobuf compiler libraries : /home/wang/CLionProjects/lidar_slam/L6/cartographer/protobuf/install/lib/libprotoc.a
-- Protobuf lite libraries : /home/wang/CLionProjects/lidar_slam/L6/cartographer/protobuf/install/lib/libprotobuf-lite.a
```
But  when `ninja`
Cartographer still use protobuf2.6.1, output like below:
```
In file included from /usr/local/include/google/protobuf/generated_message_table_driven.h:35:0,
                 from ./cartographer/mapping/proto/trajectory_node_data.pb.h:25,
                 from ../cartographer/mapping/trajectory_node.h:26,
                 from ../cartographer/mapping/trajectory_node_test.cc:17:
/usr/local/include/google/protobuf/map_entry_lite.h: At global scope:
/usr/local/include/google/protobuf/map_entry_lite.h:532:26: error: ‘MapEntry’ in namespace ‘google::protobuf::internal::internal’ does not name a type
   friend class internal::MapEntry;
```
Could you please provide some advice?

MichaelGrupp, 2020-10-14:
Is this still an issue with newer Ubuntu releases? Because on Ubuntu 18 & 20 there is no need to install a custom Protobuf version for Cartographer, it works totally fine with the protobuf version from the official package archives (installed via apt or rosdep). This is also what is used in the CI Docker script now ([change](https://github.com/cartographer-project/cartographer_ros/commit/9cde1b4f5c3c4683df67c86868b11a8e4d31f1e4#diff-d37cbca5e1304461c1e35e446459cb780471dd12b2f563694cb1f706b8a22af2L48)).

MichaelGrupp, 2020-10-27:
I have updated the compilation instructions. Now it should work fine if you follow the instructions on a clean Ubuntu / ROS: https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html

closed 2020-10-27 by MichaelGrupp

HarshalDawar, 2022-01-25:
For future reference, in case of having to remove cartographer's older system wide protoc/protobuf install, one could check out osrf/gazebo#2496