cartographer-project/cartographer_ros #626: 2D || VPL-16 || w/o IMU || w/ odom

TobiMiller, 2017-12-13:
Hey,

i've got a bagfile containing a odom recorded by ZED camera and a velodyne pointcloud. I dont have any IMU data. So i need to use the 2D mapbuilder.

My question before i start to finetune for hours. Is it possible to configure cartographer with this setup to get satisfying results? Or is the only way to use 3D mapbuilder with IMU data? What data from the IMU will be used? only orientation? Why is it not using the data from odom? it should be precise enough.

until now i ve got everything running. but the output is not as expected. I cannot provide the bag files because of NDA :(

gaschler, 2017-12-13:
Right, if you don't have IMU data, you can only use the 2D pipeline.
Odometry input will be treated like wheel odometry, PoseExtrapolator expects odometry to be noisy because that is really the default case.
Sure it can give good results, but use mostly the VPL-16 data.

TobiMiller, 2017-12-14:
@gaschler 
Thanks for your answer. What information do you use from the IMU? Just the Orientation? Then we can fake the IMU message. We are just missing the acceleration part.

gaschler, 2017-12-14:
You shouldn't fake the IMU message.

If you are willing to put some effort into this project, the best solution I see with 3D laser scans and good 6D odometry is to create a fork and
1. replace PoseExtrapolator by your good 6D odometry. As a side effect, you no longer need IMU data.
2. insert 6D odometry constraints into the global optimization problem in mapping_3d/pose_graph/optimization_problem.cc similar to fixed frame pose constraints.

gaschler, 2017-12-18:
Closing as answered, feel free to re-open as needed.

closed 2017-12-18 by gaschler

kajep09, 2018-01-17:
Sorry to resurrect a closed issue, but I would like some clarification about the IMU data usage and the statement "You shouldn't fake the IMU message".

If you are only moving in 2D and using the 2D map builder does cartographer use both the IMU acceleration and angular velocity or only one of them?
In the cartographer_turtlebot repository there is a "flat_world_imu_node" which as far as I understand it takes incoming IMU messages and hardcodes the acceleration as (x = 0, y = 0, z = 9.8). In this case does cartographer still use the angular velocities? Otherwise what is the point in (partly) faking the IMU messages?