cartographer-project/cartographer_ros #629: Huge Drift In Map (Loop Not Closing)

shreyashub, 2017-12-18:
Hi,
I have used cartographer to create an outdoor map using a 3D lidar(Quanergy M8). Data is collected from a car moving at about 12 miles/hr in an dynamic environment.
It works fine with small maps([image](https://learnermanipal-my.sharepoint.com/personal/shreyas_kumar3_learner_manipal_edu/_layouts/15/guestaccess.aspx?docid=17fb4af73fa394f3b89d07a6a8b3be6ec&authkey=AbKOURMk85p2jZAS4HhN-Cc&expiration=2018-01-17T14%3A51%3A11.000Z&e=54b6461faf2a4d0d89e28711e6699a60), [bagfile](https://learnermanipal-my.sharepoint.com/personal/shreyas_kumar3_learner_manipal_edu/_layouts/15/guestaccess.aspx?docid=1dc4efe3067f4424f87074080b6cbedc5&authkey=AXvUPsKNPx6d2S5NEfqDfqM&expiration=2018-01-17T15%3A01%3A08.000Z&e=d9e24c367d6245039f29c040d4b89e90) ),
But with a large map, there is a huge drift( [image](https://learnermanipal-my.sharepoint.com/personal/shreyas_kumar3_learner_manipal_edu/_layouts/15/guestaccess.aspx?docid=1e828965789b0418e8422111619a87243&authkey=ARXPHgL1wDIV0cZmfi5fJ20&expiration=2018-01-17T14%3A54%3A35.000Z&e=c136c80f08dc4b3d9eb0f182ef0bf543),  [odometry from ins ](https://learnermanipal-my.sharepoint.com/personal/shreyas_kumar3_learner_manipal_edu/_layouts/15/guestaccess.aspx?docid=1fa3e745d88834e46b6cefa3f1219b113&authkey=AdfQoDcgU2VPGfuTdN4i4Aw&expiration=2018-01-17T14%3A58%3A53.000Z&e=1353c732c30f46ac81631bdfc16e4e70) )  
Even if I don't provide odometery to it, there is no difference in the map that is generated.
It also takes the ground as an obstacle and gives it a dark color.
Also is there an way to remove dynamic objects to not be mapped.
Attachments:
rosbag_validate output : [link](https://gist.github.com/shreyashub/8d5e321139a000bc07aeeea24945f994)
bag_file : [link](https://learnermanipal-my.sharepoint.com/personal/shreyas_kumar3_learner_manipal_edu/_layouts/15/guestaccess.aspx?docid=1d51ac7e3fc044426bab82f403d1bf128&authkey=ARXYlbNWuExlpTIG8sJ5WwU&expiration=2018-01-17T13%3A58%3A48.000Z&e=8fd5bd6d3be64dc69ee6ac6927a954df)
URDF, rosbag_validate CSV ,Config Files and Launch files [link](https://github.com/shreyashub/Cartographer-Files)
Forked Repo : [link](https://github.com/shreyashub/cartographer_ros)

gaschler, 2017-12-20:
Having just a quick look at your configuration, 
1. `TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 1` is too low, this is ~0.01 s, I'd try something in the 0.5 s range first.
2. Lidar has a very high jitter, I'd speculate you are dropping most frames.

Concerning your questions,
(4.) Of course ground is part of the submaps, it's good for scan matching.
(5.) There is no such feature.

shreyashub, 2017-12-20:
Changing `TRAJECTORY_BUILDER_3D.num_accumulated_range_data ` to 2 or 4, doesn't seem to make a big change. But I'll try again.
I'll try fixing the lidar data too.
In the demo bagfile, it assigns the ground as white(very light). But in my case it's pretty dark. Is that a problem?
Any other insights will be deeply appreciated.

gaschler, 2018-01-03:
Since your range data messages are very small, I'd recommend to try
`TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 50` or similar to accumulate one lidar revolution.

shreyashub, 2018-01-05:
Thanks, It (loop closure) works almost perfectly [(image)](https://learnermanipal-my.sharepoint.com/personal/shreyas_kumar3_learner_manipal_edu/_layouts/15/guestaccess.aspx?docid=1c9c84772baec46e182ec508ae6c93233&authkey=AXFgIgpvBoPkXaAwB43ByiI&expiration=2018-02-04T14%3A05%3A11.000Z&e=2af4ae7edae348f1bffe661baf769583).
But now the maps loose detail, and at a particular point its very hazy [old(clear)](https://learnermanipal-my.sharepoint.com/personal/shreyas_kumar3_learner_manipal_edu/_layouts/15/guestaccess.aspx?docid=12c514c17a96c4a22a2e254b1fe090fbd&authkey=AbgW4ziQq5BCmX0CEOmDaAc&expiration=2018-02-04T14%3A06%3A35.000Z&e=d24d63381e224865b39d4bcd92c3d40f)  , [new(not clear)](https://learnermanipal-my.sharepoint.com/personal/shreyas_kumar3_learner_manipal_edu/_layouts/15/guestaccess.aspx?docid=17cb5c26e0c4647219d732573da7a133a&authkey=ASDG771COFGAfEeftV9nPLE&expiration=2018-02-04T14%3A06%3A57.000Z&e=7d29e71879a2443688c21ccff996c00a)

gaschler, 2018-01-09:
I believe the result quality are limited by the high Lidar frame drop.

The color of the x-ray images in rviz are some projection of the submap data for visualization only, and IMHO not a strong indicator for the quality of the submaps.

Closing, feel free to re-open, for instance when better Lidar recordings are available.

closed 2018-01-09 by gaschler

tiberium24, 2018-12-11:
Hello, 

I am trying to use the same Quanergy M8 lidar with Cartographer in ROS. My approach is similar to yours. I am first trying to test it only with lidar point cloud without IMU or other sensors. 
I get the following error however and no map is built or scan match points published:

[image]

My launch file:

`<launch>
  <param name="robot_description"
    textfile="$(find cartographer_ros)/urdf/test1.urdf" />

  <node name="robot_state_publisher" pkg="robot_state_publisher"
    type="robot_state_publisher" />

  <node name="cartographer_node" pkg="cartographer_ros"
      type="cartographer_node" args="
          -configuration_directory $(find cartographer_ros)/configuration_files
          -configuration_basename test1.lua"
      output="screen">
    <remap from="points2" to="Sensor/points" />
 
  </node>

  <node name="cartographer_occupancy_grid_node" pkg="cartographer_ros"
      type="cartographer_occupancy_grid_node" args="-resolution 0.05" />

  <node name="rviz" pkg="rviz" type="rviz" required="true"/>
</launch>
`

My lua file:

`include "map_builder.lua"
include "trajectory_builder.lua"

options = {
  map_builder = MAP_BUILDER,
  trajectory_builder = TRAJECTORY_BUILDER,
  map_frame = "map",
  tracking_frame = "base_link",
  published_frame = "base_link",
  odom_frame = "base_link",
  provide_odom_frame = false,
  publish_frame_projected_to_2d = false,
  use_odometry = false,
  use_nav_sat = false,
  use_landmarks = false,
  num_laser_scans = 0,
  num_multi_echo_laser_scans = 0,
  num_subdivisions_per_laser_scan = 1,
  num_point_clouds = 1,
  lookup_transform_timeout_sec = 0.2,
  submap_publish_period_sec = 0.3,
  pose_publish_period_sec = 5e-3,
  trajectory_publish_period_sec = 30e-3,
  rangefinder_sampling_ratio = 1.,
  odometry_sampling_ratio = 1.,
  fixed_frame_pose_sampling_ratio = 1.,
  imu_sampling_ratio = 1.,
  landmarks_sampling_ratio = 1.,
}

TRAJECTORY_BUILDER_3D.num_accumulated_range_data = 65

MAP_BUILDER.use_trajectory_builder_3d = true
MAP_BUILDER.num_background_threads = 7
POSE_GRAPH.optimization_problem.huber_scale = 5e2
POSE_GRAPH.optimize_every_n_nodes = 320
POSE_GRAPH.constraint_builder.sampling_ratio = 0.03
POSE_GRAPH.optimization_problem.ceres_solver_options.max_num_iterations = 10
POSE_GRAPH.constraint_builder.min_score = 0.62
POSE_GRAPH.constraint_builder.global_localization_min_score = 0.66

return options
`

My urdf:

`<robot name="cartographer_backpack_3d">
  <material name="orange">
    <color rgba="1.0 0.5 0.2 1" />
  </material>
  <material name="gray">
    <color rgba="0.2 0.2 0.2 1" />
  </material>

  <link name="Sensor">
    <visual>
      <origin xyz="0.0 0.0 0.0" />
      <geometry>
        <cylinder length="0.07" radius="0.05" />
      </geometry>
      <material name="gray" />
    </visual>
  </link>

  <link name="base_link" />

  <joint name="Lidar_joint" type="fixed">
    <parent link="base_link" />
    <child link="Sensor" />
    <origin xyz="0.01 0. 0.19" rpy="0. -0.1745 3.1416" />
  </joint>

</robot>`

rdesc, 2018-12-12:
@tiberium24 just add ``TRAJECTORY_BUILDER_2D.use_imu_data = false`` in your lua file. By default it is set to true

Duke-Allen, 2019-03-03:
Hello, I am using a 16-lines LIDAR and IMU to build 3D map by cartographer.
I tun the parameters and run it with no error, the results are as follows:
[image]
[image]

There is a big gap with the real environment. I think the map is messy. What is the problem of this result? How can I solve it?
This is the validate result of my bag:

- E0227 22:30:57.092067 19056 rosbag_validate_main.cc:365] Average IMU linear acceleration is 10.6084 m/s^2, expected is [9.5, 10.5] m/s^2. Linear acceleration data should include gravity and be given in m/s^2.
I0227 22:30:57.093359 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/IMU" (frame_id: "imu"):
Count: 37968  Min: 0.004133  Max: 0.016211  Mean: 0.010000
[0.004133, 0.005340)	                    	Count: 2 (0.005268%)	Total: 2 (0.005268%)
[0.005340, 0.006548)	                    	Count: 3 (0.007901%)	Total: 5 (0.013169%)
[0.006548, 0.007756)	                    	Count: 8 (0.021070%)	Total: 13 (0.034239%)
[0.007756, 0.008964)	                    	Count: 828 (2.180784%)	Total: 841 (2.215023%)
[0.008964, 0.010172)	 ###################	Count: 35958 (94.706070%)	Total: 36799 (96.921089%)
[0.010172, 0.011380)	                    	Count: 345 (0.908660%)	Total: 37144 (97.829750%)
[0.011380, 0.012588)	                    	Count: 820 (2.159714%)	Total: 37964 (99.989464%)
[0.012588, 0.013796)	                    	Count: 1 (0.002634%)	Total: 37965 (99.992096%)
[0.013796, 0.015004)	                    	Count: 0 (0.000000%)	Total: 37965 (99.992096%)
[0.015004, 0.016211]	                    	Count: 3 (0.007901%)	Total: 37968 (100.000000%)
E0227 22:30:57.094650 19056 rosbag_validate_main.cc:382] Point data (frame_id: "rslidar") has a large gap, largest is 0.101409 s, recommended is [0.0005, 0.05] s with no jitter.
I0227 22:30:57.094745 19056 rosbag_validate_main.cc:398] Time delta histogram for consecutive messages on topic "/rslidar_points" (frame_id: "rslidar"):
Count: 3766  Min: 0.100213  Max: 0.101409  Mean: 0.100801
[0.100213, 0.100333)	                    	Count: 1 (0.026553%)	Total: 1 (0.026553%)
[0.100333, 0.100452)	                    	Count: 1 (0.026553%)	Total: 2 (0.053107%)
[0.100452, 0.100572)	                    	Count: 9 (0.238980%)	Total: 11 (0.292087%)
[0.100572, 0.100691)	                    	Count: 33 (0.876261%)	Total: 44 (1.168348%)
[0.100691, 0.100811)	      ##############	Count: 2717 (72.145515%)	Total: 2761 (73.313858%)
[0.100811, 0.100931)	               #####	Count: 972 (25.809877%)	Total: 3733 (99.123741%)
[0.100931, 0.101050)	                    	Count: 25 (0.663834%)	Total: 3758 (99.787575%)
[0.101050, 0.101170)	                    	Count: 6 (0.159320%)	Total: 3764 (99.946892%)
[0.101170, 0.101289)	                    	Count: 1 (0.026553%)	Total: 3765 (99.973450%)
[0.101289, 0.101409]	                    	Count: 1 (0.026553%)	Total: 3766 (100.000000%)

[image]

Here are my files:
[demo_rsldiar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/demo_rslidar_3d.launch)
[rslidar_3d.launch](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_3d.launch)
[mydemo_3d.rviz](https://github.com/Duke-Allen/cartographer-configuration/blob/master/mydemo_3d.rviz)
[rslidar_2d.urdf](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_2d.urdf)
[rslidar_scan_3d.lua](https://github.com/Duke-Allen/cartographer-configuration/blob/master/rslidar_scan_3d.lua)
[my data bag](https://pan.baidu.com/s/1chlbMpfUJeY-JdXoPzxzPw)
Do you know the problem?