cartographer-project/cartographer_ros #630: Map topic is being updated in pure localization mode

doisyg, 2017-12-18:
Hello,
I am doing experiment with the pure localization mode and one of the issue I have is that the map that is published on the `/map` topic is being updated with the new information (which is basically the submaps from the  `/submap_list` topic aggregated by the `occupancy_grid_node` node).
Is this the desired behavior? Is there a way to keep it constant?
The map is being updated with new obstacles and slightly resized/moved. As I would like to run navigation on top of it, this is a problem. For instance I would like the waypoints or goals that I defined against the initial map to stay valid. Same story for additional map layers that would be defined in respect to this map.
This is what I ran:

1. Generate a map with `roslaunch cartographer_ros offline_backpack_2d.launch bag_filenames:=BAG_ONE.bag` and close

2. Use the generated map pbstream file as an input for pure localization:
`roslaunch keylo_mapping try_2d_localization.launch map_filename:=BAG_ONE.bag.pbstream`
launch and config files: https://gist.github.com/doisyg

3. Try pure localization against another bag:
`rosbag play BAG_TWO.bag`

doisyg, 2017-12-18:
I shoot to fast. The submap trajectory 0, which corresponds to the previously build map stays unchanged, it is the new trajectory from the localization which can be a bit different. Closing. Sorry.

closed 2017-12-18 by doisyg

kajep09, 2018-01-05:
I was wondering about the same thing, but I don't entirely follow your own answer. Can you state the actions needed to be taken to stop the map topic from being updated while in localization mode? 
Do you still run the occupancy_grid_node alongside the cartographer localization or is there a specific order in which you need to finish / start trajectories or something else entirely :)? 
Any help is appreciated.

doisyg, 2018-01-05:
> Do you still run the occupancy_grid_node alongside the cartographer localization or is there a specific order in which you need to finish / start trajectories or something else entirely :)?

No, as the occupancy_grid_node is publishing a map on the ros format on the /map topic from an aggregation of all the submaps, if you want a fix map, don't run this node, or remap the map topic.

Just run map server with a static map genereted from the .pbstream that you use with pure localization in order to have a non changing map.

kajep09, 2018-01-05:
That was my thought, but in that case does the cartographer localization still provide the correct (or a usable) "map_frame" to "tracking_frame" transform or do you only use the odometry information from the localization with e.g. AMCL against a static map? If you let cartographer localization run for an extended amount of time (with an updating map), there is a very good chance that the newest map will have drifted somewhat compared to the starting map right? Perhaps I am thinking about it the wrong way, but I will try what you suggest the next time I am at the robot.

doisyg, 2018-01-06:
> That was my thought, but in that case does the cartographer localization still provide the correct (or a usable) "map_frame" to "tracking_frame" transform or do you only use the odometry information from the localization with e.g. AMCL against a static map?

From my limited experiments, cartographer in pure localization can work as a replacement of AMCL as it publish the appropriate tf transform map -> odom

> If you let cartographer localization run for an extended amount of time (with an updating map), there is a very good chance that the newest map will have drifted somewhat compared to the starting [...]

I did not have time to test it properly, and I will be interested by your results if you do. My idea was to make the robot come back to a known location periodically and to start a new trajectory with a known initial pose from there.