cartographer-project/cartographer_ros #635: Unable to produce map (scan topic from rplidar not utilised by cartographer

JnusK, 2017-12-26:
Hi, sorry but I am a novice at ROS and cartographer and while trying to set up cartographer with a RPLIDAR A2, I encountered the issue where the /scan topic published by the rplidar_ros is not being used by cartographer_node. I think it might be an issue with my configuration files. Here are my files [Link](https://github.com/JnusK/cartographer_ros ).

JnusK, 2017-12-26:
[image]
The rqt_graph

JnusK, 2017-12-26:
Realised my issue was that scan should be remapped to base_link instead

closed 2017-12-26 by JnusK

joyde3382, 2019-03-25:
Hey @JnusK I am having the exact same issue right now with an RPLidar a1. Are your config files up to date? I tried using the one you have in your github but it did not change anything. Also my rqt_graph looks exactly like yours. 

Any suggestions to what i might be missing? :)