cartographer-project/cartographer_ros #638: Generated map by cartographer has a large difference with reference floor plan

danningwang, 2017-12-27:
Hello,
Sorry to disturb you. Maybe you are still on holiday. 

I am doing my RnD project about comparing mapping methods like cartographer, gmapping, and so on. After tuning parameters again and again, but map generated by cartographer is still not very good enough. So I wish you can help me to point out my problem when you are free.

Actually poses(in the beginning part) generated by cartographer comparison   with ground truth poses are really good enough. And I saw several loops were closed very well during mapping process, which is surprising. I think cartographer is a good way to build accurate map, the problem is that tuning parameter is difficult for me.

My problems are:

1.  laser scans jump very quickly and was not alligned with corridor walls. so I fixed frames in myrobot.urdf and removed extra frames in my launch file using tf_remove.py provided by  cartographer and also filtered my bagfile using prefix_remove.py provided by cartographer.  I dont know if this phenomenon affect my generated map, but it seems not affect. 

2. map generated by cartographer has a large difference with reference floor plan. It is good in the beginning part(right side loop in my submited map) but worse in the end of corridor(top part ). I thought the reason was I did not use odometry data, but the fact is that map is worse when using odometry data than not using odometry data. During tuning process, I changed some parameters which are listed in myrobot.lua(some of them are commented now). 

3. In the map, you can see some obstacles in the corridor(I think it is generated by our operators during recording bagfile). Then I tried to use only base_laser_front/scan topic to build map to avoid these obstacles(because I want to use the map to do navigation part), but it generated a really bad map that cannot be used. 

I will push my bag file link and launch file soon. 

Looking forward to your reply. 
Happy new year!

danningwang, 2017-12-27:
This is my googledrive link which contains my bagfile and launch file and generated maps. 
https://drive.google.com/drive/folders/1onjPgTtOv3fMELIPbtKeUsYjnfCLavCm?usp=sharing

gaschler, 2018-01-03:
As explained in the issue template, I would be more efficient if you also 
1. report the results of `rosbag_validate` which does some checks on your sensor data, and
2. prepare a fork that contains all the configuration, launch, and URDF files.

danningwang, 2018-01-04:
Thank you for your reply! 

The following is my bagfile validate result.  I will push my files on github soon. 

$ rosrun cartographer_ros cartographer_rosbag_validate -bag_filename groundfloor.filtered.bag

E0104 10:41:11.314782 31324 rosbag_validate_main.cc:262] frame_id "base_laser_front_link" is send on multiple topics. It was seen at least on /base_laser_front/scan_raw and /base_laser_front/scan
E0104 10:41:11.336108 31324 rosbag_validate_main.cc:262] frame_id "base_laser_rear_link" is send on multiple topics. It was seen at least on /base_laser_rear/scan_raw and /base_laser_rear/scan
E0104 10:41:11.336249 31324 rosbag_validate_main.cc:262] frame_id "base_laser_front_link" is send on multiple topics. It was seen at least on /base_laser_front/scan_raw and /base_laser_front/scan
I0104 10:41:48.528877 31324 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/base_laser_front/scan_raw" (frame_id: "base_laser_front_link"):
Count: 117231  Min: 0.000000  Max: 0.083435  Mean: 0.039389
[0.000000, 0.008343)	          ##########	Count: 58616 (50.000427%)	Total: 58616 (50.000427%)
[0.008343, 0.016687)	                    	Count: 0 (0.000000%)	Total: 58616 (50.000427%)
[0.016687, 0.025030)	                    	Count: 0 (0.000000%)	Total: 58616 (50.000427%)
[0.025030, 0.033374)	                    	Count: 0 (0.000000%)	Total: 58616 (50.000427%)
[0.033374, 0.041717)	                    	Count: 0 (0.000000%)	Total: 58616 (50.000427%)
[0.041717, 0.050061)	                    	Count: 0 (0.000000%)	Total: 58616 (50.000427%)
[0.050061, 0.058404)	                    	Count: 0 (0.000000%)	Total: 58616 (50.000427%)
[0.058404, 0.066748)	                    	Count: 0 (0.000000%)	Total: 58616 (50.000427%)
[0.066748, 0.075091)	                    	Count: 10 (0.008530%)	Total: 58626 (50.008957%)
[0.075091, 0.083435]	          ##########	Count: 58605 (49.991043%)	Total: 117231 (100.000000%)
I0104 10:41:48.532030 31324 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/base_laser_rear/scan_raw" (frame_id: "base_laser_rear_link"):
Count: 117227  Min: 0.000000  Max: 0.083536  Mean: 0.039389
[0.000000, 0.008354)	          ##########	Count: 58614 (50.000427%)	Total: 58614 (50.000427%)
[0.008354, 0.016707)	                    	Count: 0 (0.000000%)	Total: 58614 (50.000427%)
[0.016707, 0.025061)	                    	Count: 0 (0.000000%)	Total: 58614 (50.000427%)
[0.025061, 0.033414)	                    	Count: 0 (0.000000%)	Total: 58614 (50.000427%)
[0.033414, 0.041768)	                    	Count: 0 (0.000000%)	Total: 58614 (50.000427%)
[0.041768, 0.050122)	                    	Count: 0 (0.000000%)	Total: 58614 (50.000427%)
[0.050122, 0.058475)	                    	Count: 0 (0.000000%)	Total: 58614 (50.000427%)
[0.058475, 0.066829)	                    	Count: 0 (0.000000%)	Total: 58614 (50.000427%)
[0.066829, 0.075182)	                    	Count: 22 (0.018767%)	Total: 58636 (50.019192%)
[0.075182, 0.083536]	          ##########	Count: 58591 (49.980808%)	Total: 117227 (100.000000%)
E0104 10:41:48.532248 31324 rosbag_validate_main.cc:316] Point data (frame_id: "base_link") has a large gap, largest is 0.158063 s, recommended is [0.0005, 0.05] s with no jitter.
I0104 10:41:48.536350 31324 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/scan_unified" (frame_id: "base_link"):
Count: 58602  Min: 0.072569  Max: 0.158063  Mean: 0.078794
[0.072569, 0.081119)	####################	Count: 58521 (99.861778%)	Total: 58521 (99.861778%)
[0.081119, 0.089668)	                    	Count: 69 (0.117743%)	Total: 58590 (99.979523%)
[0.089668, 0.098217)	                    	Count: 0 (0.000000%)	Total: 58590 (99.979523%)
[0.098217, 0.106767)	                    	Count: 0 (0.000000%)	Total: 58590 (99.979523%)
[0.106767, 0.115316)	                    	Count: 0 (0.000000%)	Total: 58590 (99.979523%)
[0.115316, 0.123866)	                    	Count: 0 (0.000000%)	Total: 58590 (99.979523%)
[0.123866, 0.132415)	                    	Count: 0 (0.000000%)	Total: 58590 (99.979523%)
[0.132415, 0.140964)	                    	Count: 0 (0.000000%)	Total: 58590 (99.979523%)
[0.140964, 0.149514)	                    	Count: 0 (0.000000%)	Total: 58590 (99.979523%)
[0.149514, 0.158063]	                    	Count: 12 (0.020477%)	Total: 58602 (100.000000%)
I0104 10:41:48.545804 31324 rosbag_validate_main.cc:332] Time delta histogram for consecutive messages on topic "/base/odometry_controller/odometry" (frame_id: "odom_combined"):
Count: 461665  Min: 0.002284  Max: 0.020956  Mean: 0.010004
[0.002284, 0.004151)	                    	Count: 36 (0.007798%)	Total: 36 (0.007798%)
[0.004151, 0.006019)	                    	Count: 81 (0.017545%)	Total: 117 (0.025343%)
[0.006019, 0.007886)	                    	Count: 720 (0.155957%)	Total: 837 (0.181300%)
[0.007886, 0.009753)	            ########	Count: 185573 (40.196465%)	Total: 186410 (40.377762%)
[0.009753, 0.011620)	        ############	Count: 273715 (59.288662%)	Total: 460125 (99.666428%)
[0.011620, 0.013487)	                    	Count: 1342 (0.290687%)	Total: 461467 (99.957115%)
[0.013487, 0.015355)	                    	Count: 152 (0.032924%)	Total: 461619 (99.990036%)
[0.015355, 0.017222)	                    	Count: 35 (0.007581%)	Total: 461654 (99.997620%)
[0.017222, 0.019089)	                    	Count: 5 (0.001083%)	Total: 461659 (99.998703%)
[0.019089, 0.020956]	                    	Count: 6 (0.001300%)	Total: 461665 (100.000000%)

danningwang, 2018-01-04:
Hi gaschler, 

This is my github link. https://github.com/danningwang/cartographer

My launch file is cartographer_sim.launch in launch folder.  My configuration file is myrobot.lua  and urdf is myrobot_2d.urdf. 

Thank you for your help.

gaschler, 2018-01-09:
I downloaded your scenario and found a couple of points.
My general impression is that local pose extrapolation is completely off and this issue is in part invisible because global SLAM can correct some of it.

1. High inter residuals
In the current configuration, there are wrong inter constraints found with high residuals.
Ideally, the lines should be very short to invisible:
[image]
These outlier constraints explain the double walls and mediocre quality.
I think the outliers happen because local slam is wrong.

2. Unusual tf usage
Honestly, I have no clue what magic you are doing with tf so I cannot check this part.

3. Local bias
What also looks fishy is that local SLAM has a bias to turn left while moving forward.
This could indicate a wrong urdf.
Please re-run without global SLAM and fix any obvious drift before turning on global SLAM again.

**Please start from tuning for local SLAM only.** It must be able to write a consistent map with very little drift over ~30 seconds. After that, you can tune global SLAM.

The only correction I can directly offer is to use both front and back laser scanners.
Cartographer performs better when you can see in >270 deg directions.

Here are the changes I tried, but it doesn't fix the root cause:

Use and show combined scan:
```
<remap from="scan" to="scan_unified" />

 <node name="rviz" pkg="rviz" type="rviz" required="true"
      args="-d $(find cartographer_ros)/configuration_files/demo_2d.rviz" />
```

Don't trust pose extrapolation (as a work-around):
```
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.translation_weight = 1
TRAJECTORY_BUILDER_2D.ceres_scan_matcher.rotation_weight = 0.1
```

Coarser grid to work with noisy input:
```
TRAJECTORY_BUILDER_2D.submaps.resolution = 0.08
```

Rule out some other things:
```
num_subdivisions_per_laser_scan = 1
TRAJECTORY_BUILDER_2D.num_accumulated_range_data = 2
TRAJECTORY_BUILDER_2D.use_online_correlative_scan_matching = false
```

Closing as resolved for now. Feel free to re-open if needed.

closed 2018-01-09 by gaschler